ghdl/ghdl #68: NULL access dereferenced

Paebbels, 2016-05-12:
From time to time GHDL shows `NULL access dereferenced` messages at runtime.

Currently, I haven't found a way for a deterministic reproduction. As far as I know my VHDL code doesn't use any access types. Besides normal VHDL code, it only uses protected and file types. The messages are printed at runtime, because some output from VHDL user-code is written to STDOUT.

The error message is:

```
C:\Tools\GHDL\0.34dev\bin\ghdl.exe:error: NULL access dereferenced
C:\Tools\GHDL\0.34dev\bin\ghdl.exe:error: simulation failed
```

Can you confirm that this error message is a message caused by GHDL itself, or is it caused by VHDL code? Normally I would expect a stack trace for GHDL's messages as in the past - very strange...

I did 5 other runs with the same testbench and neither aborted. I also try to run just one testbench and all of my testbenches in batch mode, still not reproducible in a deterministic way. (Batch mode, does purge the TempDir between testbench runs)

[image]

tgingold, 2016-05-15:
The error message can be due to null access in vhdl code (could easily happen if you use std.textio).
Unfortunately, to investigate I need a reproducer.  Which code generator are you using ?

Paebbels, 2016-05-16:
The backend is `mcode` on Windows 7 x86-64, maybe it's happening with `mcode` on Linux too, but I have only `llvm` backended GHDL versions on my Linux test machines.

So it's an error cause by VHDL and not internally by GHDL? I started several of these tests and I needed more than 30 runs to get another `simulation failed` message.

The *reproducer* is quite big.One solution is to checkout PoC from GitHub and run the testbench with GHDL:

```
git clone --recursive https://github.com/VLSI-EDA/PoC.git PoC
cd PoC
./poc.sh configure
./poc.sh -d ghdl PoC.sort.sortnet.*
```

The configuration process can be skipped, by pressing <kbd>p</kbd> until you reach the GHDL configuration. The the verbosity level of PoC can be controlled with the command line option `-d` for debug, `-v` for verbose, nothing for normal and finally `-q` for quiet. In verbose or debug mode, poc will show you the used GHDL calls incl. parameters. To run just a single test replace the `*` by `OddEvenMergeSort`. As far as I can tell, I only saw this bug occur if at least 2 testbenches were run consecutive.

Can I assist in any other way?

tgingold, 2016-05-17:
Can you confirm the issue only appears on Windows 7 with mcode, but not on Linux ?

I think this error message could also be generated internally by GHDL.

Paebbels, 2016-05-18:
I haven't seen this error on Linux with llvm backend (until now). I can try to compile GHDL with mcode backend in my VM and run the tests 100 times.

Paebbels, 2016-05-18:
Update for GHDL 0.34dev with mcode backend on Linux x86-64. 100 full testsuite runs, but no error.

tgingold, 2016-05-18:
On 18/05/16 11:15, Patrick Lehmann wrote:
> Update for GHDL 0.34dev with mcode backend on Linux x86-64. 100 full
> testsuite runs, but no error.

Can you try with mcode for linux i386 ?  This is more like the windows
configuration.

Paebbels, 2016-05-18:
Do I need a full i386 Linux configuration + i386 compiler environment to compile and run GHDL i386?
I thought you updated the mcode backend to x86-64 or is this Linux only?

tgingold, 2016-05-18:
On 18/05/16 20:34, Patrick Lehmann wrote:
> Do I need a full i386 Linux configuration + i386 compiler environment to
> compile and run GHDL i386?

You only need an i386 compiler + libraries.  Or you can download the 
latest release unless you need more advanced features.

> I thought you updated the mcode backend to x86-64 or is this Linux only?

Yes, x86-64 is not yet supported on windows.

Tristan.

Paebbels, 2016-05-18:
An i386 compile/runtime environment will exceed my VMs diskspace (0.9 GiB free).
It's the latest mcode release compiled with mcode x86-64 on Travis-CI?

tgingold, 2016-05-19:
Yes, all builds from Travis-CI are for x86-64.  So you should try with ghdl 0.33 (if this is possible).

Tristan.

Paebbels, 2016-05-19:
If I overlooked something? The GitHub release page for 0.33 has mostly x86-64 or amd64 builds. The only i386 build (ghdl_0.33-1jessie1_i386.deb) reports an LLVM backend. Is there any other download source for 0.33?

tgingold, 2016-05-20:
On 19/05/16 22:56, Patrick Lehmann wrote:
> Have I overlooked something?
>
> The GitHub release page for 0.33 has mostly x86-64 or amd64 builds. The
> only i386 build (ghdl_0.33-1jessie1_i386.deb) I can see, reports an LLVM
> backend. Is there any other download source for 0.33?

My fault.  I was pretty sure an mcode was built.

Paebbels, 2016-05-24:
I think we can close this. I strongly suppose that it can only reproduced in a very special setup.

GHDL on Windows running in a Python script, wrapped in a PowerShell script. Under special conditions, PoSh does not flush the buffer to the console, which might cause a back-pressure into GHDL, so std.textio or GHDL itself might get into trouble.

I'll try to investigate why PowerShell has his buffer issue. If necessary, I'll reopen this issue later.

closed 2016-05-24 by Paebbels

tgingold, 2016-05-25:
On 24/05/16 22:41, Patrick Lehmann wrote:
> I think we can close this. I strongly suppose that it can only
> reproduced in a very special setup.
>
> GHDL on Windows running in a Python script, wrapped in a PowerShell
> script. Under special [...]

Humm.

I will try to have a good traceback in case of NULL access.

Tristan.