ghdl/ghdl #69: Compiling with llvm/gcc backend and using VHDL-93/2008 simultanously

Paebbels, 2016-05-12:
Sorry to file another "issue" :)

We would like to compiling the vendor libraries e.g. from Xilinx ISE for VHDL-93 and VHDL-2008 into the same folder. Currently that's no problem for the `mcode` backend, because the generated `*.cf` files have a seperate suffix (`*-obj93.cf`/`*-obj08.cf`) and other files are not written to the disk.

But for the `gcc` and `llvm` backend, it's a problem to compile both language versions into one folder, because both backends generate `*.o` files. So compiling with the opposite VHDL version will overwrite existing `*.o` files.

Is it possible to add a suffix for VHDL-93/2008, too? E.g. `*.v93.o`/`*.v08.o` like for the shipped VHDL packages in `/lib/ieee/...`?

Our current, but nasty, workaround uses a symlink so switch the pre-compile folders before every GHDL run.

tgingold, 2016-05-15:
If I guess correctly you'd like to use the same set of -P for all the standards.  Is that correct ?
I am thinking about changing the meaning of -Pxxx. GHDL will search a library LIB at:
 xxx (like before), and
 xxx/LIB/vNN
Would that fit you need ?

Paebbels, 2016-05-16:
Yes I would like to use the same `-P` switch for all VHDL versions. It's hard enough to switch Xilinx libraries between ISE and Vivado source, I don't want to have another switching dimension for the VHDL version :).

So I think this should work for me.

Where is the `*.cf` file located?
In `xxx` or in `xxx/LIB`?

P.S. Be aware of case-sensitive file systems, when storing case insensitive VHDL library names on disk. Some big simulator vendors are still struggling about this issue.... :).

tgingold, 2016-05-18:
Now implemented and documented.

See testsuite/gna/issue69 for an example of use.

closed 2016-05-18 by tgingold