ghdl/ghdl #70: Compiler error with string in records (VHDL 2008 only)

thomashk0, 2016-05-13:
Hi,

I'm using the last GHDL release (but I also tested on 0.34 with LLVM backend):
```bash
$ ghdl -v
GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: GPL 2014 (20140331)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Compiler crashes when using string in records and compiling in VHDL2008. With other standards it produces a clean compile error:
```
error_string_in_records.vhd:10:5: element declaration of unconstrained array type "string" is not allowed
```
But instead I got:

```
******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2014 (20140331)
In directory: /mnt/nobackup/th243905/home/ghdl-bugs/
Command line:
/mnt/nobackup/th243905/tools/ghdl-0.33-x86_64-linux/libexec/gcc/x86_64-linux-gnu/4.9.3/ghdl1 --std=08 -P/mnt/nobackup/th243905/tools/ghdl-0.33-x86_64-linux/lib/gcc/x86_64-linux-gnu/4.9.3/vhdl//v08/std/ -P/mnt/nobackup/th243905/tools/ghdl-0.33-x86_64-linux/lib/gcc/x86_64-linux-gnu/4.9.3/vhdl//v08/ieee/ -quiet -o error_string_in_records.s error_string_in_records.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: trans-chap7.adb:4550
Call stack traceback locations:
0x61aee2 0x61bc9a 0x64bcd1 0x635b3a 0x65dd03 0x62d72a 0x65f7b5 0x5b9183 0x595a9e 0x96b87a 0x96d17e 0x5b91e9 0x592fb1 0x7f86db0b4b43 0x593457
******************************************************************

Execution terminated by unhandled exception
Exception name: TYPES.INTERNAL_ERROR
Message: trans-chap7.adb:4550
Call stack traceback locations:
0x61aee2 0x61bc9a 0x64bcd1 0x635b3a 0x65dd03 0x62d72a 0x65f7b5 0x5b9183 0x595a9e 0x96b87a 0x96d17e 0x5b91e9 0x592fb1 0x7f86db0b4b43 0x593457
ghdl: compilation error
```
The bug should be reproducible with the following script:

```bash
#/bin/bash

set -e

vhdl_file=error_string_in_records.vhd

cat > $vhdl_file << EOF
library ieee;
use ieee.std_logic_1164.all;

entity error_string_in_records is
end entity;

architecture test of error_string_in_records is

  type foo is record
    BAR : string;
  end record;

begin
end architecture;
EOF

ghdl -i --std=08 $vhdl_file
ghdl -m --std=08 error_string_in_records
```

Best regards

tgingold, 2016-05-15:
Yes, this 2008 feature is not yet supported.

tgingold, 2017-02-25:
Now partially supported.

closed 2017-02-25 by tgingold