ghdl/ghdl #71: Big output file with Adesto device library and crash of GtkWave

cclassic, 2016-05-20:
Hi,

I am currently trying to simulate a design using this DataFlash model:
http://www.adestotech.com/sites/default/files/tools_download/DataFlash_Vhdl_model.zip

While the simulation itself does run, it produces a rather big ghw file which does not open in GtkWave. It simply crashes with a Segmentation Fault. No additional info is displayed.

Not sure whether the problem is related to GtkWave or GHDL. Interesting to note though is that if I use vcd as output format, everything works perfectly fine. Also, the size of the .ghw file doesn't really seem to change if I simulate for a longer period of time. Additionally, the simulation as .vcd runs a lot faster than as .ghw.

Here are some file sizes of my current design for reference:
```
$ ./test --stop-time=1ns --wave=wave_short.ghw
$ ./test --stop-time=1us --wave=wave.ghw
$ ./test --stop-time=10us --wave=wave_long.ghw
$ ./test --stop-time=1ns --vcd=wave_short.vcd
$ ./test --stop-time=1us --vcd=wave.vcd
$ ./test --stop-time=10us --vcd=wave_long.vcd

$ ls -lah wave*
-rw-r--r-- 1 markus markus 4.2M May 20 19:40 wave_short.ghw
-rw-r--r-- 1 markus markus 6.3M May 20 19:39 wave.ghw
-rw-r--r-- 1 markus markus 6.3M May 20 19:55 wave_long.ghw
-rw-r--r-- 1 markus markus  31K May 20 19:40 wave_short.vcd
-rw-r--r-- 1 markus markus  84K May 20 19:39 wave.vcd
-rw-r--r-- 1 markus markus 492K May 20 20:02 wave_long.vcd

$ gtkwave wave.ghw 
GTKWave Analyzer v3.3.72 (w)1999-2016 BSI
Segmentation fault (core dumped)

$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 5.3.0

$ time ./test --stop-time=10us --vcd=wave_long.vcd
real	0m0.523s
user	0m0.420s
sys 	0m0.100s

$ time ./test --stop-time=10us --wave=wave_long.ghw
real	0m18.896s
user	0m18.817s
sys 	0m0.080s
```

Can anyone reproduce the issue? My project can be downloaded here: https://notsyncing.net/downloads/temp/flashrom-wb-0.zip (Directory names denote the library)
I compile using these flags: `--std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/path/to/altera/lib`

Should I open a bug report for GtkWave? Though I think the main problems lies within GHDL here...

Markus

tgingold, 2016-05-22:
On 20/05/16 20:09, Markus wrote:
> Hi,
>
> I am currently trying to simulate a design using this DataFlash model:
> http://www.adestotech.com/sites/default/files/tools_download/DataFlash_Vhdl_model.zip
>
> While the simulation itself does [...]

I will try to investigate the GtkWave issue.

> Not sure whether the problem is related to GtkWave or GHDL. Interesting
> to note though is that if I use vcd as output format, everything works
> perfectly fine. Also, the size of the .ghw file doesn't [...]

The vcd file is much smaller than than the ghw because arrays of arrays 
aren't dumpable in vcd.

> Can anyone reproduce the issue? My project can be downloaded here:
> https://notsyncing.net/downloads/temp/flashrom-wb-0.zip (Directory names
> denote the library)
> I compile using these flags: |--std=93 [...]

I still have to try with gtkwave.
You can use vcd or fst format instead of ghw.

Regards,
Tristan.

cclassic, 2016-05-22:
Oh right, now it makes sense that the vcd is a lot smaller...

Would it help if I uploaded my .ghw file?

Also, I just managed to also break the .fst format with my code^^. This time, the ghdl-generated binary crashes with a segmentation fault:
```
markus@Core-i7:/tmp/SigasiCompileCache4458821358421238926/flashrom-wb/mentor$ ./bench_hwbench_initial_tests --stop-time=10us --fst=1.fst
Segmentation fault (core dumped)
markus@Core-i7:/tmp/SigasiCompileCache4458821358421238926/flashrom-wb/mentor$ ./bench_hwbench_initial_tests --stop-time=10us --vcd=1.vcd
[...]
./bench_hwbench_initial_tests:info: simulation stopped by --stop-time
```
The crash disappears when I remove the `pageRam_inst` instantiation from design/flashrom-wb.vhd

Code link: https://notsyncing.net/downloads/temp/flashrom-wb-fstFail.zip
Top level: hwbench/hwbench_initial_tests

Should this go into a separate bug report? Can I help in any way to troubleshoot these issues?

Thanks,
Markus

tgingold, 2016-05-23:
On 22/05/16 20:53, Markus wrote:
> Oh right, now it makes sense that the vcd is a lot smaller...
>
> Would it help if I uploaded my .ghw file?
>
> Also, I just managed to also break the .fst format with my code^^. This
> time, the [...]

No, no need to file the ghw file and no need to create a new report.

I can reproduce a crash and I am investigating.

Regards,
Tristan.

tgingold, 2016-05-25:
I cannot reproduce the crash with flashrom-wb-fstFail.zip, but I get warnings during elaboration.
Which version of Quartus are you using ?

Tristan.

cclassic, 2016-05-25:
I use Quartus Prime 15.1 Build 185.

Here are the commands I issue compile the design:
```
ghdl -i --work=ip --workdir=/tmp/sim --std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/alteraLibPath/altera  /sourceDir/ip/intercon/rwp_ram_package.vhd /sourceDir/ip/altera/pageRam.vhd /sourceDir/ip/intercon/fifo_package.vhd /sourceDir/ip/intercon/rwp_ram.vhd /sourceDir/ip/intercon/wishbone_package.vhd /sourceDir/ip/altera/spiFifo.vhd /sourceDir/ip/intercon/crossbar_package.vhd /sourceDir/ip/intercon/dma.vhd /sourceDir/ip/intercon/test_slave.vhd /sourceDir/ip/intercon/fifo.vhd /sourceDir/ip/intercon/clock_crossing.vhd /sourceDir/ip/intercon/test_slave_package.vhd /sourceDir/ip/intercon/crossbar.vhd /sourceDir/ip/intercon/clock_crossing_package.vhd /sourceDir/ip/intercon/dma_package.vhd

ghdl -i --work=design --workdir=/tmp/sim --std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/alteraLibPath/altera  /sourceDir/design/flashrom-wb.vhd

ghdl -i --work=hwbench --workdir=/tmp/sim --std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/alteraLibPath/altera  /sourceDir/hwbench/hwbench_initial_tests.vhd

ghdl -i --work=bench --workdir=/tmp/sim --std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/alteraLibPath/altera  /sourceDir/bench/devicemodels/dataflash2.1.vhd /sourceDir/bench/bench_test.vhd /sourceDir/bench/bench_hwbench_initial_tests.vhd

ghdl -m --std=93 --work=bench --workdir="/tmp/sim"  --ieee=synopsys -fexplicit -frelaxed-rules -P/alteraLibPath/altera bench_hwbench_initial_tests
--> WARNINGS

/tmp/sim/bench_hwbench_initial_tests --stop-time=100ns --fst=bench_hwbench_initial_tests.fst
--> SEGFAULT
```

Full unedited log:
```
markus@Core-i7:/tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor$ ghdl -i --work=ip --workdir=/tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor --std=93  --ieee=synopsys -fexplicit -frelaxed-rules -P/home/markus/projects/ext/ghdl-github/ghdl/libraries/vendors/altera  /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/rwp_ram_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/altera/pageRam.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/fifo_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/rwp_ram.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/wishbone_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/altera/spiFifo.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/crossbar_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/dma.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/test_slave.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/fifo.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/clock_crossing.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/test_slave_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/crossbar.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/clock_crossing_package.vhd /home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/dma_package.vhd
markus@Core-i7:/tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor$ ls
ip-obj93.cf
markus@Core-i7:/tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor$ cat ip-obj93.cf 
v 4
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/dma_package.vhd" "e37086c07b63a7f3794cf7e36908416cb359b79a" "20160525190940.928":
  package dma_package at 1( 0) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/clock_crossing_package.vhd" "ed8530e7d40c617cfde40dcbf2ed5d27051446d2" "20160525190940.928":
  package clock_crossing_package at 1( 0) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/crossbar.vhd" "352c8556282d1d65c7f1e233258a7ecac0988142" "20160525190940.928":
  entity crossbar at 19( 768) + 0 on 4;
  architecture rtl of crossbar at 45( 1634) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/test_slave_package.vhd" "bc179b4e9bdfcf009fb39fca28c0acf8314591c9" "20160525190940.928":
  package test_slave_package at 1( 0) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/clock_crossing.vhd" "acea15eabd502074a351072c1ba192e04c0f19d5" "20160525190940.928":
  entity clock_crossing at 1( 0) + 0 on 4;
  architecture rtl of clock_crossing at 24( 626) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/fifo.vhd" "8e3c3a49a67c44a35875b13413318c96fe43fb4e" "20160525190940.928":
  entity fifo at 1( 0) + 0 on 4;
  architecture rtl of fifo at 31( 887) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/test_slave.vhd" "5fa24e9df813d14c3c92f1d86749b5cc5804b7f0" "20160525190940.928":
  entity test_slave at 1( 0) + 0 on 4;
  architecture rtl of test_slave at 20( 394) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/dma.vhd" "156bdf105cea23e4b56ccfa5c2f0957da21f69bd" "20160525190940.928":
  entity dma at 1( 0) + 0 on 4;
  architecture rtl of dma at 35( 1020) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/intercon/crossbar_package.vhd" "a9f4c253609dcb8f120b3d4ffedf46a819185b32" "20160525190940.928":
  package crossbar_package at 15( 666) + 0 on 4;
file / "/home/markus/projects/sigasiStudio/flashrom-wb/ip/altera/spiFifo.vhd" "2959f11e51ba60d99e7b79cfbb3fe7dd287af2df" "20160525190940.928":
  entity spififo at 37( 1429) + 0 on 4;
  architecture syn of spififo at 59( 1836) + 0 on 4;
[... cut ...]
/home/markus/projects/sigasiStudio/flashrom-wb/ip/altera/spiFifo.vhd:101:8:warning: port "eccstatus" of entity "dcfifo" is not bound
/home/markus/projects/sigasiStudio/flashrom-wb/ip/altera/spiFifo.vhd:59:14:warning: (in default configuration of spififo(syn))
markus@Core-i7:/tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor$ /tmp/SigasiCompileCache7987690334949030508/flashrom-wb/mentor/bench_hwbench_initial_tests --stop-time=100ns --fst=bench_hwbench_initial_tests.fst
Segmentation fault (core dumped)
```

cclassic, 2016-05-25:
I just cloned and compiled ghdl again so that I am on the latest commit. The error still persists. That's how I compile it: https://git.notsyncing.net:8080/fpga/build-ghdl/src/77fdf021a5f24e16dbb121f072ed13fa061e63f2/build.sh

Anything wrong with that?

tgingold, 2016-05-27:
On 25/05/16 21:42, Markus wrote:
> I just cloned and compiled ghdl again so that I am on the latest commit.
> The error still persists. That's how I compile it:
> https://git.notsyncing.net:8080/fpga/build-ghdl/src/77fdf021a5f24e16dbb121f072ed13fa061e63f2/build.sh
>
> Anything wrong with that?

That looks correct.

I have just pushed a change that fixes a crash in fst dump.  Could you 
try it?
I don't know whether it fixes your issue or not, as I wasn't able to 
reproduce it exactly.

Tristan.

cclassic, 2016-05-27:
Yes it did fix the issue. Fst works now.

Markus

tgingold, 2016-05-28:
On 27/05/16 15:50, Markus wrote:
> Yes it did fix the issue. Fst works now.

Great. I close this issue.

Tristan.

closed 2016-05-28 by tgingold

cclassic, 2016-05-28:
What about the .ghw / GtkWave issue? Did you find anything on that side, because it still doesn't work for me? Should I create a bug report on their project?

.fst does not have all features of .ghw, correct? (Like enums or arrays of arrays)

tgingold, 2016-05-29:
On 28/05/16 08:58, Markus wrote:
> What about the .ghw / GtkWave issue? Did you find anything on that side,
> because it still doesn't work for me? Should I create a bug report on
> their project?

Which version of gtkwave are you using ?  I get no crash with 3.3.73
(just built).

> .fst does not have all features of .ghw, correct? (Like enums or arrays
> of arrays)

Correct.

Tristan.

cclassic, 2016-05-29:
I just compiled GtkWave from https://sourceforge.net/p/gtkwave/code/HEAD/tree/gtkwave3/ (GTKWave Analyzer v3.3.73 (w)1999-2016 BSI). However, it still crashes.

Can you open my output file?  https://notsyncing.net/downloads/temp/bench_hwbench_initial_tests.ghw

cclassic, 2016-05-29:
Okay I found the problem. It is an error by the package maintainer in Arch. He compiled the program without compiler optimization which is required for some part of GtkWave. I will contact him.

Thank you for your help! This issue can remain closed.

tgingold, 2016-05-30:
On 29/05/16 09:40, Markus wrote:
> Okay I found the problem. It is an error by the package maintainer in
> Arch. He compiled the program without compiler optimization which is
> required for some part of GtkWave. I will contact him.

Yes, I can confirm that my compiled version of gtkwave opens your ghw 
file without crashing.

> Thank you for your help! This issue can remain closed.

Thank you for the status update.

Tristan.