ghdl/ghdl #72: Library use clause with identifier does not work.

pmarkkula, 2016-05-25:
"use work.lib.all" works, but "use work.lib.some_type_from_lib" does not.

[issue.zip](https://github.com/tgingold/ghdl/files/282451/issue.zip)

Paebbels, 2016-05-25:
What are your GHDL command line calls for analyze, elaborate and run?

pmarkkula, 2016-05-25:
Commands:

ghdl_mcode -a --std=08 issue_pkg.vhdl
ghdl_mcode -a --std=08 issue.vhdl

Output:
issue.vhdl:24:16: no function declarations for operator "="

Paebbels, 2016-05-25:
That's strange, because there seams to be no complain about line 11, where `t_one_two` is used... It's complaining at the equal operator

mitchsm, 2016-05-25:
The operators that get created for custom types are not made visible when you "use" the type, since you've only selected the type, not all of the operations that apply to it.

pmarkkula, 2016-05-25:
Is this the way it should be? I kind of assumed that it should work, because I have used it successfully with synthesis tools, but I don't really know the standard that well.

tgingold, 2016-05-25:
On 25/05/16 20:10, pmarkkula wrote:
> Is this the way it should be? I kind of assumed that it should work,
> because I have used it successfully with synthesis tools, but I don't
> really know the standard that well.

I think the VHDL standard is clear about that:

10.4 Use clauses

Each selected name in a use clause identifies one or more declarations 
that will potentially become directly visible. If the suffix of the 
selected name is a simple name, character literal, or operator symbol, 
then the selected name identifies only the declaration(s) of that simple 
name, character literal, or operator symbol contained within the package 
or library denoted by the prefix of the selected name.

So you should add:
use work.issue_pkg."=";

[ I agree this is a weird rule...]

Tristan.

pmarkkula, 2016-05-25:
> [ I agree this is a weird rule...]

I can totally understand why tool vendors chose to automatically include operators that get generated for user defined types.

Should I just close this issue?

tgingold, 2016-05-25:
On 25/05/16 20:35, pmarkkula wrote:
>     [ I agree this is a weird rule...]
>
> I can totally understand why tool vendors chose to automatically include
> operators that get generated for user defined types.

Not that the rules have changed with vhdl 2008: implicit declarations 
for a type are automatically made visible.

This is currently not implemented in GHDL, so...

> Should I just close this issue?

... I keep this ticket open to implement it.

Regards,
Tristan.

tgingold, 2016-05-31:
Now implemented.
--std=93 and --std=08 work according to the vhdl LRM.
Use -frelaxed-rules (with --std=93) to have direct visibility on operators.

Tristan.

closed 2016-05-31 by tgingold