ghdl/ghdl #73: crash in trans-chap7.adb:827

tsailer, 2016-05-26:
$ ghdl -a x.vhd

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 5.3.1 20160406 (Red Hat 5.3.1-6)
In directory: /var/tmp/g/
Command line:
/usr/libexec/gcc/x86_64-fedora_ghdl-linux/4.9.2/ghdl1 -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/std/ -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/ieee/ -quiet -o x.s x.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: trans-chap7.adb:827 access check failed
Call stack traceback locations:
0x5e0e05 0x5e0f46 0x5e4ae6 0x5e4d2e 0x613d67 0x61695e 0x618c2d 0x62e8d8 0x62135a 0x608d19 0x63405e 0x600293 0x635dc5 0x58234d 0x55ea51 0x95b54f 0x95d286 0x5823a9 0x55bde7 0x7ff1cf40a57e 0x55c427 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: trans-chap7.adb:827 access check failed
Call stack traceback locations:
0x5e0e05 0x5e0f46 0x5e4ae6 0x5e4d2e 0x613d67 0x61695e 0x618c2d 0x62e8d8 0x62135a 0x608d19 0x63405e 0x600293 0x635dc5 0x58234d 0x55ea51 0x95b54f 0x95d286 0x5823a9 0x55bde7 0x7ff1cf40a57e 0x55c427 0xfffffffffffffffe
ghdl: compilation error

Catchpoint 2, CONSTRAINT_ERROR at 0x00000000005e0e07 in trans.chap7.translate_implicit_array_conversion (expr=..., expr_type=852, res_type=1450, loc=887)
    at /usr/src/debug/gcc-4.9.2-20141101/gcc/vhdl/trans-chap7.adb:827
827	      case Ainfo.Type_Mode is
(gdb) p Ainfo
$1 = (access trans.ortho_info_type) 0x0
(gdb) p Einfo
$2 = <optimized out>
(gdb) p Res_Type
$3 = 1450
(gdb) p Expr_Type
$4 = 852
(gdb) p Disp_Type(Res_Type, "")
$5 = "the anonymous  defined at x.vhd:25:11"
(gdb) p Disp_Type(Expr_Type, "")
$6 = " ""ya_t"""

tsailer, 2016-05-26:
Reproducer
[x.vhd.txt](https://github.com/tgingold/ghdl/files/283919/x.vhd.txt)

tgingold, 2016-05-26:
> Reproducer
> x.vhd.txt <https://github.com/tgingold/ghdl/files/283919/x.vhd.txt>

I can reproduce the crash, I will investigate.

tgingold, 2016-05-28:
Now fixed.  Than you for the report.

closed 2016-05-28 by tgingold