ghdl/ghdl #83: Embed vendor library compile scripts into makefile for installation.

Paebbels, 2016-06-17:
I'm currently reworking the vendor library compile scripts for Linux (see #82). I'm filing this issue, so nobody forgets about this tasks (either you or me :) ).

The `make install` target should be extended to copy these Bash scripts into the installation directories. I'm not sure where Linux distributions usually store such scripts, which are not "binaries".

Further extensions to make and configure, but not enabled by default:
1. New `configure` option like this:  
    `--with-altera-library=/opt/altera/16.0/quartus/eda...` to set the Altera path
    `--with-lattice-library=/opt/diamond/3.7_x86-64/simulation/...` to set the Lattice path
    `--with-xilinx-library=/opt/xilinx/14.7/ISE_DS/...` to set the Xilinx path
2. New `make` targets:  
    `make libs`
    `make lib.altera`
    `make lib.lattice`
    `make lib.xilinx-ise`

A `make install` could copy the result into the installation folder like `ieee`.

---------
**Note**: Compiling Altera, Lattice + 2x Xilinx requires circa 500 MB (large files where skipped) with llvm backend and takes > 1 hour to compile.

tgingold, 2016-06-17:
On 17/06/16 04:50, Patrick Lehmann wrote:
> I'm currently reworking the vendor library compile scripts for Linux
> (see #82 <https://github.com/tgingold/ghdl/pull/82>). I'm filing this
> issue, so nobody forgets about this tasks (either you or me [...]

What about lib/ghdl/vendors ?

Tristan.

Paebbels, 2016-06-17:
What `lib` directory exactly? My Debian has at least 4 `lib` directories :).

tgingold, 2016-06-18:
On 17/06/16 13:01, Patrick Lehmann wrote:
> What |lib| directory exactly? My Debian has at least 4 |lib| directories :).

In the lib directory of the ghdl installation.

Paebbels, 2016-06-19:
`lib/ghdl/vendors` sounds good to me. Alternative: `lib/ghdl/scripts`

Can you append such an installation rule to the makefile(s). I'm not sure which one is responsible for such a copy/install task.

What needs to be copied?
- common files
  - config.sh
  - shared.sh
- compile scripts
  - compile-altera.sh
  - compile-lattice.sh
  - compile-osvvm.sh
  - compile-vunit.sh
  - compile-xilinx-ise.sh
  - compile-xilinx-vivado.sh
- GRC rules
  - ghdl.grcrules
  - ghdl.skipwarning.grcrules

I think it could be simplified to `*.sh` and `*.grcrules`.

closed 2016-06-22 by tgingold

Paebbels, 2016-06-22:
The bash scripts are not executable.

Adding exec rights in a lib directory is strange, should we recommend to launch the scripts with bash?