ghdl/ghdl #85: compile-lattice.sh: unknown option '-o' for command '-a'

Paebbels, 2016-06-17:
@cclassic

I tested the reworked `compile-lattice.sh` in my Debian virtual machine with Bash and in my MinGW64 Bash on Windows 7. The latter one uses GHDL with mcode backend from Windows. It reports:

```bash
Paebbels@ite190 MINGW64 /d/git/PoC/temp/precompiled/ghdl (paebbels/master)
$ /d/git/GitHub/ghdl/libraries/vendors/compile-lattice.sh -a --src /d/Lattice/diamond/3.7_x64/cae_library/simulation/vhdl/
Creating vendor directory: '/d/git/PoC/temp/precompiled/ghdl/lattice'
Compiling library 'ec'...
Analyzing file '/d/Lattice/diamond/3.7_x64/cae_library/simulation/vhdl/ec/src/ORCA_CMB.vhd'
C:\Tools\GHDL\0.34dev\bin\ghdl.exe: unknown option '-o' for command '-a'
Analyzing file '/d/Lattice/diamond/3.7_x64/cae_library/simulation/vhdl/ec/src/ORCA_SEQ.vhd'
C:\Tools\GHDL\0.34dev\bin\ghdl.exe: unknown option '-o' for command '-a'
...
```

The mcode back-end supports no `-o <path>` option. I assume GHDL will have the same behavior on Linux with mcode back-end. **What was you intention to explicitly specify an output filename?**

My rework of your initial script, uses the new GHDL search rules for the `-P` option.
So each VHDL library is created into an own directory wherein a version directory is created.

Example:

```Bash
D:\git\PoC\temp\precompiled\ghdl\
  altera\
    ...
  lattice\
    ec\
      v93\
    ecp\
      v93\
    ecp2\
      v93\
    ...
  xilinx-ise\
    ...
```

**Is this renaming still required?**

You can contact me directly by [email](https://github.com/VLSI-EDA/PoC/blob/master/AUTHORS.md), so we can skip the issue reporting step on Tristans repo :). Or is there any other way to directly contact GitHub users?

cclassic, 2016-06-18:
Whoops sorry, the -o was just a remnant of when I was trying to compile everything into a single directory (see #63). It can be safely removed. I tested it without and everything seems to compile (mcode and gcc).

I am currently experiencing some problems with the created lib though. I'll look into that later today. I'll also take a look and getting it to work with VHDL 2008. (Though I remember some problems with vitals and such...)

**About the Altera libraries:**
* When I run compile-altera.sh without any parameters, it does nothing and produces (-a required):
```
markus@Core-i7:~/projects/ext/ghdl-github/ghdl/libraries/vendors$ ./compile-altera.sh 
Creating vendor directory: '/home/markus/projects/ext/ghdl-github/ghdl/libraries/vendors/altera'
--------------------------------------------------------------------------------
Compiling Altera Quartus libraries [SUCCESSFUL]
```
* You forgot to initialize SKIP_LARGE_FILES to 0 at the top -> creates a syntax error when bash evaluates the if conditions further down.
* Improvement: When I compile using -a, it does not detect which devices I have installed and will fail when for example the Arria devices are missing. Maybe auto detect if the device directory exists and skip?

Apart from that, I can confirm that the scripts for Altera work (mcode and gcc). I only tested a FIFO for a Cyclone V but if that works the rest should be OK too.

Markus

Paebbels, 2016-06-19:
Hello Markus,

I'll push an update to fix this issues (see #88).

* Option `-o` is now removed.
* If no parameter is given, the help page will be displayed.  
  (I changed variable types from string ("TRUE"/"FALSE") to integer 0/1 for most boolean values.
* `SKIP_LARGE_FILES`: This should be fixed now
* VHDL-2008: I think we need a new VHDL-2008 flavor to solve this.  
  Such a new flavor should help to compile Altera and Xilinx primitives for 2008. See #86.

**Detecting installed devices:**
Yes, this is an unsolved issue. Unfortunately, Altera has no device directories.
There are only files like `arriaii_atoms.vhd` so I need to check if a file with a certain prefix exists.

Paebbels, 2016-06-20:
@cclassic The commit was merge and is available for testing.

**Known issues:**
 1. Compile only available Altera devices.
 2. Mac OS: `readlink` does not the same thing on Mac OS as on Linux. I have already found a work around, which will be committed soon. The work around requires the `coreutils` package, which can be installed with Homebrew. It installs a GNU readlink clone called `greadlink`, which will be chosen if `Darwin` is detected.

tgingold, 2016-06-22:
Can I close this ticket ?

cclassic, 2016-06-22:
The scripts are OK. We are currently investigating a related problem, but this ticket can be closed.

Paebbels, 2016-06-22:
Everything should be fixed, accept for "Known Issue 1", which could be considered a feature for a next script update :).

closed 2016-06-22 by Paebbels

arcturus140, 2019-09-17:
> The mcode back-end supports no -o <path> option. I assume GHDL will have the same behavior on Linux with mcode back-end. What was you intention to explicitly specify an output filename?

I know this question has arised times before, so, instead of a feature request I wanted to share some additional input. 

Currently my build system is expecting an object file with the naming `.vhd.o` or `.vhdl.o`. I am trying to override the build rule ATM. As a workaround I can copy the files for both locations. 

The lack of support for `-o` argument has given me issues early on, since the build system was not designed with VHDL in mind. Meanwhile I found the proper way to do it - also I am not sure how far I'd have gotten with my hacks, though.

I think It would be nice if GHDL resembles GCC behaviour.

1138-4EB, 2019-09-17:
@arcturus140, are you aware that using GHDL with either LLVM or GCC backends will produce those `*.o` files as a result of analyzing VHDL files? You can then use `ghdl --bind` and `ghdl --list-link` to build an Ada/C/C++ applications.

Alternatively, you can provide a C/C++ wrapper to `ghdl -e` (with any backend) to avoid manually handling all the `*.o` files.

Does this sound useful to you?

arcturus140, 2019-09-17:
> are you aware that using GHDL with either LLVM or GCC backends will produce those *.o files as a result of analyzing VHDL files?

Yes, I am using GCC backend. My build step looks something like 
`%.vhd.o: %.vhd` which I need to change into 
`%.o: %.vhd` so it does not constantly rebuild the sources

1138-4EB, 2019-09-17:
You want to explicitly set where to place and how to name each object file produced as a result of `ghdl -a`? If that is the case, I don't know if GHDL supports that. It does multiple assumptios about where internal files are located, and these are considered internal.

See #781 for a similar issue.

Nonetheless, let's wait for @tgingolds comment.

arcturus140, 2019-09-17:
Yes, I haven't found any way to modify the build command. My workaround for now is to add a `touch <OBJ>` to the build step to make the 3rd-party happy. To me, that's good enough.

arcturus140, 2019-09-17:
in [the documentation](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=--workdir#elaboration-e) we read that 

> Option -o followed by a filename can override the default executable filename.

How to use it? I am using GCC backend and `hello_world` is an entity:

    ghdl -e hello_world -o hello
    ghdl:error: too many unit names for command '-e'

1138-4EB, 2019-09-17:
`ghdl -e -o hello hello_world`? Flags before positional args, I think.

tgingold, 2019-09-17:
That's correct.  It is only possible to specify the filename of the executable.