ghdl/ghdl #86: Is it possible to create a new VHDL-2008 flavor with Synopsys extensions in ieee?

Paebbels, 2016-06-17:
I would like to compile Altera packages with VHDL-2008. It's currently not possible, because it expects Synopsys packages to be in ieee. Is it possible to create some kind of dirty VHDL flavor like for VHDL-93 for 2008?

tgingold, 2016-06-18:
On 17/06/16 18:44, Patrick Lehmann wrote:
> I would like to compile Altera packages with VHDL-2008. It's currently
> not possible, because it expects Synopsys packages to be in ieee. Is it
> possible to create some kind of dirty VHDL flavor like for VHDL-93 for 2008?

Why not. But I fear there are some conflicts between synopsys and 
vhdl-08 packages, like std_logic_textio.

Paebbels, 2016-06-19:
Are the currently available ieee library flavors (standard, synopsys, metor) hard coded in GHDL or is it enough to compile a new ieee library like it's already done for `--ieee=synopsys`?
So the `--ieee` value is interpreted as a directory name?

tgingold, 2016-06-20:
On 19/06/16 20:14, Patrick Lehmann wrote:
> Are the currently available ieee library flavors (standard, synopsys,
> metor) hard coded in GHDL or is it enough to compile a new ieee library
> like it's already done for |--ieee=synopsys|?
> So the |--ieee| value is interpreted as a directory name?

No, they are hardcoded.  But synopsys is already known.
So you just have to create a synopsys directory in v08/

Tristan.

go2sh, 2016-06-20:
I changed the Altera packages to use the ieee standard packages. It works as expected.

Paebbels, 2016-06-21:
I don't want to patch hundreds for vendor files :). It's also a license issue, to patch such files.

Paebbels, 2016-06-23:
I trying to extend `libraries/Makefile.in` for a new VHDL-2008 `synopsys` flavor.

I have two questions:

 1. I found `v08/std` two times in [line 36](https://github.com/tgingold/ghdl/blob/master/libraries/Makefile.inc#L36).
    
    **Is this correct, or should one be removed?**

 2. VHDL-2008 has no Vital2000 packages. The makefile contains a FIXME note [in line 278](https://github.com/tgingold/ghdl/blob/master/libraries/Makefile.inc#L278).
    I already added Vital2000 in the standard library compile script for Windows. According to some vendor libraries (which might be wrong about it), it's required in VHDL-2008, too.
    
    **Shall I activate it?**

closed 2016-07-03 by tgingold