ghdl/ghdl #87: support for unbuffered output

wfjm, 2016-06-18:
I'm using ghdl with VHPI to setup co-simulations. In the 'external' part
stdout is either set to autoflush (e.g. in perl), or a flush is done after
each message (e.g. in C/C++).

In my test environment the output is usually send through a pipe.
Apparently ghdl uses buffered IO when vhdl textio outputs are written.

This leads to messages from ghdl and the 'external' part to be out-of-order.

Some other simulators have an option like 'UNBUFFERED' or 'AUTOFLUSH' which
will force a flush for each line for stdout writes.

If not it be great when it is added on a future release.

tgingold, 2016-06-20:
On 18/06/16 17:51, wfjm wrote:
> I'm using ghdl with VHPI to setup co-simulations. In the 'external' part
> stdout is either set to autoflush (e.g. in perl), or a flush is done after
> each message (e.g. in C/C++).
>
> In my test [...]

If you want to write portable code, you can use the new flush procedure
from vhdl 2008.

Tristan.

wfjm, 2016-06-21:
On 06/20/2016 08:00 AM, tgingold wrote:
> On 18/06/16 17:51, wfjm wrote:
>  > Some other simulators have an option like 'UNBUFFERED' or 'AUTOFLUSH' which
>  > will force a flush for each line for stdout writes.
>
> If you want to write portable code, you can use the new flush procedure
> from vhdl 2008.

Dear Tristan,

first a clarification: what I'm addressing is output to STDOUT, done by
normal write(output, line) calls or even via assert severity info. The
goal is that all output, even from libraries, is unbuffered and thus in
proper sequence with output from other processes in the co-simulation setup.

So what is needed is a mechanism which sets a global or a file descriptor
property, like in 'autoflush STDOUT' in perl, which acts on all subsequent
writes, and not a mechanism like fflush(1) which acts only once on the
current buffer.

That's why vhdl-2008 flush is not an option.

I don't think that the issue I raise can by solved at source code level,
it should be addressed at the simulator level.

ModelSim for example has an option
    UnbufferedOutput = (0 | 1)
which controls VHDL and Verilog files open for write. Can be defined
in modelsim.ini, and becomes this way a simulator property.

NCSim has a -UNBUFFERED command line option.

In case of ghdl it be great to have something similar as command line option.

Thanks in advance,    Walter

Jonsba, 2016-06-25:
Hello Walter,

I made a patch to add support for unbuffered output. Could you try it and tell me if it works as expected before I make a pull request ?
You can enable it with ghdl -r --unbuffered
Thanks
Jonas

[unbuffered.patch.zip](https://github.com/tgingold/ghdl/files/333187/unbuffered.patch.zip)

Jonsba, 2016-06-26:
You could clone my repo (https://github.com/Jonsba/ghdl) instead of applying the patch.

wfjm, 2016-06-26:
Dear Jonas,

I'll try, but it will take some time, certainly till next weekend.
	
	With best regards, 	Walter

-- 
Dr. Walter F.J. Müller      Mail:  W.F.J.Mueller@gsi.de
CBM Technical Coordinator   Phone:  +49-6159-71-2766
GSI,  Abteilung CBM         Mobile: +49-174-328-1529
                             FAX:    +49-6159-71-3026
Office: KBW.4.013           Address: GSI, Planckstr. 1, D-64291 Darmstadt

Gesellschaft mit beschränkter Haftung; Sitz der Gesellschaft:
Darmstadt; Handelsregister: Amtsgericht Darmstadt, HRB 1528;
Geschäftsführung: Ursula Weyrich; Prof. Dr. Karlheinz Langanke, Jörg Blaurock
Vors. des Aufsichtsrats: Dr. Georg Schütte; Stellv.: Dr. Rolf Bernhardt

Jonsba, 2016-06-26:
Hello Walter,

It's fine for me.
All the changes I made are in a new branch (unbuffered) created in my repo. So you will need to clone this particular branch of my repo with the following command :

git clone -b unbuffered --single-branch https://github.com/Jonsba/ghdl.git

Greetings,
Jonas

closed 2016-06-28 by tgingold

wfjm, 2016-07-02:
On 06/28/2016 04:26 AM, tgingold wrote:
> Closed #87 <https://github.com/tgingold/ghdl/issues/87> via #100 <https://github.com/tgingold/ghdl/pull/100>.

Dear Jonas, dear Tristan,

sorry, took some time, had to exercise building ghdl from sources
a bit. Finally tested in --unbuffered after Tristan merged pull/100.

Works fine for me, the output in my co-simulation environment is
finally in order.

One little remark:
I see that --unbuffered is not described in the --help text.
I'd assume that one probably has to add in

   src/grt/grt-options.adb
   procedure Help

a line like

       P (" --unbuffered      disable buffering for write or append mode files");

but I'm not familiar enough with ghdl to firmly propose that change.

	Thanks again and with best regards,   Walter

Jonsba, 2016-07-03:
Hello Walter,

Glad to hear good news from you :). I just added the description of 
--unbuffered and made a pull request. It is available when you do ghdl 
-r <exec> --help.

Greetings,
Jonas

PR #100 (2016-06-28): Jonsba/master
This pull request add an --unbuffered option for simulation. It disables buffering for stdout, stderr and for files opened in write or append mode (TEXTIO).

It should fix #87, but I need confirmation from Walter (wfjm).
commit 2ea6a590 (2016-06-28): Jonsba/master (#100)

* Add a --unbuffered option for simulation. It disables buffering for stdout, stderr and files opened in write or append mode (TEXTIO).

Fixes #87 

* Some simplifications made on the previous commit. Documentation added for --unbuffered in Simulation_and_runtime.rst.

* Some more simplifications on unbuffered writes support for TEXTIO

* Fix a style issue failure reported by Travis-CI