ghdl/ghdl #89: Embed vendor library compile scripts into "monthly build".

Paebbels, 2016-06-21:
Everything works fine with these new vendor library compile scripts, except for my own Travis-CI flow for PoC ;(. So to complete the list of work items for GHDL, here is the missing one for the monthly builds....

I think, when the makefile(s) are updated for #83. The deployment process might already gather all needed files for the *.tgz file. So let's keep this ticket as a reminder, to check if all files are packed in the *.tgz file.

tgingold, 2016-06-22:
I have just closed #83 as scripts are now installed.

Paebbels, 2016-06-22:
I added a new comment to #83. When it is solved (or worked around), then I would like to request a new tag to test it in my Travis environment :). But so far everything looks fine.

Paebbels, 2016-06-25:
I think everything looks fine and you can tag the current version.

closed 2016-10-31 by Paebbels