ghdl/ghdl #92: Is it possible to do interactive user input simulation and display outputs interactively?

cirosantilli, 2016-06-23:
For educational purposes mostly, it would be awesome if people could code simple systems / games on VHDL.

I would love to see a software simulation of something like:

- http://www.nand2tetris.org/
- https://github.com/fabioperez/space-invaders-vhdl

for those that don't have an FPGA.

closed 2016-06-23 by cirosantilli

Johnlon, 2021-05-17:
See .. https://github.com/Johnlon/spam-1/blob/master/verilog/uart/um245r.v
I use this so my verilog H/W can interact with the outsite world

umarcor, 2021-05-17:
@Johnlon have a look at [dbhi/vboard](https://github.com/dbhi/vboard) and  [ghdl.github.io/ghdl-cosim](https://ghdl.github.io/ghdl-cosim/). On the Verilog side, I recommend https://8bitworkshop.com/v3.7.2/?platform=verilog&file=framebuffer.v.