ghdl/ghdl #97: Accessing VHDL generics of DUT via VPI (Cocotb)

mzabeltud, 2016-06-24:
Hello,

thanks for fixing my last issue, now I triggered the next one. My Cocotb testbenches access the VHDL "generic" parameters of the DUT (design under test) via VPI to parametrize the golden model. Within the Python code, it is done the same way as accessing signal values. This works nice in QuestaSim. But in GHDL, VHDL generics could not be resolved during this access.

It would be nice, if this feature is also added to GHDL. An error log can be found below. Just let me know, if a minimal example is required.

Thanks, Martin

Error log from console:
```
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on GHDL version 0.1
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
  0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /home/zabel/git/poc/lib/cocotb
  0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1466786675
  0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test sort_lru_cache_cocotb.run_test_001
  0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: run_test_001
  0.00ns INFO     ..coroutine._my_test.0x7fb3f7ec2910       decorators.py:189  in send                            Starting test: "run_test_001"
                                                                                                                            Description: Automatically generated test
                                                                                                                            
                                                                                                                            
  0.00ns ERROR    ..coroutine.run_test.0x7fb3f7c52350           result.py:51   in raise_error                     Send raised exception: sort_lru_cache contains no object named ELEMENTS
                                                                                                                              File "/home/zabel/git/poc/lib/cocotb/cocotb/decorators.py", line 114, in send
                                                                                                                                return self._coro.send(value)
                                                                                                                              File "/home/zabel/git/poc/temp/cocotb/sort_lru_cache_cocotb.py", line 175, in run_test
                                                                                                                                tb = Testbench(dut)
                                                                                                                              File "/home/zabel/git/poc/temp/cocotb/sort_lru_cache_cocotb.py", line 104, in __init__
                                                                                                                                elements = dut.ELEMENTS.value;
                                                                                                                              File "/home/zabel/git/poc/lib/cocotb/cocotb/handle.py", line 261, in __getattr__
                                                                                                                                raise AttributeError("%s contains no object named %s" % (self._name, name))
                                                                                                                            
  1.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: run_test_001 (result was TestError)
  1.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
  1.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

Here `ELEMENTS` is a VHDL generic. This is the entity declaration of the DUT:
```
entity sort_lru_cache is
	generic (
		ELEMENTS			 : positive					:= 32
	);
	port (
		Clock 	: in std_logic;
		Reset 	: in std_logic;

		Insert 	: in  std_logic;
		Free 		: in  std_logic;
		KeyIn	 	: in  std_logic_vector(log2ceilnz(ELEMENTS) - 1 downto 0);

		KeyOut 	: out std_logic_vector(log2ceilnz(ELEMENTS) - 1 downto 0)
	);
end entity;
```

tgingold, 2016-06-24:
On 24/06/16 19:04, mzabeltud wrote:
> Hello,
>
> thanks for fixing my last issue, now I triggered the next one. My Cocotb
> testbenches access the VHDL "generic" parameters of the DUT (design
> under test) via VPI to parametrize the golden [...]

Yes, a minimal example would be very nice.

Tristan.

mzabeltud, 2016-07-01:
Hello Tristan,

the minimal example can be found on the branch [vlsi-eda/generics-test](https://github.com/VLSI-EDA/cocotb/tree/vlsi-eda/generics-test) in [our fork](https://github.com/VLSI-EDA/cocotb) of Cocotb.

Just change to the "examples/generics_test/tests" sub-directory and execute `make SIM=ghdl` on the command line.

An error is raised, when Cocotb tries to access the generic `N` via VPI:

```
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on GHDL version 0.1
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
  0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /home/zabel/git/cocotb
  0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1467381325
  0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test generics_test_cocotb.run_test_001
  0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: run_test_001
  0.00ns INFO     ..coroutine._my_test.0x7f0a148c5cd0       decorators.py:189  in send                            Starting test: "run_test_001"
                                                                                                                            Description: Automatically generated test
                                                                                                                            
                                                                                                                            
  0.00ns ERROR    ..coroutine.run_test.0x7f0a14659710           result.py:51   in raise_error                     Send raised exception: generics_test contains no object named N
                                                                                                                              File "/home/zabel/git/cocotb/cocotb/decorators.py", line 114, in send
                                                                                                                                return self._coro.send(value)
                                                                                                                              File "/home/zabel/git/cocotb/examples/generics_test/tests/generics_test_cocotb.py", line 44, in run_test
                                                                                                                                N = dut.N.value # read value of generic 'N' of DUT
                                                                                                                              File "/home/zabel/git/cocotb/cocotb/handle.py", line 261, in __getattr__
                                                                                                                                raise AttributeError("%s contains no object named %s" % (self._name, name))
                                                                                                                            
  1.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: run_test_001 (result was TestError)
  1.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
  1.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

Cheers,
Martin

---

If the same test is executed with QuestaSim (`make SIM=questa`), then the test completes as expected and the value of 'N' is printed in the simulator log:

```
#   0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on ModelSim for Questa-64 version 10.4d 2015.12
#   0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
#   0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /home/zabel/git/cocotb
#   0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1467381320
#   0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test generics_test_cocotb.run_test_001
#   0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: run_test_001
#   0.00ns INFO     ..coroutine._my_test.0x7fffef8c5f10       decorators.py:189  in send                            Starting test: "run_test_001"
#                                                                                                                   Description: Automatically generated test
#                                                                                                                   
[... cut ...]
#   0.00ns INFO     cocotb.generics_test               ..ics_test_cocotb.py:45   in run_test                        The value of N is 5.
#   1.00ns INFO     cocotb.regression                         regression.py:209  in handle_result                   Test Passed: run_test_001
#   1.00ns INFO     cocotb.regression                         regression.py:170  in tear_down                       Passed 1 tests (0 skipped)
#   1.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

tgingold, 2016-07-06:
It is now possible to read generics from a cocotb test bench.

However, because the simulator interface is VPI (which is the verilog one), generics are seen like a verilog parameter, which is read as a bit vector.  So in your case, N is "00000101" instead of 5.

So currently, you will have to slightly adjust your cocotb test to convert N into an integer.  Is that possible ?

The use of VPI is a limitation for your issue.  The right way to fix it would be the use of the VHPI interface, which is not implemented as an interface for the user.  VPI and VHPI are very similar, so this is possible to do but not done!

Tristan.

mzabeltud, 2016-07-06:
Thank you very much.

> So currently, you will have to slightly adjust your cocotb test to convert N into an integer. Is that possible ?

Yes. I have updated the code in the above mentioned branch to include a type check. The binary string returned when using VPI is converted to an integer.

The minimal example now works with GHDL and Questa.

closed 2016-07-06 by mzabeltud

tgingold, 2016-07-06:
On 06/07/16 15:58, mzabeltud wrote:
> Thank you very much.
>
>     So currently, you will have to slightly adjust your cocotb test to
>     convert N into an integer. Is that possible ?
>
> Yes. I have updated the code in the above mentioned [...]

Great!  Thank you for the status update.