ghdl/ghdl #99: VHDL 2008: generic package

1138-4EB, 2016-06-27:
I am trying to analyze the following file:

```
package pkg is
  generic ( gen: natural );
  constant test: natural:=gen;
end package;

package mygpkg is new work.pkg generic map ( gen => 17 );
```
with `ghdl -a --std=08 --workdir=work pkg.vhd`, and I get the following error:

```
.../pkg.vhd: In function ‘work__mygpkg__ELAB’:
.../pkg.vhd:3:0: internal compiler error: in new_selected_element, at vhdl/ortho-lang.c:1417
   constant test: natural:=gen;
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugzilla.redhat.com/bugzilla> for instructions.
ghdl: exec error
```

However, if the last line is commented, it is analyzed with no error:

```
package pkg is
  generic ( gen: natural );
  constant test: natural:=gen;
end package;

--package mygpkg is new work.pkg generic map ( gen => 17 );
```
Are generic lists in packages supported in GHDL? I would say it is not a supported feature. However, the error is given when trying to instantiate a generic package, not when declaring it. In fact, when not correctly declared, the given output behaves as if it was a supported feature:

```
package pkg is
  generic ( gen: natural )
  constant test: natural:=gen;
end package;
```

```
pkg.vhd:3:3: missing ";" at end of generic clause
ghdl: compilation error
```

PS: I am using GHDL 0.34dev (20151126) [Dunoon edition] in 4.5.7-200.fc23.x86_64.

tgingold, 2016-06-28:
On 27/06/16 21:38, 1138-4EB wrote:
> I am trying to analyze the following file:
>
> |package pkg is generic ( gen: natural ); constant test: natural:=gen;
> end package; package mygpkg is new work.pkg generic map ( gen => 17 ); |
>
> with [...]

Yes, generic packages are supported (but not generic subprograms).

I am pretty sure you cannot instantiate a generic package before 
declaring the body.  I have to double check however.

In any case, ghdl shouldn't crash.

Tristan.

tgingold, 2016-06-28:
Oops, I have just realised that the package doesn't require a body.  It works if you add a dummy body.

tmeissner, 2016-06-28:
> Yes, generic packages are supported (but not generic subprograms).

Nice. But where do I have to instantiate an package with generic list when using it in an architecture? I tried the following, which works in Modelsim, but it don't work with GHDL (all files are compile into work library):

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

package test_p is

  generic (
    G_TEST  : natural := 0
  );

  procedure test;

end package test_p;

package body test_p is

  procedure test is
  begin
    report "G_TEST: " & integer'image(G_TEST);
  end procedure test;    

end package body;
```

```
entity test_e is
end entity test_e;

architecture sim of test_e is

  package Test is new work.TestPkg generic map (G_TEST => 17);
  use Test.all;

begin

  test;

end architecture sim;
```

That gives me following error:

```
cd work; ghdl -a --std=08 -fpsl --workdir=work ../test_p.vhd ../test_e.vhd
../test_p:7:3: 'begin' is expected instead of 'package'
ghdl: compilation error
```

I've also encountered a lot of problems using generic packages with Modelsim, because the syntax is quiet complex and not very understandable documented in the "Designers Guide to VHDL 2008" which is the only book I know about the 08 features :/ But the upper solution works with Modelsim.

closed 2016-06-28 by tgingold

tgingold, 2016-06-30:
On 28/06/16 10:28, T. Meissner wrote:
>     Yes, generic packages are supported (but not generic subprograms).
>
> Nice. But where do I have to instantiate an package with generic list
> when using it in an architecture?

That's not (yet) supported by ghdl: only instantiation at design-unit 
level is support :-(

> I tried the following, which works in
> Modelsim, but it don't work with GHDL (all files are compile into work
> library):

Indeed, this is part of the language and looks correct.

> I've also encountered a lot of problems using generic packages with
> Modelsim, because the syntax is quiet complex and not very
> understandable documented in the "Designers Guide to VHDL 2008" which is
> the only book I know about the 08 features :/ But the upper solution
> works with Modelsim.

I am very reserved about generic packages in vhdl2008.  They are not
well specified and I think there are some mistakes (like generic type).
That's a pity as genericity is a well known concept.

Tristan.

1138-4EB, 2016-06-30:
@tgingold thank you very much for responding so fast. You committed a fix in less than 24h! Wow!

Just for completeness, in 

> Oops, I have just realised that the package doesn't require a body. It works if you add a dummy body.

you meant that the standard expects me to write a body, so I should always do it? Then the fix you commited would prevent GHDL from crashing, but I should get used to always declaring a body. Or, it is just a workaround until the next version is released (which will include your fix)?

@tmeissner I'm not sure of this being what you want, but it compiles in my setup:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_p is
  generic ( G_TEST  : natural := 0 );
  procedure test;
end package test_p;

package body test_p is
  procedure test is begin
    report "G_TEST: " & integer'image(G_TEST);
  end procedure test;    
end package body;

entity test_e is
end entity test_e;

package MyTestPkg is new work.test_p generic map (G_TEST => 17);

architecture sim of test_e is
  use work.MyTestPkg.all;
begin
  test;
end architecture sim;
```
I took *VHDL-2008 Just the new stuff, Peter J.Ashenden and Jim Lewis*, page 8, as a reference.

tgingold, 2016-06-30:
On 30/06/16 05:36, 1138-4EB wrote:
> @tgingold <https://github.com/tgingold> thank you very much for
> responding so fast. You committed a fix in less than 24h! Wow!

You're welcome. Thank you for using ghdl and reporting issues!

> Just for completeness, in
>
>     Oops, I have just realised that the package doesn't require a body.
>     It works if you add a dummy body.
>
> you meant that the standard expects me to write a body, so I [...]

It is just a workaround until the next version.

Tristan.

commit 9d38f14e (2016-06-28): Testcase for issue99.

Close #99