ghdl/ghdl #101: Co-Simulation with CLaSH

jgjverheij, 2016-06-28:
I am trying to create co-simulation between CLaSH (https://github.com/clash-lang/clash-compiler) and other simulators. With GHDL I want to use the VPI standard, but some (needed) things are not supported (yet)...

I am not familiar with the GHDL source-code and I want to leave this as suggestions for future implementations. 

1. I register the callback "cbStartOfSimulation" in which I want to iterate over the "vpiPorts". This is needed to check if the number and sizes of the input- and output-ports correspond to the "signals" within CLaSH. 
For this I want to use "vpiPort" instead of "vpiNet", to also use "vpiDirection". With the "vpiDirection" i can distinguish between input- and output-signals. The properties "vpiPort" and "vpiDirection" are unknown in GHDL. 

2. The width of the "signals" within CLaSH are user-defined and not restricted to 32 bits. To support more than 32 bits, I use the "vpiVectorVal" property in the "vpi_get_value" and "vpi_put_value" functions. The property "vpiVectorVal" is currently not supported in GHDL. 
A workaround is to use the "vpiBinStrVal" property, but then I have to work with string values...

tgingold, 2016-07-02:
I confirm that vpiPorts, vpiDirection and vpiVectorVal are not yet supported.

1138-4EB, 2019-09-04:
Was this issue fixed in #753? Or is `vpiVectorVal` unsupported yet?

tgingold, 2019-09-04:
I think it is not supported yet.