ghdl/ghdl #102: Building GHDL (mcode) with MinGW32 on Windows 7 64 Bit

Paebbels, 2016-06-28:
Hello,

at first, here is my setup of my machine:
- PC Intel Core i5 -> AMD64
- Windows 7 64 Bit
- MSys2 wich includes MinGW32 and MinGW64
  - MinGW32 (i686-w64-mingw32) with
    - gcc 5.4.0
    - gcc-ada 5.4.0
    - llvm 3.5.2
  - MinGW64 (x86_64-w64-mingw64) with
    - gcc 5.4.0
    - gcc-ada 5.4.0
    - llvm 3.5.2
- GNAT GPL 2016 (x86 - 32 bit) from AdaCore

----------------
I cloned a fresh GHDL and used configure as usual:

    ./configure --prefix=/c/Tools/GHDL/0.34dev-mcode-mingw32

Then I started `make`.

**Issue 1:**

As discussed in an email, `src\grt\Makefile.inc` needs to be adjusted according to the already applied changes in `compile.ps1` for Windows. In case of MinGW32, the object files `jumps.o` and `math.o` are replaced by `win32.o`:

```diff
diff --git a/src/grt/Makefile.inc b/src/grt/Makefile.inc
index f8a4d32..2ce976b 100644
--- a/src/grt/Makefile.inc
+++ b/src/grt/Makefile.inc
@@ -47,8 +47,8 @@ endif
 GRT_ELF_OPTS:=-Wl,--version-script=@/grt.ver -Wl,--export-dynamic

 # Set target files.
-ifeq ($(filter-out mingw32,$(osys)),)
-  GRT_TARGET_OBJS=jumps.o math.o clock.o
+ifeq ($(filter-out mingw32 mingw64,$(osys)),)
+  GRT_TARGET_OBJS=win32.o clock.o
 else
   GRT_TARGET_OBJS=jumps.o times.o
   ifeq ($(filter-out linux,$(osys)),)
```

**Issue 2:**

LD complains, that symbol `StackWalk64` is unknown in `win32.c:99`:
```bash
$ make
gnatmake -o ghdl_mcode -aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/mcode -gnaty3befhkmr -gnatwae -aO. -gnatf -gnat05 -g -gnata -gnatw.A ghdl_jit.adb -bargs -E -largs memsegs_c.o chkstk.o  win32.o clock.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o
gnatbind -aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/mcode -aO. -E -x ghdl_jit.ali
gnatlink ghdl_jit.ali -o ghdl_mcode.exe -g memsegs_c.o chkstk.o win32.o clock.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o
win32.o: In function `get_bt_from_context':
D:\git\GitHub\ghdl_mingw32/src/grt/config/win32.c:99: undefined reference to `_imp__StackWalk64@36'
collect2.exe: error: ld returned 1 exit status
gnatlink: error when calling C:\msys64\mingw32\bin\gcc.exe
gnatmake: *** link failed.
make: *** [Makefile:103: ghdl_mcode] Fehler 4
```

As far as I can see, this function is defined in `/mingw64/x86_64-w64-mingw32/include/psdk_inc/_dbg_common.h`.

As a temporary work-around, I *disabled* `get_bt_from_context` by an immediate return. The following diff also fixes a typo (`ctx` -> `ctxt` for x86_64):

```diff
diff --git a/src/grt/config/win32.c b/src/grt/config/win32.c
index 79935e6..11d17de 100644
--- a/src/grt/config/win32.c
+++ b/src/grt/config/win32.c
@@ -74,16 +74,20 @@ get_bt_from_context (struct backtrace_addrs *bt, CONTEXT *ctxt, int skip)
   frame.AddrStack.Offset = ctxt->Esp;
   frame.AddrStack.Mode = AddrModeFlat;

+#  warning "platform not supported"
+  return;
 #elif defined (__x86_64__)
   mach = IMAGE_FILE_MACHINE_AMD64;

   frame.AddrPC.Offset = ctxt->Rip;
   frame.AddrPC.Mode = AddrModeFlat;
-  frame.AddrFrame.Offset = ctx->Rsp;
+  frame.AddrFrame.Offset = ctxt->Rsp;
   frame.AddrFrame.Mode = AddrModeFlat;
-  frame.AddrStack.Offset = ctx->Rsp;
+  frame.AddrStack.Offset = ctxt->Rsp;
   frame.AddrStack.Mode = AddrModeFlat;

+#  warning "platform not supported"
+  return;
 #else
 #  warning "platform not supported"
   return;
```

After fixing these little things, everything compiled fine and GHDL was used to analyze the `std` and `ieee` packages (executed from within the MinGW32 shell). I continued with `make install` and GHDL was copied into the intended directory.

**Issue 3:**

I also tested GHDL from PowerShell. When I start `ghdl.exe` without any parameter I get this error message:

```PowerShell
D:\git\PoC [paebbels/master = +0 ~7 -0 | +2 ~5 -0 !]> C:\Tools\GHDL\0.34dev-mingw32-mcode\bin\ghdl.exe
C:\Tools\GHDL\0.34dev-mingw32-mcode\bin\ghdl.exe: missing command, try C:\Tools\GHDL\0.34dev-mingw32-mcode\bin\ghdl.exe --help

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
```

There seems to be a problem, when GHDL is going to gracefully terminate. When I use `ghdl.exe -v`, then there is no shutdown problem. This error message is also shown if a command line parameter check fails e.g. `unknown option '-P' for command '-a'`, which requires a shutdown.

---------------
**Overall success message:** This version executed all testbenches in PoC without any error!

tgingold, 2016-06-28:
Please, create a pull request for all your fixes.

Thanks.

tgingold, 2016-06-29:
For issue 2: ghdl should be linked with dbghelp

Paebbels, 2016-06-29:
**Appendix for issue 3:**
I managed to extend my MinGW installations with gdb. Here is the gdb output for calling GHDL without parameters:

```
(gdb) catch exception
Your Ada runtime appears to be missing some debugging information.
Cannot insert Ada exception catchpoint in this configuration.
(gdb) r
Starting program: D:\git\GitHub\ghdl_mingw32\ghdl_mcode.exe
[New Thread 10556.0x1568]
D:\git\GitHub\ghdl_mingw32\ghdl_mcode.exe: missing command, try D:\git\GitHub\ghdl_mingw32\ghdl_mcode.exe --help

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
[Inferior 1 (process 10556) exited with code 03]
```

I assume I have no influence on the Ada runtime, to insert debugging information...

Paebbels, 2016-06-30:
**Solution to issue 2:**

I populated `GRT_EXTRA_LIB` with `-ldbghelp` in case of mingw32/mingw64. This also reduces some cases when issue 3 is triggered, but not all of them.

So here is a short diff, until I finish the first comprehensive PR on all proposed diffs:

```diff
diff --git a/src/grt/Makefile.inc b/src/grt/Makefile.inc
index 2ce976b..7ac0c96 100644
--- a/src/grt/Makefile.inc
+++ b/src/grt/Makefile.inc
@@ -49,6 +49,7 @@ GRT_ELF_OPTS:=-Wl,--version-script=@/grt.ver -Wl,--export-dynamic
 # Set target files.
 ifeq ($(filter-out mingw32 mingw64,$(osys)),)
   GRT_TARGET_OBJS=win32.o clock.o
+  GRT_EXTRA_LIB=-ldbghelp
 else
   GRT_TARGET_OBJS=jumps.o times.o
   ifeq ($(filter-out linux,$(osys)),)
```

closed 2016-07-03 by tgingold