ghdl/ghdl #104: Building GHDL (llvm) with MinGW32 on Windows 7 64 Bit

Paebbels, 2016-06-28:
Hello,

at first, here is my setup of my machine:
- PC Intel Core i5 -> AMD64
- Windows 7 64 Bit
- MSys2 wich includes MinGW32 and MinGW64
  - MinGW32 (i686-w64-mingw32) with
    - gcc 5.4.0
    - gcc-ada 5.4.0
    - llvm 3.5.2
  - MinGW64 (x86_64-w64-mingw64) with
    - gcc 5.4.0
    - gcc-ada 5.4.0
    - llvm 3.5.2
- GNAT GPL 2016 (x86 - 32 bit) from AdaCore

----------------
Based on the changes of issue #102, I tested GHDL with llvm backend. So I run these lines:

    make distclean
    ./configure --prefix=/c/Tools/GHDL/0.34dev-mingw32-llvm --with-llvm-config=llvm-config
    make

**Issue 1:**

`ghdl_llvm.exe` can not find `ghdl1_llvm`. The problem is the missing file extension for executables on Windows. As a quick fix, I edited the created `Makefile` and changed line 214:

```Makefile
# line 213-215
libs.vhdl.llvm: ghdl_llvm ghdl1-llvm
	$(MAKE) GHDL=ghdl_llvm ANALYZE_OPTS="--GHDL1=ghdl1-llvm.exe" vhdl.libs.all
	$(MAKE) GHDL1=./ghdl1-llvm libs.vhdl.standard
```

Then `make` continued to build all `std` and `ieee` packages. I'm not sure how to modify `Makefile.in` so configure will create an appropriate `Makefile`. As GHDL builds all `ieee` packages without error, I expect no problems with this version.

I think the solution should be to compile the correct executable name into `ghdl.exe`. Am I right?

**Issue 2:**

I can't execute `ghdl.exe` in PowerShell, because it can't find `ghdl1-llvm`, too. Does GHDL recognize `$GHDL` or `$GHDL1`, so I could override the default executable name for a test?

Regards
    Patrick

tgingold, 2016-06-28:
Just modify configure to append the .exe suffix on windows in the generated default_pathes.ads file.

Paebbels, 2016-06-29:
**Solution for issue 1:**
As you pointed out, I modified configure to infer the file extensions from `$build`. `$LIBEXT` is a typical name to define the library extension (*.so / *.dll), but I don't remember any well known name for executables .... I chose `$EXEEXT`.

So *.exe is appended to all executables, if the platform matches `mingw*`. I hope if found all executables in the makefiles.

Paebbels, 2016-06-30:
**Addendum to issue 3:**

I think I found the problem for the false output directory. When I run GHDL (llvm) with `-v`
I can see, that `ghdl.exe` (`ghdl_llvm.exe`) passes a `-o` parameter to `ghdl1-llvm.exe`, which is
equal to the input file, except for the changed file extension.

**Example 1** - Compile `std_logic_1164.v87` for VHDL-87:

```
Paebbels@Paebbels-NB1 MINGW32 /d/git/ghdl/lib/ghdl/v87/ieee
$ ../../../../ghdl_llvm -a -v --GHDL1=ghdl1-llvm.exe --std=87 -P../std --work=ieee ../../src/ieee/std_logic_1164.v87
D:\git\ghdl\ghdl1-llvm.exe --std=87 --work=ieee -P../std\ -PD:\git\ghdl\lib/ghdl\v87\std\ -PD:\git\ghdl\lib/ghdl\v87\ieee\ -c -o ../../src/ieee/std_logic_1164.o ../../src/ieee/std_logic_1164.v87
```

<strike>When I add a `-o` parameter, then it's getting strange. `ghdl1-llvm.exe` seems to be called 3 times with:</strike>

 1. <strike>`-o ../../src/ieee/std_logic_1164.o` -> as in example 1</strike>
 2. <strike>`-o -o.o -o` -> `-o` is added 3 times, whereby the second one seems to contain the file extension</strike>
 3. <strike>`-o std_logic_1164.o` -> this is the intended call, but the input file was also changed from `*.vhdl` to `*.o`, causing a file not found error.</strike>

Argh, I get it! `-o` is considered a valid file name, so there are 3 input files...
This brings me to a question (without any judgement): Should GHDL continue if an error was found in one of the input files? In most cases further files depend on each other (compile order), so they are going to fail, too. Other tools have a *halt on error* or *resume on error* setting, parameter or Tcl command.

**Example 2** - Compile `std_logic_1164.v87` for VHDL-87 with explicit output name:

```
Paebbels@Paebbels-NB1 MINGW32 /d/git/ghdl/lib/ghdl/v87/ieee
$ ../../../../ghdl_llvm -a -v --GHDL1=ghdl1-llvm.exe --std=87 -P../std --work=ieee ../../src/ieee/std_logic_1164.v87 -o std_logic_1164.o
D:\git\ghdl\ghdl1-llvm.exe --std=87 --work=ieee -P../std\ -PD:\git\ghdl\lib/ghdl\v87\std\ -PD:\git\ghdl\lib/ghdl\v87\ieee\ -c -o ../../src/ieee/std_logic_1164.o ../../src/ieee/std_logic_1164.v87
D:\git\ghdl\ghdl1-llvm.exe --std=87 --work=ieee -P../std\ -PD:\git\ghdl\lib/ghdl\v87\std\ -PD:\git\ghdl\lib/ghdl\v87\ieee\ -c -o -o.o -o
error: missing filename to '-o'
D:\git\ghdl\ghdl1-llvm.exe --std=87 --work=ieee -P../std\ -PD:\git\ghdl\lib/ghdl\v87\std\ -PD:\git\ghdl\lib/ghdl\v87\ieee\ -c -o std_logic_1164.o std_logic_1164.o
D:\git\ghdl\ghdl1-llvm.exe: cannot open std_logic_1164.o

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
D:\git\ghdl\ghdl_llvm.exe: exec error
```

**Example 3** - New example, with `-o` at the right position:

```
Paebbels@Paebbels-NB1 MINGW32 /d/git/ghdl/lib/ghdl/v87/ieee
$ ../../../../ghdl_llvm -a -v --GHDL1=ghdl1-llvm.exe --std=87 -P../std --work=ieee -o std_logic_1164.o ../../src/ieee/std_logic_1164.v87
D:\git\ghdl\ghdl1-llvm.exe --std=87 --work=ieee -P../std\ -PD:\git\ghdl\lib/ghdl\v87\std\ -PD:\git\ghdl\lib/ghdl\v87\ieee\ -c -o ../../src/ieee/std_logic_1164.o ../../src/ieee/std_logic_1164.v87
```

So, here we go. Only one call to `ghdl1-llvm.exe`, but the output parameter was not passed to `ghdl1-llvm.exe`. `ghdl.exe` sends the default value as in example 1.

I looked into `ghdldrv.adb` and found [line 160](https://github.com/tgingold/ghdl/blob/master/src/ghdldrv/ghdldrv.adb#L160)...

Maybe, one should first think about the intended behavior of `-o` and its default value. In the current implementation (compiled with MinGW), GHDL always places the output file into the folder of the source file. This is contrary to other compilers, which consider the current working directory as an output folder and `-o` usually names an output filename and in some cases a relative
path like `-o obj/file1.o`. GCC sets `-o` by default to `a.out` or `a.exe`.

**Comparison to GHDL build on Debian with llvm backend:**

I checked it also on my Debian VM for GHDL (llvm). `ghdl` passes only the filename with changed extension to `ghdl1-llvm`:

```
paebbels@debian8:~/git/ghdl/build/llvm/lib/ghdl/v87/ieee$ ../../../../ghdl_llvm -a -v --GHDL1=ghdl1-llvm --std=87 -P../std --work=ieee ../../src/ieee/std_logic_1164.v87
/home/paebbels/git/ghdl/build/llvm/ghdl1-llvm --std=87 --work=ieee -P../std/ -P/home/paebbels/git/ghdl/build/llvm/lib/ghdl/v87/std/ -P/home/paebbels/git/ghdl/build/llvm/lib/ghdl/v87/ieee/ -c -o std_logic_1164.o ../../src/ieee/std_logic_1164.v87
```

Currently I don't see how the Ada code is effected by MinGW's gcc-ada compiler to produce different behavior for the `-o` option.

**P.S.**
I haven't investigated how `-o` is effected by `--workdir=`.

tgingold, 2016-07-01:
For -a, all arguments after the first filename are considered as a filename. So:

'ghdl -a -o x.o x.vhdl' is ok.
'ghdl -a x.vhdl -o x.o' is not ok (there are 3 files: x.vhdl, -o, x.o)

Paebbels, 2016-07-01:
All sub-issues are solved with pull request #109.

closed 2016-07-01 by Paebbels