ghdl/ghdl #105: Building GHDL (mcode and llvm) with MinGW64 on Windows 7 64 Bit

Paebbels, 2016-06-28:
Hello,

based on the changes of issue #102, I tested GHDL with mcode backend and MinGW64. So I run these lines:

    cp ./ghdl_mingw32 ghdl_mingw64
    cd ghdl_mingw64
    make distclean
    git reset HEAD --hard

Then I applied all changes from #102:

```diff
diff --git a/src/grt/Makefile.inc b/src/grt/Makefile.inc
index f8a4d32..95cedff 100644
--- a/src/grt/Makefile.inc
+++ b/src/grt/Makefile.inc
@@ -47,9 +47,9 @@ endif
 GRT_ELF_OPTS:=-Wl,--version-script=@/grt.ver -Wl,--export-dynamic

 # Set target files.
-ifeq ($(filter-out mingw32,$(osys)),)
-  GRT_TARGET_OBJS=jumps.o math.o clock.o
-else
+ifeq ($(filter-out mingw32 mingw64,$(osys)),)
+  GRT_TARGET_OBJS=win32.o clock.o
+else^M
   GRT_TARGET_OBJS=jumps.o times.o
   ifeq ($(filter-out linux,$(osys)),)
     GRT_EXTRA_LIB=-ldl -lm $(GRT_ELF_OPTS)
diff --git a/src/grt/config/win32.c b/src/grt/config/win32.c
index 79935e6..11d17de 100644
--- a/src/grt/config/win32.c
+++ b/src/grt/config/win32.c
@@ -74,16 +74,20 @@ get_bt_from_context (struct backtrace_addrs *bt, CONTEXT *ctxt, int skip)
   frame.AddrStack.Offset = ctxt->Esp;
   frame.AddrStack.Mode = AddrModeFlat;

+#  warning "platform not supported"
+  return;
 #elif defined (__x86_64__)
   mach = IMAGE_FILE_MACHINE_AMD64;

   frame.AddrPC.Offset = ctxt->Rip;
   frame.AddrPC.Mode = AddrModeFlat;
-  frame.AddrFrame.Offset = ctx->Rsp;
+  frame.AddrFrame.Offset = ctxt->Rsp;
   frame.AddrFrame.Mode = AddrModeFlat;
-  frame.AddrStack.Offset = ctx->Rsp;
+  frame.AddrStack.Offset = ctxt->Rsp;
   frame.AddrStack.Mode = AddrModeFlat;

+#  warning "platform not supported"
+  return;
 #else
 #  warning "platform not supported"
   return;
```

And started `configure` and `make` again:

    ./configure --prefix=/c/Tools/GHDL/0.34dev-mingw64-mcode
    make

**Issue 1:**

Due to the architecture change from `i686` to `x86_64`, 3 'opcode' are unknown. Here are the reported error messages:

```Bash
gnatbind -aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/mcode -aO. -E -x ghdl_jit.ali
gnatlink ghdl_jit.ali -o ghdl_mcode.exe -g memsegs_c.o chkstk.o win32.o clock.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__moddi3[.refptr.__moddi3]+0x0): undefined reference to `__moddi3'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__divdi3[.refptr.__divdi3]+0x0): undefined reference to `__divdi3'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__muldi3[.refptr.__muldi3]+0x0): undefined reference to `__muldi3'
collect2.exe: error: ld returned 1 exit status
gnatlink: error when calling C:\msys64\mingw64\bin\gcc.exe
gnatmake: *** link failed.
make: *** [Makefile:103: ghdl_mcode] Fehler 4
```

**Issue 2:**

Last but not least, I tested GHDL (llvm) + MinGW64, based on the changes from above. As expected, `ghdl.exe` can't find `ghdl1-llvm`, because of the missing extension.

tgingold, 2016-06-28:
For issue1, Flags.M64 should be false if you compile on x86-64.  That would fix the issue.

Paebbels, 2016-06-29:
I'll assemble a PR when the changes work.

Solution for issue 1:
I duplicated the file `ortho_code-x86-flags_windows.ads` to `ortho_code-x86-flags_windows64.ads`. Then I edited configure, so it selects the `***64` file on MinGW64, like it already selects `linux64` or `macos64` on other platforms.

So `Flags.M64` is set to `false`.

Paebbels, 2016-06-30:
Everything is solved by pull request #108.

closed 2016-06-30 by Paebbels