ghdl/ghdl #106: VHDL 2008: elsif/else and case clauses in generate statements

1138-4EB, 2016-06-30:
Are *elsif*/*else* and *case* clauses supported for generate statements?

I have tried both, and I get errors:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
end entity ent;

architecture arch of ent is
  signal test: natural;
begin
  LL: case test generate
    when =>
  end generate;
end architecture arch;
```
results in:
```
....vhdl:11:7: unexpected token 'case' in a concurrent statement list
ghdl: compilation error
```
and
```
architecture arch of ent is
  signal test: natural;
begin
  LL: if test=10 generate
  elsif test=5 generate
  end generate;
end architecture arch;
```
shows
```
....vhdl:12:3: unexpected token 'elsif' in a primary
....vhdl:12:9: 'generate' is expected instead of 'test'
ghdl: compilation error
```

I actually don't demand it to be supported. It's just to get to know which of the features I expect to use, are actually supported in GHDL.

tgingold, 2016-06-30:
On 30/06/16 05:47, 1138-4EB wrote:
> Are /elsif///else/ and /case/ clauses supported for generate statements?

In fact, only /else/ clauses are supported now.

> I have tried both, and I get errors:
>
> |library ieee; use ieee.std_logic_1164.all; use ieee.numeric_std.all;
> entity ent is end entity ent; architecture arch of ent is signal test:
> natural; begin LL: [...]

Yes, case generate statements are not (yet) supported.

> and
>
> |architecture arch of ent is signal test: natural; begin LL: if test=10
> generate elsif test=5 generate end generate; end architecture arch; |
>
> shows
>
> |....vhdl:12:3: unexpected token 'elsif' in a primary ....vhdl:12:9:
> 'generate' is expected instead of 'test' ghdl: compilation error |

The elsif is almost supported and will be soon. I keep this issue to
implement them.

> I actually don't demand it to be supported. It's just to get to know
> which of the features I expect to use, are actually supported in GHDL.

Understood.
Do you need case generate statements ?

Tristan.

1138-4EB, 2016-07-01:
They certainly help keeping the code readable. Moreover, they help avoiding human coding errors, since all the cases can be covered. I'd certainly love to have it (if you can add support for it without much effort compared to `elsif`). However, it's not a must have, since the same functionality can be achieved either with `if` statements or with `if` + `elsif` statements. It's not a totally new feature, but an enhancement.

Thank you again. You're so fast :)

tgingold, 2016-07-01:
On 01/07/16 08:40, 1138-4EB wrote:
> They certainly help keeping the code readable. Moreover, they help
> avoiding human coding errors, since all the cases can be covered. I'd
> certainly love to have it (if you can add support for it [...]

I keep that in my mind.

Tristan.

tgingold, 2016-07-09:
case generate statements are now supported.

Tristan.

closed 2016-07-09 by tgingold