ghdl/ghdl #107: Assertion failed when instantiate generic package

tmeissner, 2016-06-30:
I have a strange behavior when instantiating a generic packe in my design. It works when I try the example which @1138-4EB mentions in this issue https://github.com/tgingold/ghdl/issues/99#issuecomment-229551434. Imitating this in my bigger design, it fails. The package is almost the same:

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

package TestPkg is

  generic (
    G_TEST  : positive := 8
  ); 

end package TestPkg;

package body TestPkg is

  procedure TestReport is
  begin
    report "G_TEST :" & to_string(G_TEST);
  end procedure;

end package body;
```

 but I instantiate it in an entity of my own like this:

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

library std;
  use std.env.all;

use work.ApbPkg.all;

library osvvm;
  use osvvm.TbUtilPkg.all;
  use osvvm.AlertLogPkg.all;

entity ApbMasterBfmE is
  generic (
    G_ADDR_WIDTH  : positive := 8;  --* address bus width
    G_DATA_WIDTH  : positive := 8;  --* data bus width
    G_SLAVE_COUNT : positive := 1
  );
  port (
    --+ test if
    Test_io       : inout t_apb_test := C_TEST_INIT;
    --+ apb system if
    PRreset_n_i   : in std_logic;
    PClk_i        : in std_logic;
    --+ apb master if
    ApbMaster_o   : out t_apb_master;
     --+ apb slave if
    ApbMaster_i   : in  t_apb_slave
  );
end entity ApbMasterBfmE;

package MyTestPkg is new work.TestPkg generic map (G_TEST => 17);

architecture sim of ApbMasterBfmE is

  use work.MyTestPkg.all;

begin

  TestReport;

  ...

end architecture sim;
```

When trying to analyzing this, I get an Assertion failure with the current HEAD of GHDL (ecf964d):

```
cd work; ghdl -a --std=08 -fpsl --workdir=work -Posvvm ../ApbPkg.vhd ../TestPkg.vhd ../ApbMasterBfmE.vhd ../ApbSlaveBfmE.vhd
../TestPkg.vhd:18:14:warning: package "testpkg" does not require a body
../TestPkg.vhd:21:13:warning: procedure "testreport" is never referenced
../TestPkg.vhd:18:14:warning: package "testpkg" does not require a body
../TestPkg.vhd:21:13:warning: procedure "testreport" is never referenced
Assertion failed: (isOpaque() && "Struct body already set!"), function setBody, file Type.cpp, line 421.

Execution terminated by unhanded exception
Exception name: PROGRAM_ERROR
Message: unhandled signal
Load address: 0x107be4000
Call stack traceback locations:
0x108557e27 0x7fff948c6528
ghdl: compilation error
```

The strange thing is: Before I done the latest git pull, I had GHDL with commit id dc8680d. With that version I got another error when allaying the files:

```
../TestPkg.vhd:18:14:warning: package "testpkg" does not require a body
../TestPkg.vhd:21:13:warning: procedure "testreport" is never referenced
../TestPkg.vhd:18:14:warning: package "testpkg" does not require a body
../TestPkg.vhd:21:13:warning: procedure "testreport" is never referenced
../ApbMasterBfmE.vhd:48:3: no declaration for "test report
ghdl: compilation error
make: *** [compile] Error 1
```

So, it didn't either work with the older version (what I don't really understand, because apart from the bigger design, the package and the instantiation is the same). But there was no Assertion error as I get in the latest version.

I'm using GHDL with OS X 10.11:
```
name -a
Darwin TMs-Air.fritz.box 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64
```

PS: Maybe we should create a list of VHDL 08 features, which are supported by GHDL? That would be helpful for users as they don't have to try out, which one they can use. On the other hand, trying out triggers bugs sometimes, so it isn't a really bad thing I assume ;)

tgingold, 2016-07-01:
I can reproduce the crash and I am investigating.

tgingold, 2016-07-01:
Note that your issue is really a corner case: the procedure in the package body is completely useless.  Did you forget to declare it in the package specification ?

tmeissner, 2016-07-01:
Oh, wait, yes, I forgot the declaration ;) I have to correct the example and try again later.

1138-4EB, 2016-07-01:
I had to comment the usage of `work.ApbPkg.all`, and the three statements where the types which are expected to be declared in it are used. I also had to add an old `TbUtilPkg.vhd`, because it is not part of the official OSVVM release since 2013.5. Once done that, the following code compiles with GHDL 0.33 in Windows 10.
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

package TestPkg is
  generic ( G_TEST  : positive := 8 );
  procedure TestReport;
end package TestPkg;

package body TestPkg is
  procedure TestReport is
  begin
    report "G_TEST :" & to_string(G_TEST);
  end procedure;
end package body;

---

library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

library std;
  use std.env.all;

--use work.ApbPkg.all;

library osvvm;
  use osvvm.TbUtilPkg.all;
  use osvvm.AlertLogPkg.all;

entity ApbMasterBfmE is
  generic (
    G_ADDR_WIDTH  : positive := 8;  --* address bus width
    G_DATA_WIDTH  : positive := 8;  --* data bus width
    G_SLAVE_COUNT : positive := 1
  );
  port (
    --+ test if
    --Test_io       : inout t_apb_test := C_TEST_INIT;
    --+ apb system if
    PRreset_n_i   : in std_logic;
    PClk_i        : in std_logic
    --+ apb master if
    --ApbMaster_o   : out t_apb_master;
     --+ apb slave if
    --ApbMaster_i   : in  t_apb_slave
  );
end entity ApbMasterBfmE;

package MyTestPkg is new work.TestPkg generic map (G_TEST => 17);

architecture sim of ApbMasterBfmE is
  use work.MyTestPkg.all;
begin
  TestReport;
end sim;
```

```
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/NamePkg.vhd
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/OsvvmGlobalPkg.vhd
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/TranscriptPkg.vhd
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/TextUtilPkg.vhd
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/TbUtilPkg.vhd
$GHDL -a --std=08 --work=osvvm --workdir=osvvm OSVVM-master/AlertLogPkg.vhd

$GHDL -a --std=08 -fpsl --workdir=work -Posvvm sim.vhd
$GHDL -e --std=08 --workdir=work -Posvvm ApbMasterBfmE
```

tgingold, 2016-07-01:
So, can I close this issue ?

tmeissner, 2016-07-01:
After I added the declaration of TestReport procedure, the example works, as @1138-4EB wrote. So I think this issue can be closed :)

tgingold, 2016-07-02:
Ok, closing.

closed 2016-07-02 by tgingold