ghdl/ghdl #111: GHDL: a complete VHDL parser

1138-4EB, 2016-07-02:
I've been thinking long about having some multiplatform (windows and gnu/linux), free (libre), lightweight and standalone tool to analize large bodies of VHDL 2008 code at block/RTL level. That tool would provide a hirarchy browser of the compiled/synthetized design, so that all the parameterized types and components are resolved. Ideally, the info in the hierarchy would be enough to build a command line tool to draw diagrams with different levels of abstraction (entities, processes and concurrent black boxes) for documentation purposes. As a dream, a GUI could be built with linked references between diagrams and sources.

Among those features, hierarchy browsers are already available in all the big (and heavy) EDA tools and simulators; diagrams and linked references are also supported by most of the big editors. However, most of those tools require several GB to be installed, since they provide lower levels of abstraction and many other utilities. On top of that, the heavy GUIs must be launched even to just analize the hierarchy.

The closest alternative I am aware of is [RTLVision](http://www.concept.de/nlview.html#FAMILY). It [seems](http://www.concept.de/flash/RTL_basic.swf) to be very complete technically, it is supported in multiple platforms (windows/linux/solaris), and accepts VHDL, Verilog and System Verilog. Moreover, the [engine](http://www.concept.de/nlview.html#FAMILY) can be used in many GUI environments. Indeed, I believe it is used by some of the big players in the industry. It is not specified which the supported versions of the standard are, but I believe it should be quite complete for 2008, since they [use](http://www.concept.de/nlview.html#FAMILY) [Verific](http://www.verific.com/)'s parser. However, I don't know how fast it is, given that it accepts multiple languages, and that it includes a waveform viewer to allow signal tracing. In any case, it is neither libre nor free or open source.

There are some projects from academia, such as [VHDLVisualizer](http://www2.fiit.stuba.sk/~macko/VHDLVisualizer/) and [HDLVisualizer](http://labss2.fiit.stuba.sk/TeamProject/2011/team05pkss/docs/); and even web solutions: [vhdltolatex.mamikon.net](http://vhdltolatex.mamikon.net/). But all of them lack proper language support. Is worth referring to [VisualHDL](http://visualhdl.sysprogs.org/) even though its input is THDL++ and outputs VHDL.

A a result, I realized that the starting point should be a reliable VHDL parser. So I searched for free alternatives:

  - Many grammars and parsers for VHDL 1993 (or version not specified) [[tams.informatik.uni-hamburg.de/vhdl](https://tams.informatik.uni-hamburg.de/vhdl/index.php?content=07-tools#grammar)]
  - Not complete VHDL 2008 grammar for Coco/R [[github.com/wltr/coco-vhdl-2008](https://github.com/wltr/coco-vhdl-2008)]
  - Handmade prototype of parser based on regexps [[bitbucket.org/AngelEzquerra/vhdl_analyzer](https://bitbucket.org/AngelEzquerra/vhdl_analyzer)].
  - Parser for Perl last updated in 2002 [[cpan.org/~gslondon/Hardware-Vhdl-Parser-0.12](http://search.cpan.org/~gslondon/Hardware-Vhdl-Parser-0.12/)]
  - Grammar for Lex-Yacc (version not specified) [[vhdlverif.sourceforge.net](http://vhdlverif.sourceforge.net/index.html)]
  - Not complete VHDL 2008 AST [[github.com/grwlf/vsim/blob/master/grammar/vhdl-grammar](https://github.com/grwlf/vsim/blob/master/grammar/vhdl-grammar)]
  - ANTLR4 grammars [[github.com/antlr/grammars-v4](https://github.com/antlr/grammars-v4)]

Since I found no valid candidate, I had to accept that parsing VHDL is not straightforward. Some others reached the same conclusion before: [Parsing VHDL is [very] hard ](http://eli.thegreenplace.net/2009/05/19/parsing-vhdl-is-very-hard), [A Note on the Parsing of Complete VHDL-2002](http://ceur-ws.org/Vol-255/paper11.pdf).

Then, even though ANTLR4 could be the least bad starting point, maybe parsing VHDL with a general purpose engine is not the way to go. Since GHDL is a compiler, it actually parses at least part of the code. It's fast, lightweight, multiplatform, standalone and free. Moreover, it is known for being strict at conforming the standard. However, I know nothing about Ada and less about how compilers work. I don't know how much of the structure of the design has to be explicitly known by the GHDL core in order to generate machine code; i.e., whether blocks of VHDL code can be fed directly into an Ada compiler. Anyway, I have analized the content of a library file:

```
  package pkg at 1( 0) + 0 on 11;
  package body pkg at 6( 93) + 0 on 12;
  package mygpkg at 13( 232) + 0 on 13;
  entity ent at 17( 293) + 0 on 14;
  architecture arch of ent at 23( 400) + 0 on 15;
```

It seems that I could build a hierarchy view of the packages, entities and architectures defined in each file. On top of that, the [pretty print command](http://ghdl.readthedocs.io/en/latest/Invoking_GHDL.html#pretty-print-command) and the [cross-reference command](http://ghdl.readthedocs.io/en/latest/Invoking_GHDL.html#cross-reference-command) syntax highlight every word; the second one also links every identifier (packages, entities, architectures, types, signals...) to it's declaration. Then, although I have not found explicit references in the user guide, I would say that GHDL could be a good candidate. Nevertheless, the [interfacing to other languages](http://ghdl.readthedocs.io/en/latest/GHDL_implementation_of_VHDL.html#interfacing-to-other-languages) section of the user guide explains how to import subprograms, but not how to access to information about the design. I would say that the only choice at now is `AVHPI`, which is *a very light and slightly adapted version of VHPI*. I suppose that it is a very first step forward to something similar to [Aldec's VHPI Applications](https://www.aldec.com/en/support/resources/documentation/articles/1457).

In the end, the questions:

- @tgingold, is GHDL suitable for this purposes, or is it focused on generating executable machine code?
- Is Ada the only language which can be used to extract information of the elaborated design from GHDL?
- Which are the supported VHPI functions and objects?
- Given the actual status, how much effort would be needed to print the complete hierarchy of a design (top entity, ports, signals, processes, generate blocks, instantiated modules...) in any format (ascii tree to stdout, json, xml...).

tgingold, 2016-07-02:
On 02/07/16 07:52, 1138-4EB wrote:
> I've been thinking long about having some multiplatform (windows and
> gnu/linux), free (libre), lightweight and standalone tool to analize
> large bodies of VHDL 2008 code at block/RTL level. That tool [...]

Wow, a lot of references.  I haven't yet read all these pages, but I can
still try to answer to your questions.

> In the end, the questions:
>
>   * @tgingold <https://github.com/tgingold>, is GHDL suitable for this
>     purposes, or is it focused on generating executable machine code?

Yes, it is suitable.  GHDL is able to parse and analyze vhdl sources and
uses an AST as an internal representation of vhdl source.  This part is
clearly separated from the code generation and simulation part.
For example, the cross-reference only uses the parser and analyzer, but
not at all the code generation part.

>   * Is Ada the only language which can be used to extract information of
>     the elaborated design from GHDL?

No.  The AST representation (documented in iirs.ads) could be read from
any language.  It shouldn't be very hard to have a python interface
for example.

>   * Which are the supported VHPI functions and objects?

Far from being complete.  VHPI is only supported for the simulation
aspect, and just complete enough to support cocotb.

>   * Given the actual status, how much effort would be needed to print
>     the complete hierarchy of a design (top entity, ports, signals,
>     processes, generate blocks, instantiated modules...) in any format
>     (ascii tree to stdout, json, xml...).

It is already possible to display the hierarchy using the --disp-tree
option.  The output is an ascii tree, but it wouldn't be that difficult
to generate a json file.

I will comment on the first part of your message later.

Tristan.

tgingold, 2016-07-03:
On 02/07/16 07:52, 1138-4EB wrote:
> I've been thinking long about having some multiplatform (windows and
> gnu/linux), free (libre), lightweight and standalone tool to analize
> large bodies of VHDL 2008 code at block/RTL level. That tool [...]

In fact you want more than a vhdl parser: you also want a vhdl 
elaborator to get access to all instantiated units, signals and connections.

GHDL has all these data, at least internally.  The options --disp-tree 
can be used to dump some of them and vpi/vhpi could also be used to get 
more details.  However, this won't be as complete as RTL view, as ghdl 
is not a synthetizer and doesn't create a netlist.

Generating a dot/graphviz connection net shouldn't be very difficult.

Tristan.

tgingold, 2016-07-06:
I close this issue as this was just a question.
But do not hesitate to continue the discussion.

Tristan.

closed 2016-07-06 by tgingold

flip111, 2016-07-21:
@1138-4EB 

Nice idea, i had something similar in mind. Perhaps you can leave your e-mail address to stay in touch? So i will comment with what i know, point by point.

- Lightweight .. seems a bit of odd requirement. Performant seems more relevant.
- Block level visualization .. this is a problem of placing a graph in a 2D space. Ideally with the least amount of crossing edges, this can be done by algorithms that performs "Planarization". This package looked interesting http://www.ogdf.net/doku.php
- RTL level visualization .. this could be a flowchart per process. Possibly also a flowchart of concurrent processes.
- Parser .. yes you need one.
- Elaborator .. this is basically an analysis on the Abstract Syntax Tree. The parser can only detect syntax errors, but after analysis of the AST it can also detect logic errors. For example a signal assignment on a variable. It doesn't have to be a complete elaborator, some pretty simple analysis can already provide pretty useful information.

I'm doing the parser and hopefully later the elaborator. I'm building this in haskell and the progress is slow but steady. I haven't started on the problem of Planarization so this area is still wide open. I decided to start with parsing because this would already be useful, and other tools build on top of it. For visualization javascript would be preferred because it plays well with html documents, publication on the web and atom editor.

I was planning to leverage ghdl to provide values of signals at a point in time.  Or even to solve small snippets of logic. For example, given the following code:

```vhdl
variable A, B, C, D : boolean;
if notA && (B or C) then
  D := '1';
else
  D := '0';
end if;
```
- The parser could figure out that `notA` is a typo.
- The elaborator could figure out that `'1'` is not a valid value for type `boolean`.
- The RTL visualizer could draw a flowchart of this piece of logic.
- The RTL simulator (such as GHDL) can figure what will happen. Given the values of A, B and C what will be the result of expression `not A && (B or C)`.

Too bad i was unable to install ghdl, now i'm looking to interface with modelsim, but this is also not an easy thing to do. Have you got ghdl working?

> However, this won't be as complete as RTL view, as ghdl is not a synthetizer and doesn't create a netlist.

I'm not sure if a synthesis step would actually make it "more complete" ... after synthesis a lot of information is actually gone.

tgingold, 2016-07-22:
On 21/07/16 17:19, flip111 wrote:

[...]
> Too bad i was unable to install ghdl, now i'm looking to interface with
> modelsim, but this is also not an easy thing to do. Have you got ghdl
> working?

Can you clarify why you were unable to install ghdl ?  We do provide 
binaries and building it from scratch is not that difficult.

Tristan.

flip111, 2016-07-22:
@tgingold i wrote about the gcc install here https://github.com/tgingold/ghdl/issues/52

tgingold, 2016-07-22:
On 22/07/16 19:55, flip111 wrote:
> @tgingold <https://github.com/tgingold> i wrote about the gcc install
> here #52 <https://github.com/tgingold/ghdl/issues/52>

Ah, ok I remember.  But you should try the llvm or mcode backend instead 
of the gcc one.

jgjverheij, 2016-07-22:
@flip111  

> I'm doing the parser and hopefully later the elaborator. I'm building this in haskell and the progress is slow but steady. I haven't started on the problem of Planarization so this area is still [...]

I am trying to make co-simulation between CλaSH and Verilog/VHDL;  CλaSH is actually just Haskell code... Co-simulation with Icarus-Verilog and Modelsim works currently. Simulation can be done with source-files or with quasiquoters, in which the module only as string is used (written to file) currently. A future idea is to use a parser within Haskell to for example splice Haskell values in the DSL. 

Will your Haskell parser be freely available? If yes, can your parser be used in this context?

After I installed Icarus Verilog on a clean Ubuntu 16.04, I installed GHDL with the following commands:

`sudo apt-get install zlib1g-dev 
sudo apt-get install llvm-3.5 
sudo apt-get install clang
sudo apt-get install libedit-dev

# GNAT GPL 2016
wget http://mirrors.cdn.adacore.com/art/5739cefdc7a447658e0b016b
tar xf 5739cefdc7a447658e0b016b
cd gnat-gpl-2016-x86_64-linux-bin/
sudo ./doinstall
PATH="/usr/gnat/bin:$PATH"; export PATH

# GHDL
git clone https://github.com/tgingold/ghdl
cd ghdl
./configure --with-llvm-config=/usr/bin/llvm-config-3.5
make
sudo "PATH=$\$\;$PATH" make install`

flip111, 2016-07-22:
@jgjverheij you can not simulate CλaSH code directly as far as i know. CλaSH code compiles to vhdl or verilog. If you wanted to simulate CλaSH code with existing designs in RTL then you can generate RTL code first and just port map it like usual.

Thanks for the installation instructions ... it installed. But while elaborating it seems a file was not installed properly.

```
ghdl -e pattern_recognizer
gcc: error: /usr/local/lib/ghdl/v93/std/std_standard.o: No such file or directory
ghdl: compilation error
```
```
ll /usr/local/lib/ghdl/v93/std/
total 72K
-rw-r--r-- 1 root root  297 jul 22 22:47 std-obj93.cf
-rw-r--r-- 1 root root  59K jul 22 22:47 textio_body.o
-rw-r--r-- 1 root root 4,6K jul 22 22:47 textio.o
```
This `std_standard.o` file is still in `/home/flip101/Downloads/ghdl/`

jgjverheij, 2016-07-23:
@flip111 currently it is correct that is not possible to simulate CλaSH code with existing designs in RTL. That is why I am implementing co-simulation to make this possible.

Hereby a small Verilog example:

```
verilog_square :: (t ~ Signed 1000) => Signal t -> Signal t
verilog_square x    = coSim source "mult" x x
    where
        source      = [vsim| 
                                module mult (Out, Left, Right);              
                                
                                parameter data_width = 1000;
                        
                                input  signed [0:data_width-1] Left;  
                                input  signed [0:data_width-1] Right;                                                       
                                output signed [0:data_width-1] Out;                          
                                                                                                           
                                assign Out = Left * Right;                        
                                                                                  
                                endmodule                                       
                      |]
```

Which gives the following result:
```
λ> sampleN 10 $ verilog_square $ fromList [1..]
[1,4,9,16,25,36,49,64,81,100]
```

To splice Haskell (CλaSH) values into this module, for example to create this:

```
......
                                assign Out = $(x) * $(x);  
......
```
...a parser has to be created (in this case Verilog). When you said you are building a VHDL parser in Haskell, it got my attention, because that is something I can use to give more possibilities within the co-simulation.

flip111, 2017-11-13:
@jgjverheij Sorry for not gettin back to this earlier. Congratz on getting your verilog/clash interface repo up! I haven't checked tried to run it though as i am just interested in VHDL for the moment.

> Simulation can be done with source-files or with quasiquoters, in which the module is used as string (written to file) currently.

> A future idea is to use a parser within Haskell to for example splice Haskell values in the DSL.

Both these sentences i don't understand what you're getting at.

Unfortunately work on the VHDL parser is stalled because i'm lacking the appropriate haskell skills to finish it (after already many attempts). The state of it is that i have ~4500 lines of code ... all the rules are implemented (but not working) in VHDL 1993. Got basic machinery up for testing and pretty printing. For the moment the following rules are working:

* Grammar Level 0 (terminals only): AddingOperator, BaseSpecifier, BasedLiteral, BitValue, CharacterLiteral, DecimalLiteral, Direction, EntityClass, Exponent, ExtendedDigit, Identifier, Mode, MultiplyingOperator, RelationalOperator, ShiftOperator, Sign, SignalKind, StringLiteral, SubprogramKind
* Grammar Level 1 (all rules that can be constructed from terminal rules): AbstractLiteral, BaseUnitDeclaration, BitStringLiteral, EntityClassEntry, EnumerationLiteral, IdentifierList, IncompleteTypeDeclaration, Label, LogicalName, OperatorSymbol, SimpleName
* Grammar Level 2 (all rules that can be constructed with level 0 and 1): AliasDesignator, AttributeDesignator, Designator, EntityClassEntryList, EntityTag, EnumerationTypeDefinition, InstantiationList, LogicalNameList, NullStatement, Suffix
* Grammar Level 3: GroupTemplateDeclaration, LibraryClause

That's it ... the top rule `designFile` is on level 33 so still a long way to go. The concrete problems i'm facing are (in level 4):
* I used [lookahead](https://hackage.haskell.org/package/megaparsec-6.2.0/docs/Text-Megaparsec.html#v:lookAhead) in the terminal definition to check if the  subsequent character is in the set `&'()*+,-./:;<=>|[]` in which case a space is not needed. This works fine but now i have problems generating my [Arbitrary](https://hackage.haskell.org/package/QuickCheck-2.10.1/docs/Test-QuickCheck-Arbitrary.html#t:Arbitrary) instances which don't have the lookahead possibility.
* I also have rules ahead for which i need to start keeping State so that i can figure out how to interpret syntax based on earlier findings (see the article @1138-4EB linked)

Nic30, 2017-11-15:
Hello,
did you know that there are verilog/vhdl grammars for ANTLR4 [hdlConvertor](https://github.com/Nic30/hdlConvertor/tree/master/grammars).
It is possible to generate parser from them.
I do not know GHDL code, I just saying it may be useful for you...

Paebbels, 2017-11-15:
As far as I know, there is no free working VHDL-2008 grammar available on the internet. 

The grammar you are pointing to is GPL 3.0 license so it prohibits a lot of use cases... It also contains no special case handling rules for all the corner cases in the VHDL language. As far as I know, the ANTLR4 grammar needs a lot of additional manual handling to get it working with VHDL.

That's why I started my own Python implementation of a special kind of VHDL parser: [pyVHDLParser](https://github.com/Paebbels/pyVHDLParser/).

flip111, 2017-11-15:
@Paebbels i'm still looking for a library that can display entities and connections visually, do you know a good one?

Paebbels, 2017-11-15:
@flip111 
Now, not really. For entity instance dependencies or for port to signal mappings?

flip111, 2017-11-15:
@Paebbels i mean a digital schematic as can be generated by for example Vivado. It would be nice to use parts of the schematic as images in documentation.

Nic30, 2017-11-15:
@flip111 
I have some experimental "IDE" which does this thing, but it is not working because there is new D3.js library and i did only half of port.
It needs an implementation of algorithm form component placement from http://rtsys.informatik.uni-kiel.de/%7Ebiblio/downloads/theses/msp-dt.pdf p. 23.
I will do this on the end of semester.
[image]

flip111, 2017-11-15:
@Nic30 whao that's very cool you found a good algorithm. I searched for this for a long time. So your solution is al javascript based with D3.js? That would be perfect.

Nic30, 2017-11-15:
Yes, D3.js is used for graph rendering, Angular is used for for data handling and flask is used as python server.

1138-4EB, 2017-12-07:
@Nic30 , related to the topic, how do you get the structure of the modules? Do you use any parser?

flip111, 2017-12-07:
@1138-4EB Nic30 his software seems to be located here https://github.com/Nic30/hdlConvertor

I also want to mention https://github.com/Paebbels/pyVHDLParser which seems to be a nice parser which also supports VHDL 2008

Also found this https://kevinpt.github.io/hdlparse/ which is a newer library (between 2 and 7 months old) to generate documentation.

Nic30, 2017-12-08:
As @flip111  wrote. Also I am searching for cooperation. I am offering my help on similar project or anyone can help me. I am interested in everything which is related to HW especially HLS.

1138-4EB, 2017-12-09:
@Nic30 I have a meaningful update to this thread in the oven. It is focused on trying to come up with a collaboration scheme so that we can try to integrate the pieces and proofs of concept we have been developing independently. As you can see, I already edited the first comment, to include explicit references to pyVHDLParser (which I knew about, but I had not given it the relevance I feel it has) and to your solution.

Just to be clear, I do think that ANTLR4 is a dead path for the matter at hand, in the sense that at some point (which can be quite late) it will be harder to 'hack' ANTLR, instead of going with a different concept (say pyVHDLParse and/or GHDL). However, don't take me wrong, I find the frontend part of your project very interesting and inspiring. Indeed, I am ok with using ANTLR as a quick input source to the frontend, as long as we use some intermediate format between the parser and the visualizer, which let's us swap either of them.

---

NOTE: the content below might be copied and/or moved to a different thread soon. See note at the end.

---

I somehow drafted that (JSON) format [here](https://github.com/1138-4EB/elide/wiki/Data-format%28s%29#definition) which is an extended (and IMHO improved) version of the format used in icestudio, by taking ideas which are useful for a complete design flow (from edition, to simulation, synthesis, P&R and programming/configuration). In also considers the possibility to define data in a single or multiple files (this part is also above, but it is still in spanish). However, there might already exist a format which can be directly used. I'd like to hear about alternatives.

My proposal is to use this thread to discuss *parsing*. Since this is the repo of GHDL, I would focus on how to use/enhance `--disp-tree`, `--pp-html`, `--file-to-xml`... to generate the 'intermediate format'. Also, because some python features are being added to GHDL, the relation between GHDL and pyVHDLParser should also go here (see open questions in the updated first comment).

Then, I would open a different issue to discuss data requirements to build the visualization libraries/tools. We would discuss details about the intermediate format there, and set the minimum requirements which should be met by any parser in order to be usable (say GHDL, pyVHDLParser, ANTLR4). Note that the inputs can be not only parsers, but also 'documentation tools', such as [Doxygen](http://www.doxygen.org/) or [VHDLDocL](http://www.volkerschatz.com/hardware/vhdocl.html).

Last, I'd open a thread to talk about technical details to build the visualization libraries/tools. I like your approach using `D3.js`. Indeed, I considered using either that or `three.js`, but got quite stuck with graph theory. There is a JavaScript version of OGDF [too](https://github.com/Saheb/gsoc2013). Other references that I'd like to bring to discussion are [CodeMirror](https://codemirror.net/) , [Vue.js](https://vuejs.org/) and [wavedrom](wavedrom.com), [tikz-timing](https://ctan.org/pkg/tikz-timing)... However, there are two alternatives (on top of the almost fully handmade solution) which deserve attention:

- eclipse/che, which is introduced in [477#tocome](https://github.com/tgingold/ghdl/issues/477#tocome).
- [jgraph/mxgraph](https://github.com/jgraph/mxgraph).

---

When i first found mxgraph a couple of years ago, it was not open source, but a paid library. Then, although it seemed promising, I discarded it. Moreover, I was convinced that a tool for hardware design had to be written in compiled languages (say Ada, C, C++, golang) because of performance. And this is/was Java and JavaScript. Well, both issues have changed now:

- It is open source (Apache-2.0).
- I learn't a little bit about how to use golang for the backend, javascript for the frontend and AJAX to implement basic APIs; how to use docker and the browser as a 'lightweight' replacement for nodejs; and how to automate the development with watchers/livereloades (say grunt-contrib-watch).

And... well... [this](https://jgraph.github.io/mxgraph/javascript/examples/grapheditor/www/index.html) is open source, free (as in free beer), and requires only 2.34MB. That's just one among multiple [examples](https://jgraph.github.io/mxgraph/javascript/). The most exciting to me is [wires](https://jgraph.github.io/mxgraph/javascript/examples/wires.html). Note that adding a golang backend (10-20MB) and GHDL, we are talking about <30MB. Damn cool.

---

@Nic30 are you willing to consider the possibility to replace D3.js with mxgraph? I.e., use tha same ANTLR4 parser and layout algorithm you are using, but construct an XML/JSON you can provide as a model/input to mxgraph. I don't want to push you at all. Indeed, there are a couple of caveats:

- Backend examples for mxgraph are written in Java and/or PHP. I'd say no, no. So porting those functions to a different backend language would be required. Note that these 'backend' functions are to save/load files. The examples are already functional client-side (with a plain web server providing html and js files). Although I have not tried it, I'd say that it would work even on GitHub Pages. This means that, shall parsers/layout algorithms be implemented in JavaScript, we could provide a full client-side IDE.
- As stated previously, it is not reasonable to handle large projects composed of tens or hundreds of files in the browser. Therefore the possibilities to offload mxgraph should be explored to offer a performant solu [... cut ...]
Related to #477, ideally I'd like to have a ghdl organization, where we could create ghdl/gui to integrate multiple possibilities. Not only the content discussed here, but also any general topics about using GHDL with GUI tools. For example: how to compile and simulate from Atom or how to set up linting tools.

However, in the end, I really don't mind at all. We are talking about 3-6 threads for now.

---

Last thought: keeping a modular approach, but thinking big, do not forget ghdlsynth-beta, icestorm, arachne-pnr, yosys...

---

@flip111 didn't forget about you. The message in the oven is a reply to you.

Paebbels, 2017-12-09:
@1138-4EB 

The main difference or disadvantage of GHDL is that it is not Python. I don't mean this in a case of how it is written, but more in a case that several free cloud services allow us to install Python packages and addons, but no binary code. I'm using a Python based chain of cloud services to do post processing of VHDL source files. I see no way to handle this with GHDL.

The second difference is that GHDL (as any compiler) removed comments. As a tool that extracts e.g. documentation from sources, GHDL can not be used.

pyVHDLParser e.g. has an generic and abstract description of language constructs. We can move this part to a separate repository so both (or more) projects can use this description.

pyVHDLParser does not create a classic AST (Abstract Syntax Tree) as an intermediate representation. It creates a CodeDOM (Code Document Object Model). So modelling does not start at tokens as classical use; it starts at the language model and goes backward to the characters in a file.

I think that the later processing steps in pyVHDLParser could use GHDL and pyVHDLParser "frontend" as an input.

flip111, 2017-12-09:
# Parser applications #
The selection of libraries (which were already mentioned) really depend on what you want to do. I see some cons and pros about every library. https://github.com/Nic30/hdlConvertor doesn't support VHDL 2008 but should be fast because it's written in C++. https://github.com/Paebbels/pyVHDLParser supports VHDL 2008 but it's written in python so we can expect it to be slower. (i didn't look into verilog because i'm not familiar with it, but it's also something to consider).

Having broad language support is always good to have in any situation (different flavors of vhdl, different flavors of verilig, systemC, clash, ...). But the required parsing speed depends on the application where the parser is used. From slowest (parsing speed doesn't matter much) to fastest (parsing speed is critical)

* Generating documentation (both textual and generating schematics)
* Generating schematics while editing your code (you are not gonna switch between code and schematic view every second so it's ok if you have to wait a little)
* code linting / syntax checking on the command line (the context switch between CLI and editor will already be lower than with the schematic).
* IDE/editor integration .. you need nearly instant feedback.

.. feel free to extend this list with more meaningful parser applications (i can list a few more but i wanted to limit this list to the most interesting once for the moment)

# IDE #
> This means that, shall parsers/layout algorithms be implemented in JavaScript, we could provide a full client-side IDE.

> In the end, we are trying to build an IDE around GHDL.

Since you say "build an IDE around GHDL" i assume that you want the IDE to contain a waveform viewer and not just use GHDL for parsing.

Ok let's consider building an IDE on top of javascript in two scenario's:

## IDE in browser ##
You can't get around building a server-side app to interface with GHDL, python, C++ and various stuff. If you use any of them for the the critical parsing section you have to deal with additional latency of your HTTP request. For this reason editing code in the browser as a real IDE is already out of the window (a real IDE has code analyses on the fly like with [vhdl-tool LSP](https://www.vhdltool.com/)). At best you can do something similar to github which is good enough to make very simple changes to your code.

## offline IDE ##
You can still build a real IDE around javascript. The only sane way to do this is to latch onto already existing solutions, which are [atom](https://atom.io/) and [vscode](https://code.visualstudio.com/). I'm not a big fan of them because they are still slow. However it does work quite well and since there are big communities behind them it can only get better. The huge benefit here is that the javascript eco system makes it easy (compared to other solutions) to enhance these editors.

My conclusion is that the browser can never be good enough to offer serious IDE functionality (unless you host your server on the same machine of course). And atom/vscode are actually good enough. They are still slow editors but it's easy to push analytics to other tools like GHDL and dedicated parsers.

# Documentation #
For documentation i would say the browser is the preferred platform. Here you can have written documentation, display code snippets (no editing), display smaller or bigger schematics and sometimes it helps to show a waveform as well.

If the IDE solution still uses javascript for the most parts then there is real synergy between an IDE and a documentation viewer.

@Paebbels makes a good point here that it's important to be able to host on already existing infrastructure. Since maintaining infrastructure yourself is a day job on itself.

What is actually the benefit of GHDL for documentation? For the parsing there are other solutions in competition. So it's main advantage is actually what it was build for generating waveform (simulation). But i think it would be ok if the documentation writer can do the simulation offline and just upload the vcd file along with the documentation.

# Visualization #
In both IDE setting and documentation setting i think it would be useful to make visualizations in such a way that they allow you to explore the design. Of course a static image is already better than no image. But when doing things in the browser we can do better. I imagine looking at parts of your design and the clicking on the edges to start showing other parts they are connected to. If you want to do this the visualizing library needs to be fast enough for it so you don't have to wait long if your design gets big. I'm not sure which techniques have good hardware acceleration. All the libraries use either WebGL, canvas, SVG or HTML (this is the order i expect them to be fastest to slowest). Though for the moment it would be a waste of time to cook up a custom WebGL solution or something like that. Better go with an already existing solution (Nic30 d3.js or mxgraph or whatever). But it's interesting to see how these libraries respond when you load a big design into it.

But for really interesting visualizations let's move on to the next section ...

# Data flow analysis #
Data flow analysis is something compilers do for any language. In hardware design i think this is done in the elaboration phase (correct me if i'm wrong and it's actually in the analyzing phase). What is interesting about this for visualization (and actually also for when you are working w [... cut ...]  code in a CLI tool with a command line option interface. Now you have a CLI tool that can syntax check both VHDL and verilog. The same for if you want to target javascript (in a github editing way scenario). ANTLR is one of the best tools to design parsers. That doesn't mean that there are no other good solutions available (all the parsers there are right now are good parsers). I'm not saying here we should go with ANTLR, i'm just saying that it's good software and it has it's own benefits too.

Nic30, 2017-12-09:
# IDE

* browser/electron solution will be best, because there are many great libraries for graphic in JS and it works everywhere. And many HW developers that I know, are working on servers because there are licenses installed. So client-server app makes perfect sense for them.
(https://electronjs.org/ = offline version, server + chrome packed into executable)

* D3.js is one of fastest graph libraries, http://bl.ocks.org/mjromper/95fef29a83c43cb116c3
  * I used current IDE for development of AXI4 DMAs, there were 16K+ endpoints and I did not suffered from lags.

Basic display of components is not a problem. Choice mxgraph/D3.js is invariant.
With mxgraph I did not solve following problems:
* collapse and open internal structure of components
* component/path keep out
* diagrams with cyclic, self reference structures

I tried to use mxgraph json format but I did not found specification of it's json format back in times.

# Data flow analysis

* it is possible to discover clock etc, with lazy-loading and process analysis.
* It is not generally possible to discover complex interfaces like Avalon/Axi4 ...

Without HLS meta-informations and platform-informations you are limited to just view hierarchy design and operator nets. But those informations can be provided by user in form of XDC/ucf... which user probably has.

1138-4EB, 2017-12-09:
> @Paebbels
> The main difference or disadvantage of GHDL is that it is not Python. I don't mean this in a case of how it is written, but more in a case that several free cloud services allow us to install Python packages and addons, but no binary code. I'm using a Python based chain of cloud services to do post processing of VHDL source files. I see no way to handle this with GHDL.

I think this depends heavily on the context. From mine, that's precisely an advantage of GHDL over pyVHDLParser. I understand that you use several limited free services and are ok with having everything in Python, because Python itself is provided. But anyone using a clean platform which does not use python at all will have to install several additional mb in order to run pyVHDLParser. Therefore, I'd like to know which features justify the extra dependencies for someone running tests in travis, virtual machines, docker containers... In the context of ci, for example, adding extra dependencies limits the number of tasks that can be executed concurrently.

> @Paebbels
> The second difference is that GHDL (as any compiler) removed comments. As a tool that extracts e.g. documentation from sources, GHDL can not be used.

Is this a rigid 'can not be used' or does it mean 'it is not implemented'? GHDL includes `--pp-html`. Would keeping comments be just an enhancement to it?

My approach to the documentation is trying to combine Doxygen exporting XML with [pandoc](https://pandoc.org/) to convert it to md, rst, tex... But I wonder how far GHDL is in order to be able to replace Doxygen.

> @Paebbels
> pyVHDLParser e.g. has an generic and abstract description of language constructs. We can move this part to a separate repository so both (or more) projects can use this description

Where does that description fit in the flow? It's quite low level and unknown to be. Is it the output of the parser which includes all the details of the codebase so that tools can provide features such as, compilation/simulation, logic error checking, etc.?

> @Paebbels
> pyVHDLParser does not create a classic AST (Abstract Syntax Tree) as an intermediate representation. It creates a CodeDOM (Code Document Object Model). So modelling does not start at tokens as classical use; it starts at the language model and goes backward to the characters in a file.

If I don't get it wrong, GHDL, and every other 'normal' parser starts at the character level and goes up. Yours starts from the top and goes down to the details. Is it correct?

Related to this, do pyVHDLParser and GHDL have any difference/advantage when it comes to [recovering analysis](http://insights.sigasi.com/tech/recovering-vhdl-parser-0.html)?

> > @Paebbels
> I think that the later processing steps in pyVHDLParser could use GHDL and pyVHDLParser "frontend" as an input.

You mean to split it taking the generic and abstract description as the breakpoint? That sounds really good.

@Nic30, @flip111, can we agree on a couple of examples and generate outputs for them with different tools? I mean `--disp-tree`, `--pp-html`... for GHDL; the abstract description @Paebbels mentioned for pyVHDLParser; the one used in hwtIde, etc.

Nic30, 2017-12-10:
@1138-4EB Yes, lets create library of examples with example outputs for each tool or desired format.
Use cases for such a tools.
    * doc gen (data as DOM + comments in objects)
    * component diagram/hierarchy viewer (screenshot already shown in prev. comm.)
    * file dependency diagrams (format too simple, no need for attention)
    * universal HDL parser API (same data as comp. viewer or doc gen)
    * clocking analysis (same data as dep. diagram + timing information, can be saparated)
[image]

Potential use cases:
* duplicate code search
* HW regex or query language

Another features essential to hwtIde:
* wave viewer
* HLS scheduling diagram (like clocking analysis with ranges) 

And can we move to some new issue? Or where already did to  #477 and I just ignored it?

flip111, 2017-12-10:
> And many HW developers that I know, are working on servers because there are licenses installed. So client-server app makes perfect sense for them.

In this case better to enable an option "upload on save" in your favorite editor. Your linter can be local and low latency.

> D3.js is one of fastest graph libraries. there were 16K+ endpoints and I did not suffered from lags.

This is great news!

> It is not generally possible to discover complex interfaces like Avalon/Axi4

There is nothing for this now, but eventually it could be made by running all possible input combinations on that part of the circuit then derive rules from them (like [PSL](https://en.wikipedia.org/wiki/Property_Specification_Language) or any other formal verification language) and then compare with a database of rules for interfaces (which we don't have yet either).

> meta-informations and platform-informations

@Nic30 what information would be most important to you and do you have any ideas to present them in the schematic? Perhaps it would be a nice idea to have a working document.

1138-4EB, 2017-12-10:
> And can we move to some new issue? Or where already did to #477 and I just ignored it?

#477 is heavy WIP and content from there is being moved to different issues. This is one of them, but also 280, 377, 480... Therefore, content in #477 should not be overlapped with this thread. The only relation is that the possibility to create an organization is listed there, for now, and I'd like `ghdl/gui` to exist.

#477 is about using docker containers to build (GHDL, that tarballs for releases, ready-to-use images, web sites...). The conceptual IDE we are talking about in this thread will be available as a docker container (mostly because I don't install dev dependencies locally anymore), but I think that it should be deployable "standalone".

I use "standalone" because `GHDL + python/C++ parser + python/golang webserver + python/golang/C++ backend + frontend sources (html, css, js) + (optional) gtkwave + (optional) vunit + (optional) doxygen + (dependency) python runtime` is not exactly standalone, but... we get it.

About moving to some now issue to discuss GUI details, as commented above:

> Indeed, if any of these proposals is interesting to you and you agree on the proposed discussion plan, I'd encourage you not to hesitate to open a new thread.

Go on!

1138-4EB, 2017-12-11:
@flip111, @Nic30, if you agree, I can clean the wiki and the master branch in 1138-4EB/elide, and add both of you as collaborators. That will allow us to:

- Edit all these notes in the wiki in a more structured way.
- Have a site (1138-4EB.github.io/elide) to test full client-side viewers.
- Put the examples and test outputs in a branch:

> @Nic30, @flip111, can we agree on a couple of examples and generate outputs for them with different tools? I mean --disp-tree, --pp-html... for GHDL; the abstract description @Paebbels mentioned for pyVHDLParser; the one used in hwtIde, etc.

> [@Nic30](https://github.com/tgingold/ghdl/issues/111#issuecomment-350538808)
> @1138-4EB Yes, lets create library of examples with example outputs for each tool or desired format.

---

I can't remember now if I stated somewhere that my objective is to be able to make the documentation of VHDL projects as part of the creation process. This would not only make documenting more natural, but it would be very useful during the development, as a visual feedback of the sources. That is, I'd like to:

- Write comments as I describe modules in VHDL.
- Have (browsable) documentation and (interactive) structural/RTL diagrams generated in 'real-time'.
- Be able to open the sources from either the documentation or the diagrams.
- When any of the sources is edited, have the documentation and diagrams regenerated automagically.
- Ideally
  - Waveforms:
    - Feature to select some signals and a time frame, and have an image generated (SVG and or bitmap) and embedded in the documentation.
  - State Machines
  - Edit either structural/RTL diagrams or state machines graphically and have the sources updated.

Some of these features are not new (see first text block of the first comment in this thread or check commercial solutions such as Xilinx's, Mentor Graphics', Aldec's, Altera/Intel's or Sigasi's). However, I think there is no tool which satisfies all of them. The closest might be 'Sigasi Studio XL Doc' (I tried the evaluation version, which does not include all the [features](http://www.sigasi.com/products/)), but it is still missing the ideal part, and being libre.

---

> [flip111 2017-12-06](https://github.com/tgingold/ghdl/issues/111#issuecomment-349902011)
> @1138-4EB Nic30 his software seems to be located here https://github.com/Nic30/hdlConvertor

Thanks @flip111, I didn't realize that @Nic30 is the same person who wrote about ANTLR4.

> [@flip111 2016-07-21](https://github.com/tgingold/ghdl/issues/111#issuecomment-234287028)
> Lightweight .. seems a bit of odd requirement. Performant seems more relevant.

Let me elaborate: I don't like huge monolithic tools with heavy dependencies. At the same time, because VHDL projects can get so large and verbose, performance might be an issue. Therefore, I prefer to avoid interpreted languages, such as JavaScript, Python, Ruby... which is quite an issue because... you know...

At the same time, instead of very tightly coupled internal modules, I prefer to have some of those intermediate interfaces/APIs exposed so that some or all of the modules can be swapped with alternatives. Kind of 'build your own framework'. This is a compulsory requirement somehow, because we depend on external closed tools for synthesis, P&R and testing. Indeed, both VUnit and [PoC](https://github.com/VLSI-EDA/PoC) are good examples of this approach.

As a result, 'lightweight' means to use (multiple) task-specific lightweight tools/libraries in Ada, C, C++, Golang... for the backend (analysis, elaboration, graph layout, waveform generation, waveform format conversion...). If all of these are given in a known format, the frontend can be as lightweight or heavy/performant as wanted. It can be as simple as a CLI or as heavy as a Eclipse mod.

Indeed, because VUnit, pyVHDLParser and GHDL are already cli tools, I would forget about the 'lightweight' feature for now, and focus on standard interfaces instead.

>[ @tgingold](https://github.com/tgingold/ghdl/issues/111#issuecomment-230159008):
> GHDL has all these data, at least internally. The options --disp-tree
can be used to dump some of them and vpi/vhpi could also be used to get
more details. However, this won't be as complete as RTL view, as ghdl
is not a synthetizer and doesn't create a netlist.
>
> Generating a dot/graphviz connection net shouldn't be very difficult.

Let's start here:

NOTE: these are open questions that I will try to respond myself and provide minimal working examples. Of course, I'd be glad to be helped, but I don't expect anyone to do it.

- [ ] Is the output of `--disp-tree` 'good' enough to generate a dot/graphviz connection?
- [ ] Is the output of `--disp-tree` 'good' enough to generate a JSON object to be used for visualization?
  - GHDL already produces JSON with `--psl-report<=FILENAME>`
- [ ] Can `--pp-html` be modified to generate a data structure instead of raw HTML, in order to allow using static site generators (say Jekyll, Hugo, Sphinx...) to apply templates and/or refactor the content?
  - [ ] Which format should be used? Is JSON valid? XML?
  - [ ] This feature is listed in the 'use cases' of pyVHDLParser. Should both GHDL and pyVHDLParser generate compatible data structures? Doing so would make the web generator compatible with both tools/libraries for parsing.
[... cut ...]
> What is actually the benefit of GHDL for documentation? For the parsing there are other solutions in competition. So it's main advantage is actually what it was build for generating waveform (simulation). But i think it would be ok if the documentation writer can do the simulation offline and just upload the vcd file along with the documentation.

The main advantage for me is that GHDL is required. None of the other features make sense to me if I cannot simulate the code I'm writing. Therefore, using GHDL for documentation is about not adding dependencies to my framework, unless strictly required to get a feature. See references to existing GHDL options as `--pp-html` above.

1138-4EB, 2017-12-11:
@flip111, @Nic30, if you agree, I can clean the wiki and the master branch in 1138-4EB/elide, and add both of you as collaborators. That will allow us to:

- Edit all these notes in the wiki in a more structured way.
- Have a site (1138-4EB.github.io/elide) to test full client-side viewers.
- Put the examples and test outputs in a branch:

> @Nic30, @flip111, can we agree on a couple of examples and generate outputs for them with different tools? I mean --disp-tree, --pp-html... for GHDL; the abstract description @Paebbels mentioned for pyVHDLParser; the one used in hwtIde, etc.

> [@Nic30](https://github.com/tgingold/ghdl/issues/111#issuecomment-350538808)
> @1138-4EB Yes, lets create library of examples with example outputs for each tool or desired format.

---

I can't remember now if I stated somewhere that my objective is to be able to make the documentation of VHDL projects as part of the creation process. This would not only make documenting more natural, but it would be very useful during the development, as a visual feedback of the sources. That is, I'd like to:

- Write comments as I describe modules in VHDL.
- Have (browsable) documentation and (interactive) structural/RTL diagrams generated in 'real-time'.
- Be able to open the sources from either the documentation or the diagrams.
- When any of the sources is edited, have the documentation and diagrams regenerated automagically.
- Ideally
  - Waveforms:
    - Feature to select some signals and a time frame, and have an image generated (SVG and or bitmap) and embedded in the documentation.
  - State Machines
  - Edit either structural/RTL diagrams or state machines graphically and have the sources updated.

Some of these features are not new (see first text block of the first comment in this thread or check commercial solutions such as Xilinx's, Mentor Graphics', Aldec's, Altera/Intel's or Sigasi's). However, I think there is no tool which satisfies all of them. The closest might be 'Sigasi Studio XL Doc' (I tried the evaluation version, which does not include all the [features](http://www.sigasi.com/products/)), but it is still missing the ideal part, and being libre.

---

> [flip111 2017-12-06](https://github.com/tgingold/ghdl/issues/111#issuecomment-349902011)
> @1138-4EB Nic30 his software seems to be located here https://github.com/Nic30/hdlConvertor

Thanks @flip111, I didn't realize that @Nic30 is the same person who wrote about ANTLR4.

> [@flip111 2016-07-21](https://github.com/tgingold/ghdl/issues/111#issuecomment-234287028)
> Lightweight .. seems a bit of odd requirement. Performant seems more relevant.

Let me elaborate: I don't like huge monolithic tools with heavy dependencies. At the same time, because VHDL projects can get so large and verbose, performance might be an issue. Therefore, I prefer to avoid interpreted languages, such as JavaScript, Python, Ruby... which is quite an issue because... you know...

At the same time, instead of very tightly coupled internal modules, I prefer to have some of those intermediate interfaces/APIs exposed so that some or all of the modules can be swapped with alternatives. Kind of 'build your own framework'. This is a compulsory requirement somehow, because we depend on external closed tools for synthesis, P&R and testing. Indeed, both VUnit and [PoC](https://github.com/VLSI-EDA/PoC) are good examples of this approach.

As a result, 'lightweight' means to use (multiple) task-specific lightweight tools/libraries in Ada, C, C++, Golang... for the backend (analysis, elaboration, graph layout, waveform generation, waveform format conversion...). If all of these are given in a known format, the frontend can be as lightweight or heavy/performant as wanted. It can be as simple as a CLI or as heavy as a Eclipse mod.

Indeed, because VUnit, pyVHDLParser and GHDL are already cli tools, I would forget about the 'lightweight' feature for now, and focus on standard interfaces instead.

>[ @tgingold](https://github.com/tgingold/ghdl/issues/111#issuecomment-230159008):
> GHDL has all these data, at least internally. The options --disp-tree
can be used to dump some of them and vpi/vhpi could also be used to get
more details. However, this won't be as complete as RTL view, as ghdl
is not a synthetizer and doesn't create a netlist.
>
> Generating a dot/graphviz connection net shouldn't be very difficult.

Let's start here:

NOTE: these are open questions that I will try to respond myself and provide minimal working examples. Of course, I'd be glad to be helped, but I don't expect anyone to do it.

- [ ] Is the output of `--disp-tree` 'good' enough to generate a dot/graphviz connection?
- [ ] Is the output of `--disp-tree` 'good' enough to generate a JSON object to be used for visualization?
  - GHDL already produces JSON with `--psl-report<=FILENAME>`
- [ ] Can `--pp-html` be modified to generate a data structure instead of raw HTML, in order to allow using static site generators (say Jekyll, Hugo, Sphinx...) to apply templates and/or refactor the content?
  - [ ] Which format should be used? Is JSON valid? XML?
  - [ ] This feature is listed in the 'use cases' of pyVHDLParser. Should both GHDL and pyVHDLParser generate compatible data structures? Doing so would make the web generator compatible with both tools/libraries for parsing.
[... cut ...]
> [@flip111](https://github.com/tgingold/ghdl/issues/111#issuecomment-350451075)
> What is actually the benefit of GHDL for documentation? For the parsing there are other solutions in competition. So [...]

The main advantage for me is that GHDL is required. None of the other features make sense to me if I cannot simulate the code I'm writing. Therefore, using GHDL for documentation is about not adding dependencies to my framework, unless strictly required to get a feature. See references to existing GHDL options as `--pp-html` above.

Nic30, 2017-12-16:
@1138-4EB , @flip111 
I will respond after school projects and exams, my semester became much more time than I taught.

Paebbels, 2017-12-17:
@1138-4EB 

pyVHDLParser is also a demonstrator. It's meant to allow easy adding and modification of VHDL language features. If the concepts of pyVHDLParser work out, it's possible to translate them e.g. C# There is no intention to be very fast. Being very fast requires heavy optimizations and thus disallows modifications for future language investigations.

Doxygen and similar documentation tools are not capable of handling the specifics of VHDL. They are limited to classes and methods. You can not transform VHDL to these paradigms. That's why I prefer e.g. Sphinx and Restructure Text. It can be extended to all needs of VHDL.

>> pyVHDLParser e.g. has an generic and abstract description of language constructs.
>
> Where does that description fit in the flow?

pyVHDLParser provides a CodeDOM (semantic object model). It's a object-oriented model of the VHDL source code. It is not specific to any parser or compiler. [Read the articles of Ken Beckett for more information.](https://www.codeproject.com/Articles/488657/The-Future-of-Software-Development-CodeDOMs-Part)

> If I don't get it wrong, GHDL, and every other 'normal' parser starts at the character level and goes up. Yours starts from the top and goes down to the details. Is it correct?

Yes.

> Related to this, do pyVHDLParser and GHDL have any difference/advantage when it comes to recovering analysis?

As far as it feels with GHDL: It stops at the first error. pyVHDLParser is more tolerant, because it analyzes many thing in later steps, while more code was already analysed.

1138-4EB, 2017-12-18:
> Doxygen and similar documentation tools are not capable of handling the specifics of VHDL. They are limited to classes and methods. You can not transform VHDL to these paradigms. That's why I prefer e.g. Sphinx and Restructure Text. It can be extended to all needs of VHDL.

I think we are mixing concepts here. There are two different tasks: i) extract comments and metadata from sources and ii) generate a structured document (HTML and/or PDF) with the extracted info.

AFAIK, only Doxygen and similar tools (say VHDocL or gnatdoc) are capable of extracting specially decorated comments from sources. This is true for C-type languages, but also for [python](https://www.stack.nl/~dimitri/doxygen/manual/docblocks.html#pythonblocks) and (although buggy) [VHDL](https://www.stack.nl/~dimitri/doxygen/manual/docblocks.html#vhdlblocks).

A quite different issue is how to elaborate a HTML site or PDF from text-only structured documents. Sphinx, falls in this group, along with Pandoc, LaTeX, Hugo, Jekyll, wikis... Note that there is no relation to the sources, because text-only structured documents are written and updated manually.

Note: sphinx can behave as doxygen with python sources, which is what it was created for. It has been extended to support other languages, but the [list](https://bitbucket.org/birkenfeld/sphinx-contrib/src/8e295053a27d5914127fbb01e2a8fb57421897a7/README.rst?at=default&fileviewer=file-view-default) is quite short compared to doxygen.

Then, any of the tools in this last group can be extended to all needs of VHDL, as long as a parser is used to generate intermediate formats (e.g. RestructureText/Markdown). Indeed, because Sphinx is lacking a VHDL domain, [extending it](https://stackoverflow.com/questions/37420560/how-to-pre-process-source-files-while-a-sphinx-run) is not very different from replacing [vhdlparser](https://github.com/doxygen/doxygen/tree/master/vhdlparser) with a more feature-rich parser.

This parser, say pyVHDLParser, needs some mechanism to decide which comments/components should be rendered in the final report and which are to be shown only in source listings. In the end, some syntax must be defined, which will likely conflict or replicate the conventions used by Doxygen. For example, [VHDocL](http://www.volkerschatz.com/hardware/vhdocl.html), *can parse doxygen markup and come with a conversion script*. There is also an [extension](https://pythonhosted.org/sphinxcontrib-cmtinc/) for Sphinx to identify doxygen comments. What is your approach here?

Again, I think we should take into account in this thread that services as Travis exists, and can be used as a helper or a replacement for RTD. Is it possible for pyVHDLParser to output an intermediate XML instead of rst sources? With an 'XML to rst' parser it would still fit your workflow. At the same time, it would allow not python-limited environments to use different tools for rendering.

Nic30, 2018-02-05:
I am back from the dead. Thanks to overloaded semester I have nearly semester free.
I did start working on netlist query language. I rewrite Nic30/hwtIde to use jgraph/mxgraph as it's dataformat. I had many other works, but first results can appear next week.

@flip111 
> what information would be most important to you and do you have any ideas to present them in the schematic? Perhaps it would be a nice idea to have a working document.

* component/process level connections
* expandable hierarchy
* operand level connections on lowest level

I will allow to edit hwtIde wiki to all.

@1138-4EB
I have seen you wiki https://github.com/1138-4EB/elide/wiki and I agree with everything.
It may be useful to have edit rights.

Nic30/hwtIde - blocks are just hdl components they can be imported/exported etc.

(RTL) Schematic view  - expanding blocks - I do not know if they can have ports but I will find out 

> when you say that you didn't solve those problems, do you mean that those are not built-in features or that you couldn't dive into the code far enough to get it done?

My worst problem was implementation of alg. for initial layout (place components form HDL somehow to have nice diagram with minimum nets crossing).
Current alg. is A* based, need to be reworked as mentioned, I found a paper which has right. alg. for this. But it is quite complex. (the problem is NP-hard, bit it is more like "NP^3")

For everything else there is JS lib which can do it.

> leaving mxgraph apart, and focusing on your application, which features does it provide? I mean, did you implement every menu and action by hand, or are you using any library? A quick list of features:

>     Dragging.

D3.js

>    Rotation.

D3.js

>    Fixed connections.
>    Waypoints.

all connections are just json, components are placed by A* based alg. to have minimum crossing connections, these connections are just paths between route points rendered by D3.js

>    Styles for arrows, blocks, background, text...

css

>    Zoom.

D3.js

>    Grid.

simple JS

>    Outline.

D3.js

>    Import xml/json.

JS native

>    Export to png/gif/jpg/pdf/svg/xml/json.

in D3.js, not gui accessible yet

>    Copy & Paste.

simple JS

>    Contextual (right-click) menu.

none (not useful)

>    Tooltips.

D3.js plugin

>    Arrange (front/back).

possible in D3.js, not useful because components should not overlap

>    Select layout algorithm.

Select is done by D3.js and custom JS, it is easy.

>    Group elements in a container. Remove a container but leave elements connected to external blocks.

Containers are in very early prototype, renderer is just called on box, there is problem with shared history
and port placement.

>    Hierarchical navigation.

Done on file-browser level (needs to be on component diagram level)

>    Undo/Redo.

Done, custom JS, (actually back in time I had to completely rewrite IDE to support this) 

>    Add from library/list of shapes/modules.

Modules are loaded from HDL or can be crafted in GUI, only rectangle shapes yet.

>    Layers. Useful to add info not present in VHDL sources.

Maybe it would be better to support only hierarchy based layers to keep things simple.

flip111, 2018-02-05:
> I found a paper which has right. alg. for this

care to link your source? Or are you referring to the paper from before?

Nic30, 2018-02-06:
@flip111 
Yes, I was writing about algs. from http://rtsys.informatik.uni-kiel.de/%7Ebiblio/downloads/theses/msp-dt.pdf

1138-4EB, 2018-04-10:
I created a new repository a couple of days ago, where we can share/integrate the proofs of concept: https://github.com/1138-4EB/hwd-ide It started as a copy of the develop branch and the wiki of 1138-4EB/elide, but I thought it'd be better to use a new repo, because I have some personal unrelated content there. Now we can adapt/reorganize the content to fit the needs of different proofs.

@Nic30 and @flip111 were already added as collaborators. If any other wants to contribute, please let us know.

The wiki is public, so that anyone can contribute. Indeed, a couple of days ago @Paebbels also created [awesome-vhdl](https://github.com/VHDL/awesome-vhdl). I believe that some of the references in the wiki might be moved there. The same applies to many resources mentioned in this issue. However, I am not sure about the "awesomeness" of all of them, since many are not under active development. @Paebbels, do you think that it would be interesting to add not active/complete projects to the awesome list? I would try to avoid mixing "really awesome" and "not so awesome, but really interesting" efforts, but I think it'd be required to somehow state the status/issues.

---

Right now, there is a single proof of concept that integrates [filebrowser/filebrowser](https://github.com/filebrowser/filebrowser) with `ghdl/ext:vunit` and `ghdl/ext:vunit-gtkwave` (see #477). This is a very basic web-based editor with GHDL, VUnit and (optional) GtkWave in the backend.

Travis CI is set up to execute two stages after each push to branch develop:

- Build docker images, and then push the built images to [hub.docker.com/r/ghdl/ext](https://hub.docker.com/r/ghdl/ext/).
- Execute test scripts in subdir [`examples`](https://github.com/1138-4EB/hwd-ide/tree/develop/examples).

It is ready to start trying to put other stuff together!

---

I believe we can move the conversation here to the new repo. That way, we will produce less "noise" here, and we'll bother Tristan only when we need something related to GHDL. Feel free to open as many issues as you want, as we've talked about many different features.

PS: @flip111, @Nic30, I think that all the references here are already present in the wiki of the new repo, but don't hesitate to add any you might miss.

kraigher, 2018-11-08:
@Nic30 I updated the readme on how to try it out. I will add CI soon.

kraigher, 2018-11-08:
@Nic30 You can also join the gitter channel https://gitter.im/rust_hdl/Lobby to ask questions and get support.

kraigher, 2018-11-08:
@Nic30 Now there is Travis CI as well.

flip111, 2018-11-09:
@kraigher what data structures are you using for symbol table?

kraigher, 2018-11-09:
@flip111 A hash table from name to unique_id. Since VHDL is case insensitive symbols that differ only in case get the same id. The id can be used for fast comparison.

The code is here: 
https://github.com/kraigher/rust_hdl/blob/master/vhdl_parser/src/symbol_table.rs

flip111, 2018-11-09:
@kraigher i read elsewhere on the interwebz that vhdl is case insensitive so i don't doubt you about it, but i was curious how it was actually described in the standard. I had a quick look in the 1993 and 2008 standard but i couldn't find it, do you know? Perhaps it might just be so that people think VHDL is case insensitive because a lot of compilers are but it's not defined by the standard. Also with extended_identifier it seems strange to me if it were case insensitive because this is the special case of identifiers in which special characters can be used. So in this case it might also be interesting for the user to differentiate between lowercase and uppercase.

Is it already possible with the parser to lookup symbols from different scopes ?

kraigher, 2018-11-09:
The parser currently only does the parsing into an AST. Semantic analysis is the next step and then symbol references will be resolved.

Regarding case in-sensitivity the standard does not use this terminology but rather say:

*LRM 15.4.2 Basic identifiers*
> Basic identifiers differing only in the use of corresponding uppercase
and lowercase letters are considered the same.

*Also regarding keywords*
> NOTE 1—Reserved words differing only in the use of corresponding uppercase and lowercase letters are considered as
the same (see 15.4.2)

*LRM 15.4.3 Extended identifiers*
> Extended identifiers differing only in the use of corresponding uppercase and lowercase letters are distinct.
Moreover, every extended identifier is distinct from any basic identifier.
and lowercase letters are considered the same.

So normal identifiers are case insensitive but extended identifiers are not.
I currently treat extended identifiers as case in-sensitive so that is a bug. Good that you asked.