ghdl/ghdl #122: ld: cannot find -lz

RGNagel, 2016-07-24:
I followed the tutorial "Starting with GHDL" ( [](http://ghdl.readthedocs.io/en/latest/Starting_with_GHDL.html#the-hello-world-program) ) 
and I got stuck at:
$ ghdl -e -v hello_world
and it printed out:
>>>>>>>>>
/usr/bin/gcc -o hello_world e~hello_world.o /usr/lib/ghdl/v93/std/std_standard.o /usr/lib/ghdl/v93/std/textio.o /usr/lib/ghdl/v93/std/textio_body.o hello.o /usr/lib/ghdl/libgrt.a -ldl -lm -Wl,--version-script=/usr/lib/ghdl/grt.ver -Wl,--export-dynamic -L./ -lz
/usr/bin/ld: cannot find -lz 
collect2: error: ld returned 1 exit status
ghdl: compilation error
<<<<<<<<<

I installed Ghdl by binary package and I assured I had installed all dependences. 
System: Debian GNU/Linux 8 (jessie) 32-bit.

tgingold, 2016-07-24:
On 24/07/16 03:56, Rafael Nagel wrote:
> I followed the tutorial "Starting with GHDL" (
> <http://ghdl.readthedocs.io/en/latest/Starting_with_GHDL.html#the-hello-world-program>
> )
> and I got stuck at:
> $ ghdl -e -v hello_world
> and it printed [...]

You need to install zlib1g-dev

Tristan.

RGNagel, 2016-07-24:
Thanks for the fast and right reply, Tristan. It worked like a charm.

tgingold, 2016-07-24:
On 24/07/16 07:51, Rafael Nagel wrote:
> Thanks for the fast and right reply, Tristan. It worked like a charm.

Great!

closed 2016-07-25 by tgingold