ghdl/ghdl #123: Recursive instantiation of components...

Brian-Drummond, 2016-07-25:
Interesting testcase on StackExchange...

http://stackoverflow.com/questions/38567544/recursive-self-instantiation-component-vhdl

Matthew Taylor's answer to the question compiles in ghdl (0.34dev) but doesn't elaborate...

> ghdl -e adder
> 
> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
> Compiled with GNAT Version: 4.9.3
> In directory: /home/brian/Projects/ghdl/play/
> Command line:
> /usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9/ghdl1 -P/usr/local/lib/gcc/x86_64-unknown-linux-gnu/4.9/vhdl//v93/std/ -P/usr/local/lib/gcc/x86_64-unknown-linux-gnu/4.9/vhdl//v93/ieee/ --elab adder -l e~adder.lst -quiet -o e~adder.s e~adder
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
> Message: trans.adb:393
> Call stack traceback locations:
> 0xfceacf 0x571f26 0x65c351 0x669d02 0x695353 0x694da6 0x662861 0x6921af 0x6960d8 0x5ecc27 0x5c9b0e 0x9a4710 0x9a630e 0x5ecc79 0x5c6f05 0x7f93f5d415ee 0x5c7527 0xfffffffffffffffe
> ******************************************************************
> Execution terminated by unhandled exception
> 

[adder.vhd.txt](https://github.com/tgingold/ghdl/files/381942/adder.vhd.txt)

tgingold, 2016-07-25:
Works with the latest commit.

Tristan.

closed 2016-07-25 by tgingold

Brian-Drummond, 2016-07-25:
Thanks! But ... Built the latest commit, with gcc backend, according to the instructions I used last time...

> ghdl -a -v --std=08 adder.vhd
> ghdl:warning: library std does not exists for v08
> ghdl:warning: library ieee does not exists for v08
> /usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9/ghdl1 --std=08 -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -quiet -o adder.s adder.vhd
> /usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9/ghdl1: cannot find "std" library
> ghdl: compilation error

/usr/local/lib/ghdl contains ... only libbacktrace.a
Last time I built it, "make install" put libbacktrace and the libraries in
/usr/local/lib/gcc/x86_64-unknown-linux-gnu/4.9/vhdl

something changed?

Brian-Drummond, 2016-07-25:
something changed ... I see a new step in the build instructions. Forget the above...

Paebbels, 2016-07-25:
Is this related to the (old) recursive instantiation issue #2?

Brian-Drummond, 2016-07-25:
@Paebbels I'm not sure but I don't think so. It occurs at an assert in a different file, #2 was a constraint error, and my build of ghdl was fairly recent, so it's something fixed in the last month.