ghdl/ghdl #126: Segmentation fault when overloaded implicit operators are on different package.

vicencb, 2016-07-28:
Hello,
when building the attached code with any of those commands:
`ghdl -a --std=08   seg_fault.vhd && ghdl -e --std=08   e && ./e`
`ghdl -a -fexplicit seg_fault.vhd && ghdl -e -fexplicit e && ./e`
The resulting executable seg-faults.
It has been tested with gcc in linux with the released [ghdl-0.33](https://github.com/tgingold/ghdl/releases/tag/v0.33) and with [ghdl-0.34dev](https://github.com/tgingold/ghdl/commit/3e04f0aadc73a45eeba46d4ee3bc5a533b8d4a47).

Regards,
  Vicenç.

[seg_fault.vhd](https://github.com/tgingold/ghdl/files/389517/seg_fault.txt)
P.S.: GitHub did not accept the `.vhd` extension  :disappointed:

tgingold, 2016-07-29:
On 29/07/16 00:55, vicencb wrote:
> Hello,
> when building the attached code with any of those commands:
> |ghdl -a --std=08 seg_fault.vhd && ghdl -e --std=08 e && ./e|
> |ghdl -a -fexplicit seg_fault.vhd && ghdl -e -fexplicit e && ./e|
> The [...]

Hello,

this is simply an infinite recursion: pb."=" is calling itself.

We have discussed about a compile-time warning, that would be helpful.

Tristan.

> seg_fault.vhd <https://github.com/tgingold/ghdl/files/389517/seg_fault.txt>
> P.S.: GitHub did not accept the |.vhd| extension 😞

Did you try .vhdl ?

vicencb, 2016-07-29:
Oh, I see the recursion now that you point at it. That makes the side note about `use work.pa."=";` invalid and so this whole bug report.
It would be nice to have this compile-time warning, but I understand that detecting if a 
recursion is infinite or not is extremly difficult (if possible at all) for all possible cases.

Regards and thanks for your time.
 Vicenç.

>> seg_fault.vhd <https://github.com/tgingold/ghdl/files/389517/seg_fault.txt>
>> P.S.: GitHub did not accept the |.vhd| extension :disappointed:

> Did you try .vhdl ?

Yes, it neither worked.

Paebbels, 2016-07-29:
The issue was discussed in #28 [Deep recursion detection](https://github.com/tgingold/ghdl/issues/28).

closed 2016-07-29 by vicencb