ghdl/ghdl #127: compiling ghdllib requires pre-installed ghdl

vicencb, 2016-07-29:
Hello,
When building ghdllib it fails if ghdl has not been installed before.
It has been tested with d0d04ba9b19e0f4a0db16c70db1b3452e16d6923 on linux and the gcc backend.
It reports: `No rule to make target '/usr/bin/ghdl'`
The issue seems to be [here](https://github.com/tgingold/ghdl/blob/master/Makefile.in#L156).

Related: https://github.com/tgingold/ghdl/commit/e35b24adbb1d0e76c076fe475c8442738b3378be#commitcomment-18440050

Regards,
  Vicente.

tgingold, 2016-07-29:
On 29/07/16 03:00, vicencb wrote:
> Hello,
> When building ghdllib it fails if ghdl has not been installed before.
> It has been tested with d0d04ba
> <https://github.com/tgingold/ghdl/commit/d0d04ba9b19e0f4a0db16c70db1b3452e16d6923>
> on linux and the gcc backend.
> It reports: |No rule to make target '/usr/bin/ghdl'|
> The issue seems to be here
> <https://github.com/tgingold/ghdl/blob/master/Makefile.in#L156>.
>
> Related: e35b24a#commitcomment-18440050
> <https://github.com/tgingold/ghdl/commit/e35b24adbb1d0e76c076fe475c8442738b3378be#commitcomment-18440050>

I think you can work-around this issue by calling:
   make GHDL=path/to/ghdl ANALYZE_OPTS="--GHDL1=path/to/ghdl1" \
vhdl.libs.all libs.vhdl.standard

Would that work ?

I agree this is not very convenient, but I don't see any simple change 
to fix that.

Tristan.

vicencb, 2016-07-29:
The work-around needed a few more tweaks to work:
```makefile
  make -j1 \
    GHDL="${srcdir}/gcc-build/gcc/ghdl" \
    ANALYZE_OPTS="--GHDL1=${srcdir}/gcc-build/gcc/ghdl1" \
    STD_GHDL_FLAGS="--GHDL1=${srcdir}/gcc-build/gcc/ghdl1" \
    vhdl.libs.all libs.vhdl.standard libgrt.a
```
During the libs build there are a lot of warnings about the `std` and `ieee` libraries not yet compiled, but in the end it works fine.
Another issue is that parallel builds do not work, but that is a minor problem.

Regards,
  Vicenç.

tgingold, 2016-07-30:
Great.  This should be rationalised!

vicencb, 2017-05-29:
Hello Tristan,
I've tried building ghdl again and now the workaroud has changed.
Now it requires this patch:
```
--- Makefile.in
+++ Makefile.in
@@ -52,6 +52,8 @@
 
 VHDL_LIB_DIR=$(prefix)/$(libdirsuffix)
 
+GHDL_BIN?=$(bindir)/ghdl$(EXEEXT)
+
 # Debug
 OPT_FLAGS=-g
 ADA_FLAGS=-gnata
@@ -222,7 +224,7 @@
 	$(CP) -p $(srcdir)/dist/gcc/lang-specs.h $(gcc_vhdl_dir)
 
 libs.vhdl.gcc:
-	$(MAKE) -f $(srcdir)/libraries/Makefile.inc $(LIBVHDL_FLAGS_TO_PASS) GHDL=$(bindir)/ghdl$(EXEEXT) GHDL_FLAGS="$(LIB_CFLAGS)" vhdl.libs.all libs.vhdl.standard
+	$(MAKE) -f $(srcdir)/libraries/Makefile.inc $(LIBVHDL_FLAGS_TO_PASS) GHDL=$(GHDL_BIN) GHDL_FLAGS="$(LIB_CFLAGS)" vhdl.libs.all libs.vhdl.standard
 
 ghdllib: libs.vhdl.gcc $(libdirsuffix)/libgrt.a
 
```
and execute make as:
```
make -j1 GHDL_BIN="${srcdir}/gcc-build/gcc/ghdl" LIB_CFLAGS="--GHDL1=${srcdir}/gcc-build/gcc/ghdl1" ghdllib
```

Regards,
  Vicenç.

commit 4886d934 (2017-05-30): Define GHDL_GCC_BIN in Makefile.in

Fix #127
closed 2017-05-30 by tgingold