ghdl/ghdl #129: Can't assign to an array type with negative integer bounds any more

wjl, 2016-08-02:
When doing complex FPGA DSP algorithms I use tons of custom array data types such as fixed-point values. GHDL has always supported this fine, but the latest version gives me errors when trying to convert closely-related array types.

Where I actually ran into this is in my huge DSP algorithm library used for multiple designs, but here is a very small example that shows the problem:

It's kind of obvious what is going on: GHDL is failing here because the array 'low on the fixed-point type is negative, but std_ulogic_vector expects natural bounds. I've come to expect GHDL to be picky about the VHDL standard, however, I believe this is actually completely legal code and should not be failing. It has always worked before with GHDL, and it still works with at least 3 other simulators (ModelSim, Active-HDL, Xilinx ISE).

```
library ieee;
use ieee.std_logic_1164.all;

entity example is
end entity;

architecture bug of example is

   -- A fixed point type
   type fixed_point is array (integer range <>) of std_ulogic;

   signal fp   : fixed_point(5 downto -3); -- 6 integer bits, 3 fractional, length=9
   signal sulv : std_ulogic_vector(8 downto 0); -- length=9

begin

   -- This conversion works.
   fp <= fixed_point(sulv);

   -- This conversion SHOULD work, but instead it results in this error:
   -- ./example:error: bound check failure at example.vhdl:23
   -- ./example:error: simulation failed
   sulv <= std_ulogic_vector(fp);

end architecture;
```

```
$ ghdl --version
GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2016-08-03:
I am pretty sure this is not valid.
The type of fp is fixed_point (5 downto -3).
If converted to std_ulogic_vector, the type will be std_ulogic_vector (5 downto -3), which is not valid.

I think here ghdl strictly follow the LRM, but maybe this should be relaxed ?

Paebbels, 2016-08-03:
The VHDL-2008 fixed point package has a `to_slv` conversion function ([line 1039](https://github.com/tgingold/ghdl/blob/master/libraries/ieee2008/fixed_generic_pkg.vhdl#L1039-L1069)) 
to address this problem.

wjl, 2016-08-03:
I would have thought this would be perfectly legal. After all, you can do a simple assignment between, say, a std_logic_vector(7 downto 4) and a std_ulogic_vector(3 downto 0) without getting bound check violations or resorting to conversion functions with loops. It is strange that this has worked for years in many simulation and synthesis tools (all VHDL '93 based) and suddenly it doesn't work in GHDL any more. 

But, the comment in the VHDL 2008 package you referenced indeed makes it sound like this is truly invalid VHDL. If that's the case, GHDL is right, but inconsistent with most VHDL implementations (simulation and synthesis) that I have access to. Of course, it wouldn't be the first time GHDL conforms where other tools are more forgiving ... most of the time that's nice, but sometimes it's perturbing. =)

Anyway, thanks for looking into this. If it's really not a bug, feel free to close this out. I am happy to use my own `to_slv` type function in my libraries, since it sounds like this is really invalid VHDL. This is what I've already done for as a workaround -- now it sounds like I should just make it a permanent change!

Paebbels, 2016-08-04:
Having a second thought on your example:

```VHDL
signal a : std_logic_logic(3 downto 0);
signal b : std_logic_logic(7 downto 4);
-- ...
b <= a;  -- without any conversion
```

... maybe, this is related to the type conversion of closely related types.

tgingold, 2016-08-04:
On 04/08/16 00:50, Wesley J. Landaker wrote:
> I would have thought this would be perfectly legal. After all, you can
> do a simple assignment between, say, a std_logic_vector(7 downto 4) and
> a std_ulogic_vector(3 downto 0) without getting bound check violations
> or resorting to conversion functions with loops.

Yes, but there is no type conversion in assignments.

 > It is strange that this
> has worked for years in many simulation and synthesis tools (all VHDL
> '93 based) and suddenly it doesn't work in GHDL any more.

A check was missing, and I don't remember when it was fixed.

> But, the comment in the VHDL 2008 package you referenced indeed makes it
> sound like this is truly invalid VHDL. If that's the case, GHDL is
> right, but inconsistent with most VHDL implementations [...]

Maybe the behaviour of type conversion could be changed to follow other
simulator with -frelax.  Do you think it would be an improvement ?

Tristan.

wjl, 2016-08-04:
> Maybe the behaviour of type conversion could be changed to follow other simulator with -frelax. Do you think it would be an improvement ?

It could certainly help with similar code found in the wild, but personally I'm the only person I work with who even does things like define custom data types, so although I wish it worked the way I thought it did, I don't know how universal this issue is. It might just not be worth doing any work on it.

From my perspective, I now that I have learned this is actually illegal, I'm just going to update all my library code to work and be correct, since I'd rather have my stuff be as standards-compliant as possible (while still working on all implementations of interest).

diogratia, 2016-08-07:
Implicit subtype conversion has been added to type conversion in 2008. See IEEE Std 1076-2008 9.3.6 Type conversions (appended and annotated). ghdl does not (as yet) include implicit subtype conversion as specified therein.

Significantly the subtype of the target can be derived from the type when lacking a constraint as is the case here:
```vhdl
    use ieee.std_logic_1164.all;
    type STD_ULOGIC_VECTOR is array (NATURAL range <>) of STD_ULOGIC;
-- locally:
    -- A fixed point type
    type fixed_point is array (integer range <>) of std_ulogic;

   signal fp   : fixed_point(5 downto -3); -- 6 integer bits, 3 fractional, length=9
   signal sulv : std_ulogic_vector(8 downto 0); -- length=9

-- And the implicit subtype conversion:

   sulv <= std_ulogic_vector(fp);
```
The first sentence of paragraph 3, IEEE Std 1076-1993 7.3.5 Type conversions:

> The target type of a type conversion is the base type of the type mark.

Compared to -2008 9.3.6 paragraph 3:

> The target type of a type conversion is the base type of the type mark, and the target subtype of a type conversion is the type or subtype denoted by the type mark. 

See -2008 5.3.2 Array types, 5.3.2.1 General the BNF, the rules unbounded_array_definition and index_subtype_definition.

> index_subtype_definition ::= type_mark range <>

And paragraph 5:

>An unbounded array definition in which the element subtype indication denotes either an unconstrained composite subtype or a subtype that is not a composite subtype defines an array type and a name denoting that type. For each object that has the array type, the number of indices, the type and position of each index, and the subtype of the elements are as in the type definition. The index subtype for a given index position is, by definition, the subtype denoted by the type mark of the corresponding index subtype definition. The values of the left and right bounds of each index range are not defined, but shall belong to the corresponding index subtype; similarly, the direction of each index range is not defined. The symbol <> (called a box) in an index subtype definition stands for an undefined range (different objects of the type need not have the same bounds and direction).

For an unconstrained type in the type mark the target subtype is taken from the index type (the "_**type or subtype denoted by the type mark**_" in paragraph 3 of 9.3.6).

The index bound are converted to the index type of the target, see 9.3.6 paragraph 8, a) second sub paragraph:

>— If the index type of the operand and the index type of the target type are closely related, then the bounds of the index range of the result are obtained by converting the bounds of the index range of the operand to the index type of the target type, and the direction of the index range of the result is the direction of the index type of the operand. 

This represents the implicit subtype conversion for the assignment:

```vhdl
   sulv <= std_ulogic_vector(fp);
```
which should work in -2008. Implicit subtype conversion is not supported at all by the text of 7.3.5 in IEEE Std 1076-1993.

In the attached PDF of 9.3.6 changes from 7.3.5 are highlighted in yellow and orange, the orange indicating changes affecting the above implicit subtype conversion, and green highlighting for added comments.

[type_conversion_2008.pdf](https://github.com/tgingold/ghdl/files/405788/type_conversion_2008.pdf)
[type conversion 1993.pdf](https://github.com/tgingold/ghdl/files/405793/type.conversion.1993.pdf)

tgingold, 2016-08-25:
Diogratia, I don't get your point.  In the submitted case, the target  subtype is an array type (std_ulogic_vector) for which the index ranges are not defined.  The index type of the operand and the index type of the target are closely related (both are integer).  So the second sub-paragraph of a) applies.  I think we agree about that.

How does it apply ? The bound of the operand are 5 downto -3.  It is converted to the type of the target, keeping the same direction, so 5 downto -3.  But the bound of the result doesn't belong to the corresponding index subtype of the target type, hence the error.

So I would say that ghdl is correct here.

closed 2017-09-05 by tgingold