ghdl/ghdl #136: GHDL --xref-html - argument too long

Paebbels, 2016-08-05:
Hello,

I was trying out the `--xref-html` command. Unfortunately, I need to pass all files in one call, but the list of filenames exceeds the maximum command line parameter length.

Is there another way to pass circa 280 files?

Maybe, this command could run for every parsed or analyzed file if no file list is specified. So it's like combining xref with `--dir` for every found library.

```PoSh
ghdl.exe -a foo1.vhdl
ghdl.exe -a foo2.vhdl
ghdl.exe -a foo3.vhdl
ghdl.exe --xref-html
```

tgingold, 2016-08-26:
Possible ideas:
* allow response file to avoid command line limit.
* allow an entity name for --xref-html to collect all units in a design

Paebbels, 2016-08-26:
I think the command line limit is set by the command line itself, for PowerShell it's 16 KiB; for Bash 4 KiB?

Is is possible to have multiple top-levels?
(I was looking forward to map the complete PoC repo.)

tgingold, 2016-08-27:
On 26/08/16 12:48, Patrick Lehmann wrote:
> I think the command line limit is set by the command line itself, for
> PowerShell it's 16 KiB; for Bash 4 KiB?

I think on Linux it is at least 32 KB.

So, do you mean that 16KB is not enough for you ?

> Is is possible to have multiple top-levels?
> (I was looking forward to map the complete PoC repo.)

The arguments of --xref-html are files, not units.

Paebbels, 2016-08-29:
Currently, PoC's source directory contains 386 VHDL files, each filles with one entity/architecture or package/package body pair. Assembling all file names in one text file, results in a ~ 40 kiB file. It could be striped by 3 KiB (absolute path), but PoC will grow :).

One way to solve this is an argument file. Many tools support this in addition to command line arguments, especially when long command lines can be build. Each line is interpreted as an argument:

```
-a
--ieee=sysnopsys
--std=08
file1.vhdl
file2.vhdl
```

Usually, these tools allow no other command line arguments than the switch to enable the argument file itself and sometime a logfile, because if the argument file has an syntax error, the logfile option in the file might be ignored.

Example with a long switch:

    ghdl.exe --arg-file argumentfile

commit 74cf5244 (2016-08-30): ghdldrv: support response files.

Fixes #136
closed 2016-08-30 by tgingold