ghdl/ghdl #137: GHDL --dir

Paebbels, 2016-08-05:
Hello,

`ghdl.exe --dir` emits no output if all files were compiled into a user-named library. A user needs to specify that library again to get a list of compiled items.

I suggest to:
 * print an error message if the list is empty (`work-xx.cf` does not exist / work is empty), or
 * scan for all *.cf files and emit all compiled items for every library.

I think `--dir` is a debug command, to see what's already compiled. It might require many command line calls, (or a loop) to check every library.

```PoSh
ghdl.exe -a --work=poc foo1.vhdl
ghdl.exe -a --work=poc foo2.vhdl
ghdl.exe -a --work=work foo3.vhdl
ghdl.exe --dir
poc
  package foo1
  package body foo1
  entity foo2
  architecture rtl of foo2
work
  entity foo3
  architecture rtl of foo3  
```

Alternatives:
 * What about an `--all` option?
 * use `--work=poc  ` as a prefix, instead of indentation

```PoSh
ghdl.exe --dir --all
--work=poc package foo1
--work=poc package body foo1
--work=poc entity foo2
--work=poc architecture rtl of foo2
--work=work entity foo3
--work=work architecture rtl of foo3  
```

tgingold, 2016-08-26:
Libraries could be indeed searched on the path and displayed.

Enhancement.

m-kru, 2020-05-10:
Is this issue Windows related, or can it be fixed on Linux? I asked because of the `.exe` in question.

eine, 2020-05-10:
@m-kru, I think this is independent of the host OS. GHDL, by default, searches for library `work` only. It is possible to show the content of any other library with e.g. `ghdl --dir --work=poc`. However, a different command needs to be executed for each library. The proposal of @Paebbels is to add option `--all`, so that searching for libraries is done by GHDL, instead of the user.

BTW, this enhancement is similar to the error that is produced when some files are analyzed with one version of the standard, and elaborated with a different version. See the two suggested enhancements in https://github.com/ghdl/ghdl/issues/980#issue-507885991.

m-kru, 2020-05-10:
Ok, I have realized one more drawback. You have to specify the standard directly. If you specify wrong standard you get not output. I think it would be better to display the standard used for compilation, instead of asking an user which standard to show. @tgingold ?

eine, 2020-05-10:
> Ok, I have realized one more drawback. You have to specify the standard directly. If you specify wrong standard you get not output. I think it would be better to display the standard used for compilation, instead of asking an user which standard to show.

Yes, this is the point of the suggested enhancements in #980. Searching for multiple libraries and searching for multiple versions of the standard are complementary features, but not the same.

Note that `ghdl --dir work poc` is supported (https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=--dir#cmdoption-ghdl-dir). However, it seems not to work with [`--workdir`](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=--dir#cmdoption-ghdl-workdir):

```sh
root@433a10d23100:/src# mkdir wrkdir

root@433a10d23100:/src# ghdl -a --workdir=wrkdir --work=poc ent.vhd

root@433a10d23100:/src# ls wrkdir/
ent.o  poc-obj93.cf

root@433a10d23100:/src# ghdl --dir --workdir=wrkdir work poc
# Library work
# Directory: wrkdir/
*command line*:1:1:error: cannot find resource library "poc"
# Library poc
# Directory:

root@433a10d23100:/src# ghdl --dir --workdir=wrkdir 
# Library work
# Directory: wrkdir/  
  
root@433a10d23100:/src# cat wrkdir/poc-obj93.cf 
v 4
file "/src/" "ent.vhd" "b60f7888d0531f1b0b9adce24be3bae460668e68" "20200510132100.586":
  entity ent at 1( 0) + 0 on 11;
  architecture a of ent at 8( 117) + 0 on 12;
```

tgingold, 2020-05-10:
The plan to ease multi-version support is to put the version inside the library file and thus making one library file for all versions.

PR #2040 (2022-04-19): ghdldrv: add --all option to dir command
    Fix #137

Add `-all` option to `--dir` command so that :
```
$ ghdl -a --work=poc foo1.vhdl
$ ghdl -a --work=poc foo2.vhdl
$ ghdl -a --work=work foo3.vhdl
$ ghdl --dir --all
# Library poc
# Directory:
package foo1
package body foo1
entity foo2
architecture rtl of foo2
# Library work
# Directory:
entity foo3
architecture rtl of foo3  
```
commit 3b460210 (2022-04-19): ghdldrv: add --all option to dir command

    Fix #137
closed 2022-04-19 by tgingold