ghdl/ghdl #141: VHDL assertion interpreted as if it was PSL

vicencb, 2016-08-11:
Hello,
when analyzing this
```vhdl
entity e is end entity;
architecture a of e is
 signal s :boolean;
begin
 assert not s;
end architecture;
```
with
```sh
ghdl -a --std=08 bug.vhd
```
ghdl reports
```
bug.vhd:5:2: no clock for PSL directive
ghdl: compilation error
```
and it should not because the assertion is valid in VHDL.
It works fine with `--std=02`.

It has been tested with d0d04ba9b19e0f4a0db16c70db1b3452e16d6923
```
ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 6.1.1 20160707
 GCC back-end code generator
```

Regards,
  Vicente.

Paebbels, 2016-08-11:
That's strange, I thought it was fixed one or two months ago...

The problem is that VHDL and PSL asserts overlap and GHDL should analyze such asserts as pure VHDL asserts. But it recognizes it as PSL assert statements, which in turn need a PSL clock.

As a workaround, you can specify a PSL clock with a comment line.

Setting a PSL default clock for all `assert` statements:
`-- PSL default clock is (rising_edge(clk));`

tgingold, 2016-08-26:
Thank you for the report, fix will be pushed soon.

commit 313d7236 (2016-08-27): Also consider 'not' for non-psl assert statement.

Fixes #141
closed 2016-08-27 by tgingold