ghdl/ghdl #142: Cannot open std_logic_vector_1164-body.vhdl

andresdemski, 2016-08-16:
Hi,
I just intalled the llvm flavor of the ghdl simulator and i found the next error while i was trying to compile a project using ieee2008 std:

ghdl -m --std=08 --ieee=synopsys  hist_tb
analyze ../../src/ieee2008/std_logic_1164-body.vhdl
/usr/local/bin/ghdl1-llvm: cannot open ../../src/ieee2008/std_logic_1164-body.vhdl
ghdl: compilation error.

Regards,
Andres

Paebbels, 2016-08-16:
Hello,

by "installed" you mean installed from binary package or installed from fresh compilation?

Regards
    Patrick

andresdemski, 2016-08-16:
Hello, 
I compiled it following the steps described in BUILD.txt. 
I'm using linux Majaro (arch based).

Regards
Andres

Paebbels, 2016-08-16:
Please execute `ghdl` in verbose mode by adding `-v` and post the results. This should show all search directories, where GHDL looks for `std_logic_1164-body.vhdl`. The search directories are passed from `ghdl` to `ghdl1-llvm` with `-P`.

I'm a bit confused why your output from above shows ` ../../src/ieee2008/` as search directory, because normally it should be `../../lib/ghdl/v08/`. Moreover, `std_logic_1164-body.vhdl` is precompiled, why is you script compiling this file again? The original source files are stored in `../../lib/ghdl/src/ieee2008` so it looks like `lib/ghdl/` is missing in the path.

Regards
    Patrick

andresdemski, 2016-08-16:
Here is the output:
ghdl -m -v --std=08 --ieee=synopsys hist_tb
no object file for ../../src/ieee2008/std_logic_1164-body.vhdl
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/synopsys/ --work=ieee --workdir=/usr/local/lib/ghdl/v08/synopsys/ -c -o /usr/local/lib/ghdl/v08/synopsys/std_logic_1164-body.o ../../src/ieee2008/std_logic_1164-body.vhdl
/usr/local/bin/ghdl1-llvm: cannot open ../../src/ieee2008/std_logic_1164-body.vhdl
ghdl: compilation error

Yes, it's really strange. The first thing i did when this error appear was looking for these files and i found them where you said. 

/usr/local/lib/ghdl/src/ieee2008/std_logic_1164-body.vhdl

Could the relative path (../../) be the cause of the error? I don't know which is the workspace directory for saying the path is ok.

andresdemski, 2016-08-16:
And the compiled file is here: /usr/local/lib/ghdl/v08/ieee/std_logic_1164-body.o

andresdemski, 2016-08-16:
i added the PATH argument:
ghdl -m -v -P/usr/local/lib/ghdl/v08/ieee/ --std=08 --ieee=synopsys hist_tb 

I got the next output:
sp_ram.vhd:5:10: unit "std_logic_unsigned" not found in 'library "ieee"'
sp_ram.vhd:5:10:  (use --ieee=synopsys for non-standard synopsys packages)

Now the --ieee option is not working :(. 
Is other command for doing the same as -m? I tried with -a but generate the compiling order is a mess.

Paebbels, 2016-08-16:
A few comments to you posts:
- You can edit you posts (pencil icon) to add addition information to a previous post
- Please enclose single code lines in `-signs and code blocks with fenced code-blocks:

        ```Bash
        ghdl -m
        ```

If you review your GHDL compilation log you will find these major steps:
 1. `ghdl1-llvm` gets compiled
 2. `ghdl_llvm` gets compiled (it will be renamed to to `ghdl` while installing)
 3. The shipped libraries are pre-compiled for VHDL-87, 93, 2008:
     1. A set of `sed` commands extracts a version specific source file and saves it in a `v87/v93/v08` directory. The extension is changed to `.v87/.v93/.v08`.
     2. `ghdl1-llvm` compiles library `std` in bootstrap mode
     3. Library `ieee` is compiled
     4. Vendor flavors like `mentor` and `synopsys` get compiled (while reusing existing pre-compiled `ieee` packages.
 4. `make install` copies all files to your (default) `PREFIX`.

The command line option `--std=XX` selects the appropriate `vXX` directory from `/usr/local/lib/ghdl` and the `--ieee=` option selects the flavor (standard -> ieee, mentor -> mentor, synopsys -> synopsys)

The paths in `-P` should be relative to the binary directory.

I have never used `-m` for my self ... Is this a big project or can you rerun the analyze steps manually?

andresdemski, 2016-08-16:
Thanks you for your reply. 
First of all, sorry about the comments. I'm new with github issues editor.
 
The error only happens with the `-m` and `--gen-makefile` commands. Now I'm trying the `-c files -e unit` and i have no problems.

Paebbels, 2016-08-16:
Hello again,

I compiled one of my examples with [PoC](http://poc-library.readthedocs.io/) and then used `ghdl --gen-makefile > makefile` to generate a makefile for my example. Then I removed all local object files (`*.o`) and executed make. It reports:

    This file was not locally built (/home/paebbels/git/PoC/tb/common/my_config_GENERIC.vhdl)

But `ghdl -m` works and reanalyzes all files. So I think we need to wait for a comment from Tristan :).

andresdemski, 2016-08-16:
Hi!!!
I'm trying to find the error. It's strange that with one command it works and with the other it doesn't, it can't be a path or environment error. 
I installed the ghdl in other pc (Ubuntu 14.04) with the basic compilation (without gcc neither llvm) and the commands `-m` works with the same project without problems. I can't try with the llvm because the ubuntu repositories have the llvm 3.4 and the compilation needs the llvm 3.5.

Paebbels, 2016-08-16:
One hint: The compilation without gcc or llvm backend is mcode. This backend doesn't generate object files. It only updates these `*.cf` database files, wherein GHDL logs where to find VHDL primary and secondary units for the (final) run.

tgingold, 2016-08-26:
@andresdemski: do you have a test case so that I can investigate this issue ?

tgingold, 2016-08-26:
(I was not able to reproduce it on a simple design).

Brian-Drummond, 2016-08-27:
Is this related to the new "make ghdllib" step somehow?

tgingold, 2016-08-27:
I would say no, as packages were compiled.
Despite some attempts, I was not able to reproduce this issue :-(

closed 2016-09-23 by tgingold

xnorme, 2019-01-02:
This issue is still unresolved and reproducible, for instance, on PoC.
Suppose `poc.files` contains list of vhdl files:
```
cat poc.files | xargs ghdl -i --std=08 --work=PoC --workdir=PoC/PoC/08
ghdl --gen-makefile --std=08 --work=PoC --workdir=PoC/PoC/08 ocram_sp > Makefile
```

tgingold, 2019-01-02:
It would be nice to have a standalone reproducer.

Paebbels, 2019-01-02:
@xnorme Please open a new issue for your follow-up issue. This one is closed since 2 years. If you think it's related, please add a reference (e.g. link) to this issue.

While your example looks a little bit like IP core `PoC.mem.orcram.sp` from the PoC-Library, the PoC Library does not contain `*.files` files that can be easily parsed by `cat` and `xargs`.

So currently we cannot comprehend how your example looks like and how reproduce it.