ghdl/ghdl #143: Conversion of universal_real to real when performing physical units multiplication

kmtaylor, 2016-08-16:
Hi Tristan,

I've found and issue when trying to run the following testbench:
[universal_conversion.tar.gz](https://github.com/tgingold/ghdl/files/419690/universal_conversion.tar.gz)

No method is found for performing the multiplication between the universal_real (left) and the physical unit (time)

Without fully understanding the inner workings of conversion between universal_real and real, I can get it to work with the following patch:
[universal_conversion.txt](https://github.com/tgingold/ghdl/files/419689/universal_conversion.txt)

Maybe this is a bug?

Paebbels, 2016-08-16:
Hello,

please post the full error message.

The expression in `constant TEST_VAL : time := 1000.0/100 * 1 ns;` is evaluated from left to right, because both operators have the same precedence. You can rewrite your code to:

```VHDL
constant TEMP_VAL : real := 1000.0/100;
constant TEST_VAL : time := TEMP_VAL * 1 ns;
```

Now you can see that operator `/` is not defined for the signature `[real, integer return real]`. You need to divide the real value by a real value.

```VHDL
constant TEMP_VAL : real := 1000.0 / 100.0;
constant TEST_VAL : time := TEMP_VAL * 1 ns;
```

kmtaylor, 2016-08-16:
Thanks Patrick.

The exact error message is:
`ghdl -i "../testbench/universal_conversion_tb.vhd"`
`ghdl -m universal_conversion_tb`
`../testbench/universal_conversion_tb.vhd:11:44: no function declarations for 'operator "*"'`

I understand that the correct result can be achieved by dividing by 100.0 (real). This way we have (from left to right):

1. (real/real) * physical
2. real * physical
3. physical

With 100 as an integer, we have:

1. (real/integer) * physical
2. universal_real * physical
3. No "*" operator found

I'm not sure if the standard allows the latter case, but it seems intuitive to me that:
1000.0/100 * 1 ns;
and
1000.0/100.0 * 1 ns;
Should result in identical behaviour (as the would in C).

Thanks for your quick response.

Paebbels, 2016-08-16:
Hello,

here is an short example:

```VHDL
entity e is
end;

architecture a of e is
  constant a : real := 1000.0 / 100;
  constant b : time := 1000.0 / 100 * 10 ps;
begin
end;
```

```Bash
ghdl -a test.vhdl
.\test.vhdl:6:31: no function declarations for 'operator "/"'
.\test.vhdl:7:42: no function declarations for 'operator "*"'
```

This is what I expect to happen, because there is no real divide by integer operator define. The question is why GHDL reports no operator for multiply instead of divide in line 7. Maybe because `*` was the last operator in an internal list ...

Regards
    Patrick

kmtaylor, 2016-08-16:
Hmmm. Not quite I think.

The line:
`constant a : real := 1000.0/100;`

Will result in a search for an operator that has the return type "real" with the left operator "real" and the right operator "integer". This is not found in (std_standard) and the error is reported.

However when the two operations are placed on one line, the division operation matches the available function: return: universal_real, l_operator: real, r_operator: integer.

This is processed in sem_expr.adb within the function Sem_Operator().

It seems to me that universal_real * physical = physical should be available, although I don't actually see it in the LRM.

Paebbels, 2016-08-17:
Ah OK. I see your point. I think this is a question for Tristan :).

diogratia, 2016-08-18:
For the value expression in the constant declaration:

```vhdl
constant TEST_VAL : time := 1000.0/100 * 1 ns;
```
The left hand expression fits with
````vhdl
  -- function "/" (<anonymous>: <UNIVERSAL_REAL>;
  --               <anonymous>: <UNIVERSAL_INTEGER>) return <UNIVERSAL_REAL>;
```
(displayed via ghdl --disp-standard).

The type of the left operand of the "\*" is dependent on the left expression (1000.0/100).

There are two  possibilities for the "\*" operator:
```vhdl
  -- function "*" (<anonymous>: <integer>;
  --               <anonymous>: <time>) return <time>;
  -- function "*" (<anonymous>: <time>;
  --               <anonymous>: <real>) return <time>;
```
And it would be the second of these. Note this also requires an implicit type conversion from type universal_real to real.

Why this implicit type conversion doesn't work can be found in 7.3.5 Type conversions, the last paragraph:

>In certain cases, an implicit type conversion will be performed. An implicit conversion of an operand of type universal_integer to another integer type, or of an operand of type universal_real to another floating point type, can only be applied if the operand is either a numeric literal or an attribute, **or if the operand is an expression consisting of the division of a value of a physical type by a value of the same type**; such an operand is called a convertible universal operand. An implicit conversion of a convertible universal operand is applied if and only if the innermost complete context determines a unique (numeric) target type for the implicit conversion, and there is no legal interpretation of this context without this conversion. 

The left hand of the "*" operation operand, a universal_real does not qualify for implicit type conversion to type real. The implicit type conversion you want to make with your patch is specifically not allowed.

There's a type conversion that proves the issue is 7.3.5 blocking implicit type conversion:
```vhdl
constant TEST_VAL : time := real(1000.0/100) * 1 ns;
```
This works by explicit type conversion of the result of the first expression.  And about here we're clear why the blame is put on the "\*" operator. There is no "\*" operator with a signature [universal_real, time return time] predefined.

It's also worth noting:
```vhdl
constant TEST_VAL : time := 1000.0/real(100) * 1 ns;
```
works by returning a real. Explicit type conversion is possible between any numeric types (they're all closely related).

For a 7.3.5 implicit type conversion:
```vhdl
    constant TEST_VAL : time := 1000.0 ns/100 ns * 1 ns;
```
works. This  a legal implicit type conversion under 7.3.5 of the result of the left hand expression to type integer from type universal_integer where the left hand expression uses the predefined:
```vhdl
--  function "/"      (anonymous, anonymous: TIME) return universal_integer;
```
and we use:
```vhdl
  -- function "*" (<anonymous>: <integer>;
  --               <anonymous>: <time>) return <time>;
```
In summary it's a feature and not a bug.

kmtaylor, 2016-08-18:
Thanks for the detailed response Dio. I agree with your analysis. Although I do think it is somewhat counter-intuitive that VHDL doesn't provide:
` -- function "*" (<anonymous>: <time>; <anonymous>: <UNIVERSAL_REAL>) return <time>;`

But I suppose that's a VHDL shortcoming, and not GHDL's problem.

I know that at least Xilinx's XST provides this conversion, but it tends to be more permissive in general (not necessarily a good thing).

Maybe this could be enabled with -frelaxed-rules ?

If not, I'm happy just applying the patch here, since we have some packages that cause problems. I suppose that's the great thing about open source software.

tgingold, 2016-08-26:
I'd prefer not to change -frelaxed-rules here, as it may introduce incompatibilities.

kmtaylor, 2016-08-26:
Ok. Fair enough. Thanks again for such a brilliant piece of software.

closed 2016-08-26 by kmtaylor

tgingold, 2016-08-26:
And thank you for using it!

Tristan.