ghdl/ghdl #146: Error in `ghdl --help -r` output

pacalet, 2016-08-31:
```
$ ghdl --help -r
-r UNIT [ARCH] [OPTS]      Run UNIT
...
```
should probably be:
```
$ ghdl --help -r
-r [OPTS] UNIT [ARCH]      Run UNIT
...
```

tgingold, 2016-08-31:
Simulation options are after unit name.
OTOH, it is possible to have some options before in some cases, but that's almost useless.

I don't think I will change this help line.

pacalet, 2016-08-31:
I see. But when using the mcode backend one must pass elaboration options too. Example:

```
$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ghdl -r counter --std=08 --ieee=standard --work=counter_lib --workdir=counter_lib/lib -Pcounter_lib/lib
ghdl: cannot find entity or configuration counter
$ ghdl -r --std=08 --ieee=standard --work=counter_lib --workdir=counter_lib/lib -Pcounter_lib/lib counter
... (simulation OK) ...
```
Perhaps should there be a different help for mcode?

tgingold, 2016-08-31:
On 31/08/16 11:51, pacalet wrote:
> I see. But when using the mcode backend one must pass elaboration
> options too. Example:
>
> |$ ghdl --version GHDL 0.34dev (20151126) [Dunoon edition] Compiled with
> GNAT Version: 4.9.2 mcode code [...]

Like this ?

$ ./ghdl_llvm -h -r
-r UNIT [ARCH] [OPTS]      Run UNIT
Main options (try --options-help for details):
[...]

$ ./ghdl_mcode -h -r
-r,--elab-run [OPTS] UNIT [ARCH] [RUNOPTS]  Run UNIT
Main options (try --options-help for details):
[...]

pacalet, 2016-08-31:
Yep, that would be fine, I think.

tgingold, 2016-08-31:
On 31/08/16 15:27, pacalet wrote:
> Yep, that would be fine, I think.

So, the current output is fine !

pacalet, 2016-09-01:
Hmm. Sorry then, I must have done something wrong during the install. I have nothing like these executables installed. Forget this issue, I must read the install instructions once more. Thanks for your answer.

closed 2016-09-01 by pacalet

tgingold, 2016-09-01:
On 01/09/16 07:39, pacalet wrote:
> Hmm. Sorry then, I must have done something wrong during the install. I
> have nothing like these executables installed. Forget this issue, I must
> read the install instructions once more. Thanks for your answer.

ghdl_mcode and ghdl_llvm are what I build for development.

Still, the idea of supporting --work and --workdir for llvm/gcc is a 
good idea.