ghdl/ghdl #147: With llvm and gcc backends, `ghdl -r` does not behave as it should

pacalet, 2016-08-31:
```
$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

With llvm and gcc backends, `ghdl -r [OPTS] UNIT [ARCH]` imposes that `UNIT` is a base name (no `mylib/myunit`) and ignores the `--workdir` option. A consequence is that the simulation can be launched only from within the directory where the `UNIT` binary resides. For various reasons this is a bit annoying. Could it look for the binary in the directory specified by `--workdir`, instead?

tgingold, 2016-08-31:
Makes sense.

tgingold, 2016-09-23:
Wait.  `ghdl -e UNIT` always create the unit in the current directory.  So changing `-r` would make the flow incompatible.  Do you suggest that `-e` should write the executable in the workdir ?  I am not sure about changing that.

I will clarify the documentation however.

tgingold, 2016-09-23:
There was a bug however: when --workdir was present --remove/--clean didn't correctly delete the elaboration files.  Being fixed.

closed 2016-10-11 by tgingold