ghdl/ghdl #148: Wrong dynamic type range checking?

pacalet, 2016-08-31:
```
$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
When simulating `bug_sim`:
```
entity bug is
	port(data: out integer);
end entity bug;

architecture arc of bug is
begin
end architecture arc;

entity bug_sim is
end entity bug_sim;

architecture sim of bug_sim is
	signal data: natural;
begin
	u0: entity work.bug
		port map(data => data);
end architecture sim;
```
`ghdl` (`llvm` backend) does not issue an error. Shouldn't it? The initialization of the `data` output port of `bug` (`integer`) should force it to `-2^{31}+1` and this should probably cause a range checking violation on signal `data` of architecture `bug_sim` (`natural`).

Paebbels, 2016-08-31:
Hello,
my local machine has currently no GHDL on it, so I can run your test...
What happens if you set an init value on the integer output (integer'low)?
Does this cause a range error?

Regards
    Patrick

pacalet, 2016-08-31:
> Hello,

Hello Patrick,

> my local machine has currently no GHDL on it, so I can run your test...
> What happens if you set an init value on the integer output (integer'low)?
> Does this cause a range error?

Nope. By the way, I did some more tests with a commercial simulator
(Mentor Graphics Modelsim). In both cases (with and without init value)
it issues a range violation error.

Best regards,

Renaud.

> Regards
> Patrick
> 
> <https://github.com/tgingold/ghdl/issues/148#issuecomment-243710354>, or
> mute the thread

diogratia, 2016-09-13:
IEEE Std 1076-2008 6.4.2.3 Signal declarations paras  5 & 6:
(IEEE Std 1076-1993 4.3.1.2 Signal declarations 6 & 7)

> If the signal declaration includes the assignment symbol followed by an expression, it shall be of the same type as the signal. Such an expression is said to be a default expression. The default expression defines a default value associated with the signal or, for a composite signal, with each scalar subelement thereof. For a signal declared to be of a scalar subtype, the value of the default expression is the default value of the signal. For a signal declared to be of a composite subtype, each scalar subelement of the value of the default expression is the default value of the corresponding subelement of the signal.
> 
> In the absence of an explicit default expression, an implicit default value is assumed for a signal of a scalar subtype or for each scalar subelement of a composite signal, each of which is itself a signal of a scalar subtype. The implicit default value for a signal of a scalar subtype T is defined to be that given by T'LEFT.

para 11 (12, -1993):

>The default value associated with a scalar signal defines the value component of a transaction that is the initial contents of each driver (if any) of that signal. The time component of the transaction is not defined, but the transaction is understood to have already occurred by the start of simulation.

14.3.4 Port clause (12.2.3 The port clause -1993)

>Elaboration of a port clause consists of the elaboration of each of the equivalent single port declarations contained in the clause, in the order given. The elaboration of a port declaration establishes that the port can subsequently be referenced.

14.3.5 Port map aspect para 6 ( 12.2.4 The port map aspect -1993)

>Similarly, if an actual signal is associated with a port of mode out, inout, or buffer, and if the type of the actual is a scalar type, then it is an error if (after applying any conversion function or type conversion expression present in the formal part) the subtype of the formal is not compatible with the subtype of the actual.

So this is not a run time check, it's done at elaboration time. The subtype of the formal is integer which is not a proper or perfect subset of the subtype of the actual which is natural.

The requirement is that the formal's set of values must be expressible in the set of values defined by the subtype of the actual.

This represents a missing semantic check of the rule of 14.3.5/12.2.4 para 6. (Which is a distinct semantic rule from that found in object declaration  14.4.2.5/12.3.1.4).

tgingold, 2016-09-23:
Yes, I agree this is a bug that should be reported during elaboration.

closed 2016-10-11 by tgingold