ghdl/ghdl #149: GHDL with GCC backend as Docker container

RafaelCatrou, 2016-09-03:
Hello,

As to build GHDL with GCC backend seems a bit complicated (compared to the other backends), I have done a [Docker ](http://www.docker.com/) container which is available on [Docker Hub](https://hub.docker.com/r/rafaelcatrou/docker_ghdl/)) so that anybody could easily deploy and use GHDL with GCC backend.

Nevertheless, even following the BUILD.txt of GHDL, I am not sure about the exact recipe for build automation. Thus I need a review and feedback on my script. It is available on my GitHub in the files:
- [Dockerfile](https://github.com/RafaelCatrou/docker_ghdl) : dockerfile for GHDL building with GCC backend. It relies on this other [dockerfile](https://github.com/RafaelCatrou/docker_gnat) for GNAT installation. 
- [src/build_ghdl_with_gcc.sh](https://github.com/RafaelCatrou/docker_ghdl/tree/master/src) : script that build GHDL with GCC (called by the dockerfile above)

Or else, I would appreciate if anyone could provide me the exhaustive list of commands to build GHDL with GCC.

Current status: as far as I used this container, it works (but for now I did only very-basic VHDL stuff). That's why I prefer to consolidate this container before going further.

Thanks for help,
Rafael

tgingold, 2016-09-03:
On 03/09/16 09:24, Rafael Catrou wrote:
> Hello,
>
> As to build GHDL with GCC backend seems a bit complicated (compared to
> the other backends), I have done a Docker <http://www.docker.com/>
> container which is available on Docker Hub [...]

Why do you configure ghdl and run make copy-sources twice ?

You can also try to run the testsuite to check whether ghdl was 
correctly built.

Tristan.

RafaelCatrou, 2016-09-03:
Hi Tristan,

For the 'configure ghdl + make copy-sources', if I don't do twice (but with different argument for --prefix) the build fails. I should have missed something in there but I have to admit that a have great room for improvement in the understanding of 'configure/make/make install' and mostly linux stuff (including GCC re-compilation). That's why I am interested in getting an already existing recipe for building GHDL with GCC backend from a fresh install of ubuntu 16.04.

Thanks for the testsuite hint, I add the [testsuite.log](https://github.com/RafaelCatrou/docker_ghdl/blob/master/testsuite.log) in my repository so that we could investigate if needed.

tgingold, 2016-09-03:
On 03/09/16 18:00, Rafael Catrou wrote:
> Hi Tristan,
>
> For the 'configure ghdl + make copy-sources', if I don't do twice (but
> with different argument for --prefix) the build fails. I should have
> missed something in there but I have to admit [...]

In the first configure, you should use --prefix=${PREFIX_FINAL_GCC_PATH}
Then remove the second one (as well as the make copy-sources).

If that doesn't work, please post the build log.

You should also remove:

cp -r ${DOWNLOAD_PATH}/ghdl/lib/ghdl/* ${PREFIX_FINAL_GCC_PATH}/lib/ghdl

> Thanks for the testsuite hint, I add the testsuite.log
> <https://github.com/RafaelCatrou/docker_ghdl/blob/master/testsuite.log>
> in my repository so that we could investigate if needed.

It is a way too big file.  Look at the last line: the testsuite stops in 
case of failure.

RafaelCatrou, 2016-09-04:
Hello Tristan,

I update the script [build_ghdl_with_gcc.sh](https://github.com/RafaelCatrou/docker_ghdl/blob/master/src/build_ghdl_with_gcc.sh) with your feedback and it works like a charm.

I also add the run of testsuite at the end of the [Dockerfile](https://github.com/RafaelCatrou/docker_ghdl/blob/master/Dockerfile). The results end with (it seems good):
`*** End of tests delete library: project /root/tools/ghdl_final/bin/ghdl --remove --std=93c --work=project delete library: random /root/tools/ghdl_final/bin/ghdl --remove --std=93c --work=random delete library: utilities /root/tools/ghdl_final/bin/ghdl --remove --std=93c --work=utilities Vests tests successful ./testsuite.sh: Success (/root/tools/ghdl_final/bin/ghdl) GHDL 0.34dev (20151126) [Dunoon edition] Compiled with GNAT Version: GPL 2016 (20160515-49) GCC back-end code generator Written by Tristan Gingold. Copyright (C) 2003 - 2015 Tristan Gingold. GHDL is free software, covered by the GNU General Public License. There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Thanks for your help, it will easier GHDL deployment.

closed 2016-09-04 by RafaelCatrou

1138-4EB, 2017-12-15:
Hi @RafaelCatrou , I am trying to reproduce your work using `debian:stretch-slim` as a base image. I am concerned with these separate compilations of gmp, mpfr and mpc: [build_ghdl_with_gcc.sh#L64-L125](https://github.com/RafaelCatrou/docker_ghdl/blob/master/src/build_ghdl_with_gcc.sh#L64-L125). According to the documentation ([GCC Backend on GNU/Linux with GCC/GNAT](http://ghdl-devfork.readthedocs.io/en/doc-gcc/building/gcc/GNULinux-GNAT.html)), running `download_prerequisites` is enough, which you already do before those compilations: [build_ghdl_with_gcc.sh#L47](https://github.com/RafaelCatrou/docker_ghdl/blob/master/src/build_ghdl_with_gcc.sh#L47).

Indeed, I tried [this](https://gist.github.com/1138-4EB/acb5a3d14c303527ee1e5b309bffe181) gist, and it passes the testsuite without errors. Could you please clarify why you compiled those dependencies explicitly?