ghdl/ghdl #150: concatenation fails with "bound check failure" during run-time

pawel256, 2016-09-07:
example:
-- concat.vhd START
library ieee; 
use ieee.std_logic_1164.all;
      
entity concat is
   port (
      a : in  std_ulogic_vector(3 downto 0);
      b : in  std_ulogic_vector(3 downto 0);
      q1 : out std_ulogic_vector(7 downto 0);
      q2 : out std_ulogic_vector(7 downto 0);
      q3 : out std_ulogic_vector(7 downto 0)
   );
end concat;
     
architecture rtl of concat is
begin
   as_q1: q1 <= "0000" & b;
   as_q2: q2 <= a & "0000";
   as_q3: q3 <= a & b;
end rtl; 
-- concat.vhd END

concat.vhd passes build with no errors/warnings
but, during run time, concatenation in q2 assignment fails:

$ ./concat --trace-processes
run process .concat(rtl).as_q1 [00000000194EB700]
run process .concat(rtl).as_q2 [00000000194EB700]
./concat:error: bound check failure at concat.vhdl:18
./concat:error: simulation failed

if q2 assignment is commented out, then q3 fails:

$ ./concat --trace-processes
run process .concat(rtl).as_q1 [0000000003E50700]
run process .concat(rtl).as_q3 [0000000003E50700]
./concat:error: bound check failure at concat.vhdl:19
./concat:error: simulation failed

Looks like in some concatenations the result size is not calculated correctly during run time?

tmeissner, 2016-09-07:
I can't reproduce this bug with the latest version of GHDL:

```
$ ./concat --trace-processes
run process .concat(rtl).as_q1 [00007F83C1C03D70]
run process .concat(rtl).as_q2 [00007F83C1C03D70]
run process .concat(rtl).as_q3 [00007F83C1C03D70]
Now is 0ms +0
Now is 9223372036854775807fs +0

TMs-Air:Temp torsten$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Hingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Build from the HEAD of the master branch.

pawel256, 2016-09-08:
Thank you for quick response

I have recompiled using ghdl-master.zip (downloaded today)
$ ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.4.6 
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

unfortunately the issue is still there
(this is the output with -g added in analysis/elaboration)

$ ./concat --trace-processes
run process .concat(rtl).as_q1 [000000001BDA2D30]
run process .concat(rtl).as_q2 [000000001BDA2D30]
./concat:error: bound check failure at concat.vhdl:18
  from: process work.concat(rtl).as_q2 at concat.vhdl:18
./concat:error: simulation failed

I see that you are using LLVM ("llvm code generator")
while all three versions I tried (0.29 binaries, 0.33 & 0.34 built by myself)
are using GCC ("GCC back-end code generator")

Could the issue only manifest itself with GCC back-end code?

Brian-Drummond, 2016-09-08:
I have just tried it here with a gcc backend ... no issue.

ghdl -v
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

One comment on your build ... "Gnat version 4.4.6"?? 
That looks like a very old version of Gnat, I'm a little surprised it built at all. Gnat version isn't hyper-critical so I'd expect anything from 4.6 or 4.7 onwards to work, but I've never tried one that old

Which gcc source revision did you use?

I'd suggest building with a newer Gnat, perhaps from getadanow.com if you need to build from source - or install from a package if there's one available for your OS.

pawel256, 2016-09-09:
Thanks for the hint

I've tried to compile with a never version of GNAT, then with another one - no difference

so I dig into the code translation - and in the process found that the generated code depends on the VHDL std option
sorry, I have not mentioned that I am using "--std=87"
(now I see that --std=93 generates different range checks for concatenation -> so boundary check does not fail)

in **trans-chap7.adb**, function **Translate_Concatenation**
fragment (for VHDL 87 std, dir == downto):

```
New_Assign_Stmt
               (M2Lv (Chap3.Range_To_Length (Get_Res_Range)),
                New_Obj_Value (Var_Length));
[...]
New_Assign_Stmt
                 (New_Obj (Var_Right),
                  New_Dyadic_Op (ON_Sub_Ov,
                                 M2E (Chap3.Range_To_Left (Get_Res_Range)),
                                 New_Convert_Ov (New_Obj_Value (Var_Length1),
                                                 Idx_Otype)));

[...]
Chap3.Check_Range (Var_Right, Null_Iir, Index_Type, Left);
```

generates following code
(from concat.vhdl.004t.gimple dumped with -fdump-tree-all option)
comments in bold

```
T0_2.BOUNDS = &T0_0;
T0_3 = 8;
T0_0.dim_1.length = T0_3;
T0_0.dim_1.left = 3;
T0_0.dim_1.dir = 1;
...
T1_0 = T0_3 + 4294967295; // this is T0_3 - 1 **// Var_Length1 := Var_Length -1 = 7**
...
D.1693 = T0_0.dim_1.left; **// Chap3.Range_To_Left (Get_Res_Range) = 3**
T1_0.4 = (std__standard__integer__BT) T1_0; **// = 7**
T1_1 = D.1693 - T1_0.4; // = 3 - 7 = -4
if (T1_1 < 0) goto <D.1695>; else goto <D.1696>; **// -4 < 0 => goto <D.1695>**

<D.1695>:
__ghdl_bound_check_failed_l1 (&_UI00000000, 18);
```

so:
the length of result - 1 (Var_Length1) is subtracted from the left boundary of the first operand (a which is (3 downto 0))
3 - (8-1) = 3 - 7 = -4
and that is compared against 0 => check fails

thanks again, and sorry for missing the important part of information

tgingold, 2016-09-11:
Hello,

> I've tried to compile with a never version of GNAT, then with another
> one - no difference
>
> so I dig into the code translation - and in the process found that
> the generated code depends on the VHDL [...]

Ah, that's the most important point.

So GHDL uses the v87 standard to compute the range of the
result, which results in a bound violation.
Vhdl 87 concatenation rules were slightly broken and fixed
in vhdl 93.

So why are you using --std=87 ?

Regards,
Tristan.

pawel256, 2016-09-12:
Thanks Tristan,

As you said: this is VHDL-87 standard "feature"
and GHDL is just following the standard

> So why are you using --std=87 ?

because the files were supposed to be written in VHDL-87 ... (I would spare you the details ...)

Great thanks to all the people who helped and all GHDL designers

Regards,
Pawel

closed 2016-09-12 by pawel256

tgingold, 2016-09-12:
On 12/09/16 16:06, Pawel wrote:
> Thanks Tristan,
>
> As you said: this is VHDL-87 standard "feature"
> and GHDL is just following the standard
>
>     So why are you using --std=87 ?
>
> because the files were supposed to be written in VHDL-87 ... (I would
> spare you the details ...)

Then you can use --std=93c, which is compatible with VHDL-87 but
use the vhdl 93 rules.

> Great thanks to all the people who helped and all GHDL designers

Thank you for using ghdl!

Tristan.