ghdl/ghdl #151: Why is the length of an unconstrained input port not considered static?

nathanaelhuffman, 2016-09-21:
The following test code issues an error during analysis with GHDL:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
 port (in_vec   : in std_logic_vector);
end entity;

architecture rtl of test is
signal A : natural;
begin

  gen2 : if in_vec'length <= 2 generate
     A <= 2;
  end generate;
  gen3 : if in_vec'length > 2 generate
     A <= 3;
  end generate;
end architecture;```

The errors issued:
'condition must be a static expression' 2x, one for each of the generates statements.

This occurs with the --std=08 flags and without, and with the -frelaxed-rules flag as an additional test.
 
I think 2008 LRM section 9.4.3 deals with this, though I admit I got a little lost in there somewhere.

My understanding also agrees with VHDL 2008- Just the New Stuff by Ashenden and Lewis  (Section 9.8 on Static Ranges ):

"VHDL-2008  addresses  this  and  similar  problems  by  recognizing  that  it  is  not  the
index range of the declared subtype of d that we are interested in. Rather, it is the index
range of d itself, which is determined at elaboration time from the actual signal associated with the port. VHDL-2008 revises the rules for globally static expressions and ranges
to allow use of attributes that provide information about the range of a prefix, provided
the prefix is one of the following:
• a signal or port
• a constant or generic constant
• a type or subtype
• a globally static function call• a variable that is not of an access type, or a variable of an access type whose desig-
nated subtype is fully constrained
For  these  prefixes,  we  can  use  attributes  such  as  'range, 'left, 'right, 'length,  and
'ascending, since they provide information about the range of the prefix that is deter-
mined at elaboration time."

Paebbels, 2016-09-21:
Hello, what's your GHDL version? I think that issue was fixed is a similar case for for-loops.

nathanaelhuffman, 2016-09-21:
Very latest tag 2016-09-14 from 6 days ago.

I actually built that version specifically to see if this was fixed and it was not.  I can build master if needed to confirm.

I think the issue is actually distinct because this is related to unconstrained ports and this syntax that is now valid in 2008 which was not before (even though most simulators would let it through anyway)

tgingold, 2016-09-21:
On 21/09/16 02:40, Nathanael Huffman wrote:
> The following test code issues an error during analysis with GHDL:
>
> library ieee;
> use ieee.std_logic_1164.all;
> use ieee.numeric_std.all;
>
> entity test is
>  port (in_vec   : in std_logic_vector);
> end entity;
>
> architecture rtl of test is
> signal A : natural;
> begin
>
>   gen2 : if in_vec'length <= 2 generate
>      A <= 2;
>   end generate;
>   gen3 : if in_vec'length > 2 generate
>      A <= 3;
>   end generate;
> end architecture;```
>
> The errors issued:
> 'condition must be a static expression' 2x, one for each of the generates statements.
>
> This occurs with the --std=08 flags and without, and with the -frelaxed-rules flag as an
> additional test.

I agree that this should be allowed by -frelaxed-rules.

> I think 2008 LRM section 9.4.3 deals with this, though I admit I got a little lost in there somewhere.

Eh, don't you find the LRM clear and understandable ?  I will try to 
decypher that for you!

Tristan.

nathanaelhuffman, 2016-09-21:
This is not mostly academic since you agreed that -frelax-rules could apply here,, but since I'm this far into it I want to make sure I understand this.
Based on my understanding, the -frelaxed-rules is a nice thing for pre'08 standard flows, but I think the above code is explicitly allowed by the 2008 LRM, which would mean that it there aren't rules to relax for the 2008 case.

I did more LRM reading, and this is certainly a bit out of my area of expertise, but from section 5.3.2.2 (VHDL2008):
The index range for each index of an array object or array subelement of a composite object is determined as follows:
...
e) For an interface object of an array type, or a subelement of an interface object for which the subele-
ment type is an array type, each index range is obtained as follows: Let the subtype index range be 
the corresponding index range of the subtype indication of the declaration of the object.
  1) ... (not our case)
  2) ... ( I think not our case)
  3) If the subtype index range is undefined, and the interface object is associated in whole (see 
6.5.7.1) or is a subelement that is associated individually by a single association element other 
than one in which the formal designator is a slice name, then the index range of the object is 
obtained from the association element in the following manner ( I think this is our case)

tgingold, 2016-09-22:
On 21/09/16 16:43, Nathanael Huffman wrote:
> This is not mostly academic since you agreed that -frelax-rules could
> apply here,, but since I'm this far into it I want to make sure I
> understand this.
> Based on my understanding, the [...]

That's correct, but the issue is not about indexes of objects, but about
staticness!

Tristan.

tgingold, 2016-09-22:
On 21/09/16 02:40, Nathanael Huffman wrote:
> The following test code issues an error during analysis with GHDL:
>
> library ieee;
> use ieee.std_logic_1164.all;
> use ieee.numeric_std.all;
>
> entity test is
>  port (in_vec   : in std_logic_vector);
> end entity;
>
> architecture rtl of test is
> signal A : natural;
> begin
>
>   gen2 : if in_vec'length <= 2 generate
>      A <= 2;
>   end generate;
>   gen3 : if in_vec'length > 2 generate
>      A <= 3;
>   end generate;
> end architecture;```
>
> The errors issued:
> 'condition must be a static expression' 2x, one for each of the generates statements.
>
> This occurs with the --std=08 flags and without, and with the -frelaxed-rules flag as an
> additional test.
>
> I think 2008 LRM section 9.4.3 deals with this, though I admit I got a little lost in there somewhere.

I think that 9.4.3 l) applies:

l) A predefined attribute that is a function, other than the predefined 
attributes 'EVENT, 'ACTIVE, 'LAST_EVENT, 'LAST_ACTIVE, 'LAST_VALUE, 
'DRIVING, and 'DRIVING_VALUE, whose prefix is appropriate for a globally 
static attribute, and whose actual parameter (if any) is a globally 
static expression

As 'length for an array is a function.  The prefix is in_vec.
But:

A prefix is appropriate for a globally static attribute if it denotes a 
signal, a constant, a type or subtype, a globally static function call, 
a variable that is not of an access type, or a variable of an access 
type whose designated subtype is fully constrained.

It is not clear whether the prefix in_vec denotes a signal: an interface 
signal is not a signal.

Tristan.

diogratia, 2016-09-23:
>> It is not clear whether the prefix in_vec denotes a signal: an interface signal is not a signal.

IEEE Std 1076-2008:

 6.5.6.3 Port clauses:
>
Ports provide channels for dynamic communication between a block and its environment. The following applies to both external blocks defined by design entities and to internal blocks defined by block statements, including those equivalent to component instantiation statements and generate statements (see 11.8).
>
port_clause ::=  
&nbsp;&nbsp;&nbsp;&nbsp;**port** ( port_list ) ;  
port_list ::= port_interface_list  
>
**The ports of a block are defined by a port interface list. Each interface element in the port interface list declares a formal port.**
>
To communicate with other blocks, the ports of a block can be associated with signals in the environment in which the block is used. Moreover, the ports of a block may be associated with an expression in order to provide these ports with constant driving values or with values derived from signals and other ports; such ports shall be of mode in. **A port is itself a signal (see 6.4.2.3);** thus, a formal port of a block may be associated as an actual with a formal port of an inner block. The port, signal, or expression associated with a given formal port is called the actual corresponding to the formal port (see 6.5.7). The actual, if a port or signal, shall be denoted by a static name (see 8.1).

**emphasis added**.

nathanaelhuffman, 2016-09-28:
So do we have a final ruling on this?  Given LRM-2008 9.4.3 and 6.4.2.3/6.5.6.3 it would seem that this should be supported natively in VHDL 2008 correct?

tgingold, 2016-09-28:
Yes, that looks correct.

----- Mail original -----

closed 2016-10-11 by tgingold