ghdl/ghdl #152: ghdl 0.33 crashes on VHPI

stdefeber, 2016-09-22:
I want to create a link between my simulation via a message queue and some yet to be defined application. See code below.

$> ghdl --version 

GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: 5.3.0

However ghdl crashes with : 

ax_wb_pli.vhdl:8:0: internal compiler error: Segmentation fault
 entity ax_wb_pli is
 ^
0x96e70f crash_signal
        ../../gcc-4.9.3/gcc/toplev.c:337
0x966dd1 place_field(record_layout_info_s*, tree_node*)
        ../../gcc-4.9.3/gcc/stor-layout.c:1098
0x96806a layout_type(tree_node*)
        ../../gcc-4.9.3/gcc/stor-layout.c:2287
0x59b9b1 finish_record_type
        ../../gcc-4.9.3/gcc/vhdl/ortho-lang.c:1140
0x6306c6 trans__chap10__pop_instance_factory
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans.adb:293
0x6488b7 trans__chap8__translate_procedure_call_state
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans-chap8.adb:2370
0x65505f trans__chap4__translate_statements_chain_state_declaration
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans-chap4.adb:2162
0x63dcd0 trans__chap9__translate_process_declarations
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans-chap9.adb:253
0x63dcd0 trans__chap9__translate_block_declarations
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans-chap9.adb:817
0x666b98 trans__chap1__translate_architecture_body
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/trans-chap1.adb:221
0x6371bb translation__translate
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/translation.adb:298
0x6688d6 ortho_front__parse
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/ortho_front.adb:407
0x5ba58e lang_parse_file
        /opt/ghdl_build/gcc-4.9.3/gcc/vhdl/ortho_gcc.adb:114
0x5979bf ortho_parse_file
        ../../gcc-4.9.3/gcc/vhdl/ortho-lang.c:496
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
ghdl: exec error

Code ax_wb_pli_pkg.vhdl :
---------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

package ax_wb_pli_pkg is

  type bus_t is access integer;

  function init_queue(queue_id : integer) return integer;
    attribute foreign of init_queue : function is "VHPIDIRECT sim_init_queue";

  procedure delete_queue(queue_id : integer);
    attribute foreign of delete_queue : procedure is "VHPIDIRECT sim_delete_queue";

end package;

package body ax_wb_pli_pkg is

  function init_queue(queue_id : integer) return integer is
  begin
    assert false report "VHPI" severity failure;
  end function;

  procedure delete_queue(queue_id : integer) is
  begin
    assert false report "VHPI" severity failure;
  end procedure;

end package body;
---------------------------------------------------------------------------
End of Code ax_wb_pli_pkg.vhdl

---------------------------------------------------------------------------

Code of ax_wb_pli.vhdl
---------------------------------------------------------------------------

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

library work;
use     work.ax_wb_pli_pkg.all;

entity ax_wb_pli is
end entity ax_wb_pli;

architecture behav of ax_wb_pli is

  constant cQUEUE_ID : integer := 0;
  shared variable queue_handle : integer;

begin

  
  init_bus_handle : process 
  begin
    queue_handle := init_queue(cQUEUE_ID);
    if (bus_handle = 0) then
      assert false report "Failed to register Message Queue";
    end if;  
    wait for 1 us;
    queue_delete(cQUEUE_ID);
  end process;

end behav;

---------------------------------------------------------------------------
End of Code of ax_wb_pli.vhdl
---------------------------------------------------------------------------
Code of ax_wb_pli.c
---------------------------------------------------------------------------
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <sys/types.h>
#include <sys/ipc.h>
#include <sys/msg.h>

int sim_init_queue (int qid)
{
  key_t  key;
  int    msgqueue_id;
  //struct mymsgbuf qbuf;

  /* Create unique key via call to ftok() */
  key = ftok(".", 'm');

  /* Open the queue - create if necessary */
  if((msgqueue_id = msgget(key, IPC_CREAT|0660)) == -1) {
          perror("msgget");
          exit(1);
  }
  return msgqueue_id;

}

int sim_delete_queue(int qid)
{

  /* Remove the queue */
  msgctl(qid, IPC_RMID, 0);

  return 0;

}

int main ()
{

  return 0;          

}
---------------------------------------------------------------------------
End of Code of ax_wb_pli.c
---------------------------------------------------------------------------

Build script :

gcc -Wall -fPIC ax_wb_pli.c -o ax_wb_pli.o
ghdl -i --work=work ax_wb_pli.vhdl ax_wb_pli_pkg.vhdl
ghdl -m -Wl,ax_wb_pli.o --work=work ax_wb_pli

Paebbels, 2016-09-22:
Hello,
your issue report contains lots of source code. Please format the code with [fenced code blocks](https://help.github.com/articles/creating-and-highlighting-code-blocks/#syntax-highlighting).
Thanks.

stdefeber, 2016-09-22:
OK. Done.

I have tried the same code on a debian machine (8.5). 
I get :
```
. ./build.sh 
analyze ax_wb_pli_pkg.vhdl
analyze ax_wb_pli.vhdl

******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: 4.9.2
In directory: /home/axign/projects/hslink/
Command line:
/usr/bin/ghdl1-llvm --work=work -P/usr/lib/ghdl/v93/std/ -P/usr/lib/ghdl/v93/ieee/ -c -o ax_wb_pli.o ax_wb_pli.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: ortho_llvm.adb:2911 access check failed
Call stack traceback locations:
0x80a9298 0x80a1a04 0x8164309 0x818fc71 0x81a4b22 0x817b260 0x817e634 0x81c417e 0x816f43e 0x81c8ead 0x80aad75 0x8075c73 0xb7062a61 0x807511d
******************************************************************
ghdl: compilation error
```
```
ghdl --version
GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
```

tgingold, 2016-09-23:
As is, the vhdl code doesn't analyse:
ax_wb_pli.vhdl:22:9: no declaration for "bus_handle"
ax_wb_pli.vhdl:26:5: no declaration for "queue_delete"

Once fixed, I remark you have an issue on filenames: both ax_wb_pli.c and ax_wb_pli.vhdl will creates ax_wb_pli.o, so you need to rename one of both.

You should also remove the main() function in the C file, otherwise that will be the main of the simulation.

The design can be analysed and elaborated with the development version of ghdl, so the crash has been fixed after the 0.33 release.

Thank you for the report,
Tristan.

closed 2016-09-23 by tgingold

diogratia, 2016-09-23:
And using Tristan's modifications the design analyzes and elaborates on OS X with llvm. There is no timeout mechanism in process init_bus_handle. When run it appears it will go on until simulation time hits time'HIGH.

Adding a timeout to the process and the simulation completes.

```
david_koontz@Macbook: ax_wb_pli
ax_wb_pli.vhdl:29:9:@11us:(report note): timeout

GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)

Commit:
2566852c5601fd03f4a9c6b5939d30a56d919c5d [2566852]
Parents:
d6cb5a78af
Author:
Tristan Gingold <tgingold@free.fr>
Date:
30 August 2016 3:00:02 pm NZST
Commit Date:
30 August 2016 3:12:39 pm NZST

```

stdefeber, 2016-09-23:
Correct. Initially I wanted to see whether I can open close message queues
via my test-bench.
So it is in its preliminary state.

stdefeber, 2016-09-23:
Btw, it is much appreciated that you have tried my code.

Bg,

ST