ghdl/ghdl #154: Feature Request: Step Model from external process

sleary78, 2016-09-23:
Hi, 

Appreciate that this might not be something you want to do but i'd really like to be able to manipulate a GHDL compiled object from an external program. Currently I can execute ghdl_main but this doesnt give me the ability to evaluate the signals. Similar to how verilator works (infact i'd like to simulate verilator models simultaneously with GHDL models). 

Apologies if this is utterly not of interest.

tgingold, 2016-09-24:
No, that's useful.
Do you need to inspect signals after a delta cycle or after a time cycle ?

sleary78, 2016-09-24:
That would be ideal. I imagine an API might look something like. 

ghdl_init(); always required.
ghdl_setsignal(const char* signame, int value); // unsure how to do > 64 bits but its a start.
int ghdl_getsignal(const char* signame);
ghdl_eval();
ghdl_destroy()

Ideally i'd like to be able to inject my clock externally via a setsignal. This would let me eval a verilator module and a ghdl module and then manually marshall the connected signals in code.

tgingold, 2016-09-28:
Note that setting and reading signals can be done with VHPI or VPI.

1138-4EB, 2017-03-02:
@sleary78 , did you try either VHPI or VPI? If so, did it fit your needs?

terriblefire, 2017-03-02:
(i am also sleary78)... no i want to mix VHDL and Verilator stuff in a C++ project... its just a fantasy that i can make ghdl do what verilator does. :)

1138-4EB, 2017-03-02:
Then, check section '7 VHPI Interface to an other language' in this tutorial: http://www.dossmatik.de/ghdl/GHDL_uart_sim.pdf That's an example of a UART. Both VHDL and C files are available at http://www.dossmatik.de/vhdl.html I think it will be less of a fantasy after you have a look at it ;). I believe that the use case you explained (injecting the clock externally) can be covered.

If it fits your needs, please close this issue, to let @tgingold know that the current status of GHDL can fit your needs through VHPI. If not, feel free to discuss the features you miss.

Good luck!

terriblefire, 2017-04-15:
This is the opposite way round to what i need. Instead of VHDL driving other code. I'd like to build a class object that i can eval() from C++ to bring it forward and make it evaluate itself internally. I'm pretty sure this wouldnt be too difficult as the simulator must already do something like this internally. Its just letting me use the HDL model like a class model in a C/C++ program.

lyssdod, 2018-08-25:
Agree. How difficult would it be to add external eval interface? If there's some core doc, I can try to make a PR

lyssdod, 2018-08-25:
It looks it's kind of possible with cocotb (Python): https://cocotb.readthedocs.io/en/latest/endian_swapper.html

Paebbels, 2018-08-25:
Cocotb is using the VPI interface, which is not so powerful as VHPI, but already supported.
VHPI is currently not supported by GHDL.

1138-4EB, 2019-03-25:
@sleary78, @terriblefire, did you get to co-execute GHDL and Verilator in the end?

terriblefire, 2019-03-25:
No i never managed to get this to work.

1138-4EB, 2019-03-26:
I'm sorry to hear that. Was it because GHDL lacks a `eval()` function and runs continuously until termination?

I have a working setup where GHDL is co-executed with some software, either by using pthreads or gRPC. Now I'd like to reproduce the setup with Verilator, and hopefully plug them together. Note that it is not mixed language simulation, because all the pieces in each module need to be written in the same language. But I think it should be doable to connect external interfaces such as FIFOs or AXI Streams. I'd very much appreciate any reference or thoughts.

terriblefire, 2019-03-26:
I never spent any more time on it to be honest. So i cant remember.