ghdl/ghdl #158: generate labels lost in ghw wave files (always 'B1')

wfjm, 2016-10-08:
Hi,

I'm using ghdl, build from git 
  commit 9ebac9a4cf1dce154a1fb7eecd75ccad189b3424
  Date:   Fri Jul 1 18:47:13 2016 +0200

build with gcc backend.
I generate ghw wave files and inspect them with gtkwave. When iterating in gtkwave over the available signals (here called facilities) I found many duplicate signals. Cross checking 
-  the vhdl code
-  ghwdump -h
-  gktwave 'facility' list

revealed that ghdl generates faulty signal names for ghw wave files.

Culprit are generate labels. From a code like

    IIST: if sys_conf_ibd_iist generate
    begin
      I0 : ibd_iist
        port map (
          CLK       => CLK,
          ...
          IIST_SRES => IIST_SRES
        );

I get in `ghwdump -h`

    generate-if B1
     instance i0
      port-in clk: slbit #412
      ...
      port-in iist_sres: iist_sres_type #11028 #11029

and consequently in gtkwave later signal names like

    top.tb_nexys4_cram.uut.ibdr_sys.B1.i0.clk
    ...
    top.tb_nexys4_cram.uut.ibdr_sys.B1.i0.iist_sres.ack_boot
    top.tb_nexys4_cram.uut.ibdr_sys.B1.i0.iist_sres.ack_lock

==> all generate labels are replaced by 'B1'
==> if one has multiple generates one gets duplicate signal names

I get for one entity, with multiple generates, in ghwdump

     instance ibdr_sys
       generate-if B1
        instance i0
        ...
       instance kw11l
        ...
       generate-if B1
        instance i0
        ...
       generate-if B1
        instance i0
        ...
       instance sdreg
        ...
       instance intmap

This is a rather simple bug, but with very serious consequences.

With best regards,   Walter

tgingold, 2016-10-10:
You're right.  Fix being tested.

diogratia, 2016-10-10:
A minimum complete and verifiable example to reproduce the issue:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ibd_iist is
    port (
        CLK:        in  std_logic;
        IIST_SRES:  in  std_logic
    );
end entity;

architecture fum of ibd_iist is
    
begin
    
end architecture;
    

library ieee;
use ieee.std_logic_1164.all;

entity foo is
end entity;

architecture fum of foo is
    constant sys_conf_ibd_iist:  boolean := true;
    signal clk:                  std_logic:= '0';
    signal IIST_SRES:            std_logic;
    component ibd_iist is
        port (
            CLK:        in  std_logic;
            IIST_SRES:  in  std_logic
        );
    end component;
begin
IIST: 
    if sys_conf_ibd_iist generate
    begin
      I0 : ibd_iist
        port map (
          CLK       => CLK,
          -- ...
          IIST_SRES => IIST_SRES
        );
    end generate;
CLOCK:
    process 
    begin
        wait for 10 ns;
        clk <= not clk;
        if now > 200 ns then
            wait;
        end if;
    end process;
end architecture;
```
And

```
ghwdump -h foo.ghw
design
 package standard
 package std_logic_1164
 instance foo
  signal clk: std_logic #1
  signal iist_sres: std_logic #2
  generate-if B1
   instance i0
    port-in clk: std_logic #1
    port-in iist_sres: std_logic #2
  process clock
```

While:

```
ghdl -r foo --disp-tree
foo [entity]
`-fum [arch]
  +-clk [signal]
  +-iist_sres [signal]
  +-iist [if-generate true]
  | `-i0 [instance]
  |   +-clk [port in]
  |   +-iist_sres [port in]
  |   `-ibd_iist [entity]
  |     +-clk [port in]
  |     +-iist_sres [port in]
  |     `-fum [arch]
  `-clock [process]
std.standard [package]
ieee.std_logic_1164 [package]
```

Which tells us it's either a ghwdump or ghdl ghw wave output issue.

You could also note that every block in a for generate will have the same label. See IEEE Std 1076-2008 14.5.3 Generate statements.

>NOTE—The repetition of the block labels in the case of a for generate statement does not produce multiple declarations of the label on the generate statement. The multiple block statements represented by the generate statement constitute multiple references to the same implicitly declared label.

(And if that doesn't confuse you...)

Also seen in IEEE Std 1076-1993 12.4.2 Generate statements.

And the question would be how is an instance_name attribute value in a for generate or case generate statement? See IEEE Std 1076-2008 16.2.5 Predefined attributes of named entities (or -1993 14.2 Predefined attributes) 

```
generate_label ::= generate_label [ ( literal ) ]
```

Where the literal in parentheses is the generate parameter used in a constant declaration in each block statement containing the concurrent statements of the generate statement body.

(And I don't recall how GHW encodes the label, the B1 stuff may be a fairly recent artifact. For generate statements should be checked too.)

closed 2016-10-11 by tgingold