ghdl/ghdl #160: VHDL 2008: cannot find entity or configuration

petspats, 2016-10-09:
Tested with latest build from sources.
top.vhd: 
```vhdl
entity top is
end entity;
```

Cmd:
```bash
ghdl -a --std=08 top.vhd
```
Resulting file work-obj08.cf:
```bash
v 4
file . "top.vhd" "05de2a35a54bbef8afc4616ac3a2cc1a60adc3a0" "20161009193833.953":
  entity top at 1( 0) + 0 on 11;
```

Cmd:
```bash
ghdl -e top
ghdl: cannot find entity or configuration top
```
Works if i change to --std=02, cf file identical but timestamp.

diogratia, 2016-10-10:
The default without specifying the std is 93c.

Using a .vhdl extension:

>%% ghdl -a --std=02 top.vhdl
>%% ghdl -e --std=02 top
>/usr/local/bin/ghdl1-llvm: entity "top" has no architecture in library "work"
ghdl: compilation error

What's missing is an architecture prior to elaboration.

Add a missing architecture for top:

```vhdl
architecture fum of top is
begin
end architecture;
```

>%% ghdl -a --std=08 top.vhdl
>%% ghdl -e --std=08 top
>%%

And top get's elaborated error free.

There are distinct libraries for each version of the standard.

tgingold, 2016-10-10:
Diogratia is correct here.  You forgot the --std switch when you elaborated the design.

I don't know what I could do to improve the error message.  Suggestions are welcome.

petspats, 2016-10-10:
Thanks guys. 
My fault, i skipped the RTFM for some example code that made me believe i only have to pass arguments to -a. Sorry for wasting time.

closed 2016-10-10 by petspats

ibkvictor, 2022-04-06:
```vhdl
library IEEE;
use ieee.std_logic_1164.all; -- contains std_logic data types
use IEEE.numeric_std.all; -- contains function like "to_unsigned"

entity systemx_tb is
    -- empty for testbenches
end entity;

architecture systemx_arch_tb of systemx_tb is
    component systemx is
        port (A, B, C : in std_logic;
            F : out std_logic);
    end component;

    signal signal_a, signal_b, signal_c, signal_f : std_logic;
    constant n : integer:= 3;

    begin
        comp : systemx port map (
            A => signal_a,
            B => signal_b,
            C => signal_c,
            F => signal_f
        );
    process
        variable i : integer;
        begin
            for i in 0 to 7 loop
                (signal_a,signal_b,signal_c) <= std_logic_vector(to_unsigned(i, n));
                report "signal_a: " & std_logic'image(signal_a);
                report "signal_b: " & std_logic'image(signal_b);
                report "signal_c: " & std_logic'image(signal_c);
                wait for 10 ns;
                report "nothing";
                report "signal_f: " & std_logic'image(signal_f);
            end loop;
    end process;

end architecture;
```

I have the same error for my testbench
`ghdl error: cannot find entity or configuration systemx_tb`

tgingold, 2022-04-06:
So same answer: you need to specify the `--std` also for elaboration and execution.