ghdl/ghdl #161: more telling ghdl -v output

wfjm, 2016-10-09:
currently `ghdl -v` always returns (for gcc backend)

    GHDL 0.34dev (20151126) [Dunoon edition]
       Compiled with GNAT Version: 4.8
       GCC back-end code generator

Whatever git version I use, I get  `0.34dev` and worse `20151126`.

It be great to have in the `ghdl -v` output a better indication of the version. In a svn environment one would simply use the svn version via a `Id:` tag in a string. In a git environment this is a bit more tricky but there are certainly good ways to achieve the goal.

With best regards, Walter

Masshat, 2016-10-09:
I think that the version is static: look at this file => https://github.com/tgingold/ghdl/blob/master/src/version.ads

But yes I think it is possible to have the tag version with git using git://github.com/tgingold/ghdl#tag=v${pkgver}
I Hope I answer to you question.

Paebbels, 2016-10-09:
The Windows build flow has a patch to overwrite the version string with build date, branch name and commit hash. I couldn't convince Tristan to enable such a patch for the make file based builds.

https://github.com/tgingold/ghdl/blob/master/dist/mcode/windows/targets.psm1#L144-L194

wfjm, 2016-10-09:
On 10/09/2016 10:46 PM, Massine wrote:
> I think that the version is static: look at this file => https://github.com/tgingold/ghdl/blob/master/src/version.ads
>
> But yes I think it is possible to have the tag version with git using git://github.com/tgingold/ghdl#tag=v${pkgver}
> I Hope I answer to you question.

I personally would prefer the string as returned by 'git describe',
that would be for the current master

   2016-09-14-65-g8b090b0

with 2016-09-14 --> last tag (here a date)
      65         --> number of commits after last tag
      g          --> for git
      8b090b0    --> last commit

still quite compact, but has very much information. The first two components

   2016-09-14-65   --> last tag + number of commits afterwards

is the closest equivalent to svn's "Id:", something linear development
and human readable. With the commit key one has a unique id in addition.

The question is more what the best way is to get this string into a
source file so it shows up with 'ghdl -v'.

Paebbels, 2016-10-09:
The number of commits since last tag and 'g' for git are not very usefull. We already know it's a git repository. Git encurages to make many commit, so a commit count is no quality mark or age since a tag. More over GHDL has 2 tag kinds: release tags and monthly build tags.

Output on Windows:

`GHDL 0.34dev (commit: 2016-09-05; git branch: paebbels/master; hash: dd78f39) [Dunoon edition]`

tgingold, 2016-10-10:
I am ok with any improvements here, but I don't know when to patch version.ads
Ideally it should be patched before a push.  I will investigate how to do that.

Paebbels, 2016-10-10:
My patch applies it at build time and restores the patched file afterwards so git doesn't show a diff. You could try git hooks e.g. pre-commit but the hash isn't known at that time.

wfjm, 2016-10-11:
On 10/10/2016 12:37 AM, Patrick Lehmann wrote:
> The number of commits since last tag and 'g' for git are not very usefull.
> We already know it's a git repository. Git encurages to make many commit,
>  so a commit count is no quality mark or age since a tag. More over GHDL
>  has 2 tag kinds: release tags and monthly build tags.

The output of 'git describe' is very compact and a careful balance
between 'human readable' and 'unique id'. See

   https://www.kernel.org/pub/software/scm/git/docs/git-describe.html

   The command finds the most recent tag that is reachable from a commit.
   If the tag points to the commit, then only the tag is shown. Otherwise,
   it suffixes the tag name with the number of additional commits on top
   of the tagged object and the abbreviated object name of the most
   recent commit.

The search for 'most recent tag' has some heuristics, but in practice
it gives what one wants.

The number of commits is imho very helpful, because is allows to compare
two 'describe strings' and one quickly sees which one is the newer version
(if tags are equal from higher commit count).

So before re-inventing the wheel I'd prefer to use the git describe output,
potentially augmented with repository and/or branch info.

		With best regards,  Walter

tgingold, 2016-10-11:
On 11/10/16 18:35, wfjm wrote:
> On 10/10/2016 12:37 AM, Patrick Lehmann wrote:
>
> The output of 'git describe' is very compact and a careful balance
> between 'human readable' and 'unique id'. See [...]

Generating the string is one step.
But we also need to decide when to generate it, and what to do with 
non-git sources (if you build from a tar).

Paebbels, 2016-10-11:
I'm using these commands in the Windows compile flow:

```PowerShell
$Git_Branch_Name =			& git rev-parse --abbrev-ref HEAD
$Git_Commit_DateString =	& git log -1 --format=%cd --date=short
$Git_Commit_ShortHash =		& git rev-parse --short HEAD
```

If the current working directory is under git control can be determined by:
```PowerShell
function Test-GitRepository
{ git rev-parse 2>&1 | Out-Null
  return $LastExitCode -eq 0
}
```

So a makefile can use the exit code of `git rev-parse` if its in a version controlled environment or extracted from a archive file. The windows flow patches the file before compilation, so HEAD is used to generate the extended version info. A command line option `-Release` can be used to disable this default behavior, so the original version e.g. `0.34dev` is used.

Paebbels, 2016-10-22:
So what format is used now? I see no linked commit.

tgingold, 2016-10-23:
Now really pushed.

commit de1c213c (2016-10-23): Create version.ads from git hash/date in version.ads

Fix #161
closed 2016-10-23 by tgingold