ghdl/ghdl #162: AMS support

femape, 2016-10-10:
Hello,
is it planned to bring AMS support to GHDL?

tgingold, 2016-10-10:
Ah good question.
There is a very preliminary, incomplete and certainly bit rotten support of AMS (try --ams) in the analyser.  I'd like to work on it but that's not my first priority.

Do you know a good ODE/DAE solver suitable for AMS simulations ?

fedmatip, 2016-10-10:
Hi Tristan,
I tried --ams and found some weird errors, then I realized that it is not yet supported. 
I know from experience 3 EDA softwares, which are quite expensive:
  - Synopsys Saber (I don't know the name of the solver)
  - Mentor Graphics System Vision (I also don't know)
  - Cadence AMS Designer (uses either Ultrasim or Spectre)

Another software I know, it's a free one, from a German university (see [here](http://www.theoinf.tu-ilmenau.de/~twangl/VHDL-AMS_online_en/)). It's a little out of date, but they do have a solution. May be it worths to contact them.

Do you work alone on this?

tgingold, 2016-10-11:
But Hamster doesn't look open source.

As far as I remember, you are the only one that expressed interest in AMS!

fedmatip, 2016-10-11:
Yes. I mean, maybe they would like to join GHDL and creating an open source AMS solution. After all, they are in the University.

The fact that I'm the only one asking seems quite surprising!

Paebbels, 2016-10-11:
The Fraunhofer Institute for Integrated Circuits IIS has an interest to use a free tool for VHDL-AMS.

tgingold, 2016-10-11:
On 11/10/16 11:31, FP wrote:
> Yes. I mean, maybe they would like to join GHDL and creating an open
> source AMS solution. After all, they are in the University.
>
> The fact that I'm the only one asking seems quite surprising!

My understanding is that AMS-Verilog is much more popular.

tgingold, 2016-10-11:
On 11/10/16 11:35, Patrick Lehmann wrote:
> The Fraunhofer Institute for Integrated Circuits IIS has an interest to
> use a free tool for VHDL-AMS.

Are they ready to put efforts for that ?

Paebbels, 2016-10-11:
I don't know.

One of the employees (@JoachimHaase) is the secretary at [IEEE P1076.1](http://www.eda-twiki.org/cgi-bin/view.cgi/P10761/WebHome). He got in contact with me to supply the [Table Driven Modelling](http://www.vhdl.net/proposals/ams/tabledrivenmodelling) packages for VHDL-AMS to [CoreLib](https://github.com/VHDL/CoreLib). I know that IIS is heavily using VHDL and this group is using AMS stuff too.

What about an email to Ernst Christen (chair) to ask if he can ask if there is a demand from a free VHDL-AMS tool?

fedmatip, 2016-10-11:
It's nice to see that my question was a trigger! I think most silicon vendors will be interested on this ;)

closed 2016-10-15 by tgingold