ghdl/ghdl #164: Error while analysing unconstrained vectors in records

mzabeltud, 2016-10-10:
Hello,

VHDL'08 allows the declaration unconstrained vectors in records, GHDL terminates with an exception.

```VHDL
package p is
  type t is record
    a : bit_vector;
  end record t;
end package p;
```

I used the GHDL version 8b090b01da269679e98a578f4fda17e3e4adce12 and executed
```
/opt/ghdl/current/bin/ghdl -a --std=08 records.vhdl
```

The GHDL output is: 
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.9.2
In directory: /home/zabel/Dokumente/projekte/VHDL/records/
Command line:
/opt/ghdl/current/bin/ghdl1-llvm --std=08 -P/opt/ghdl/current/lib/ghdl/v08/std/ -P/opt/ghdl/current/lib/ghdl/v08/ieee/ -c -o records.o records.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: trans-chap7.adb:4537
Call stack traceback locations:
0x52ab7a 0x52bc29 0x52fc39 0x598ae7 0x5b1fd7 0x558b0a 0x5c0caf 0x46f335 0x430f0f 0x7f42a84aeb43 0x430360 0xfffffffffffffffe
******************************************************************
/opt/ghdl/current/bin/ghdl: compilation error
```

Regards,
Martin

tgingold, 2016-10-10:
Yes, the support is incomplete.  I think this is now rather high on my priority list as this is often requested.

closed 2016-10-10 by tgingold

petspats, 2016-11-26:
@tgingold 
I just ran into the same problem, more specifically array of unconstrained type. I am wondering what is the progress on this issue, should i implement my own workarounds or maybe you are close to releasing something?