ghdl/ghdl #166: libgnat-4.6 on Ubuntu 16

petspats, 2016-10-15:
On Ubuntu 16, executing any Pre-Release results:
`./ghdl: error while loading shared libraries: libgnat-4.6.so.1: cannot open shared object file: No such file or directory`

Basically there is a need to run `sudo apt install libgnat-4.6`, Ubuntu 16 has no such package (12 and 14 have it).

Fix:
- `echo "deb http://ee.archive.ubuntu.com/ubuntu precise main universe" | sudo tee -a /etc/apt/sources.list`
- `sudo apt update`
- `sudo apt-get install -y gnat-4.6-base, libgnat-4.6`
- Remove last line from  /etc/apt/sources.list

petspats, 2016-10-15:
Documenting the issue, maybe saves some time for someone.

closed 2016-10-15 by petspats

Paebbels, 2016-10-15:
Hello,

you don't need a `tee` command. STDOUT redirection with `>>` appends to a file. Secondly, you can spare the delete last line command if you copy before and restore the file after your patch operation.

But to solve your issue with Ubuntu 16.04 (please note Ubuntu versions have two numers: a year and a month) you could compile GHDL with a newer GCC version e.g. GCC/GNAT 5.3. Ubuntu 16.04 should containt  these GNAT libraries.

petspats, 2016-10-16:
Hey,
Thanks for the tips.
I think that recompiling is actually harder as you must install GNAT GPL.

periata, 2018-05-25:
This approach no longer seems to work cleanly on 17.10.  The repository listed above gives errors about having an invalid signature (even if I import the relevant public key).  Switching from "precise" to "trusty" (which seems to be the last version that supported the required packages) gets rid of the signature issues, but still doesn't work (it looks as though apt is failing to understand the format of the archive... I presume some update has broken backwards compatibility).

1138-4EB, 2018-05-25:
@periata, I'd suggest to build it yourself. It will be easier, because you will use the default package versions available in Ubuntu 17.10, instead of adding possibly outdated content.

See https://github.com/ghdl/ghdl#building-ghdl and http://ghdl.readthedocs.io/en/latest/building/Building.html

PetteriAimonen, 2019-02-06:
Seems to work fine on Ubuntu 18.04 by downloading these two packages manually from links below and installing with `sudo dpkg -i xxx.deb`:

https://packages.ubuntu.com/trusty/gnat-4.6-base
https://packages.ubuntu.com/trusty/libgnat-4.6

gasparka, 2019-02-06:
Best way to run GHDL these days is trough Docker:
https://hub.docker.com/r/ghdl/ghdl

1138-4EB, 2019-02-07:
@PetteriAimonen It might work with `libgnat-4.6` on Ubuntu 18.04 because the prebuilt binaries (`ghdl-20181129-ubuntu14-llvm-3.8.tgz` and `ghdl-20181129-ubuntu14-mcode.tgz`) are built with `gnat-4.6` on Ubuntu 14.04. However, note that we cannot guarantee it. At some point some other libraries might be needed, or incompatibilities might arise. As a rule of thumb, it is not good practice to install packages for `trusty` in `bionic`.

Therefore, we recommend to use one of these alternatives:

- Use the GHDL package available in the repos for your distribution. This won't probably be the latest available GHDL version.
- Build it on your own, with the helper scripts in `dist` (https://github.com/ghdl/ghdl/blob/master/.travis.yml#L6).
- Use the docker images suggested by @gasparka, to run GHDL in a container (https://github.com/ghdl/docker/blob/master/USE_CASES.md).
- Since Ubuntu 18 is one of the platforms where we test GHDL in the CI flow (see https://travis-ci.com/ghdl/docker/builds/99877101), there are prebuilt tarballs available. However, these are not published through GitHub, but through docker. For example:

``` bash
cat <<-EOF > Dockerfile
FROM ghdl/pkg:ubuntu18-llvm-5.0 AS pkg-tmp
FROM busybox
COPY --from=pkg-tmp ./* ./ghdl-pkgs/
EOF

docker build -t save_pkg .
docker run --rm -tv /$(pwd)://src save_pkg busybox sh -c "cp -r ghdl-pkgs/ghdl-* /src"
docker rmi save_pkg
rm Dockerfile
```

will download `ghdl-20190206-75b01176a3-ubuntu18-llvm-5.0.tgz`.

The runtime dependencies are: [dockerfiles/run/ubuntu18#L7](https://github.com/ghdl/docker/blob/master/dockerfiles/run/ubuntu18#L7) and [dockerfiles/run/ubuntu18#L16](https://github.com/ghdl/docker/blob/master/dockerfiles/run/ubuntu18#L16).

milannedic, 2019-04-11:
> Preparing to unpack ghdl_0.33-1ubuntu1_amd64.deb ...
Unpacking ghdl (0.33-1ubuntu1) over (0.33-1ubuntu1) ...
dpkg: dependency problems prevent configuration of ghdl:
 ghdl depends on libgnat-4.8 (>= 4.8.2-8ubuntu3); however:
  Package libgnat-4.8 is not installed.

Ubuntu 16.04 Fix:
---------------
Had to manually download and Install dependencies:
      sudo dpkg -i gnat-4.8-base_4.8.2-8ubuntu3_amd64.deb
      sudo dpkg -i libgnat-4.8_4.8.2-8ubuntu3_amd64.deb

First download them from here: 
      https://packages.ubuntu.com/trusty/amd64/gnat-4.8-base/download
      https://packages.ubuntu.com/trusty/i386/libgnat-4.8/download

1138-4EB, 2019-04-11:
@milannedic, note that version 0.33 is several years old. It is strongly suggested to use a more recent version.

praton1729, 2019-05-19:
> On Ubuntu 16, executing any Pre-Release results:
> `./ghdl: error while loading shared libraries: libgnat-4.6.so.1: cannot open shared object file: No such file or directory`
> 
> Basically there is a need to run `sudo apt install libgnat-4.6`, Ubuntu 16 has no such package (12 and 14 have it).
> 
> Fix:
> 
> ```shell
> echo deb http://cz.archive.ubuntu.com/ubuntu precise main universe precise main universe > /etc/apt/sources.list.d/tmp.list
> apt-get update
> apt-get install -y gnat-4.6-base libgnat-4.6
> rm -f /etc/apt/sources.list.d/tmp.list
> ```

Might wanna add the `[trusted=yes]` flag to it so that it won't complain of Release signatures in the repo.

1138-4EB, 2019-05-19:
As commented above, installing packages from Precise in more recent versions (Trusty, Xenial, Bionic) is not suggested at all.

gasparka, 2019-05-20:
@tgingold 
Fact that this thread is still alive shows that something is wrong about how GHDL is distributed.
I would suggest making Docker the default, and endorsed way of running GHDL. Any reasons against?

PetteriAimonen, 2019-05-20:
Docker seems overkill, though it works. Is there a particular reason why the current Ubuntu 14.04 packages couldn't be upgraded into Ubuntu 18.04 packages?

1138-4EB, 2019-05-20:
> @gasparka:
> Fact that this thread is still alive shows that something is wrong about how GHDL is distributed.

I wouldn't say there is anything wrong about how GHDL is distributed, but there is still work to do. Precisely, we are lacking active package maintainers for the main OSs/distros which have (in)direct communication with the core of the project. We know that there are several people which keep packages somehow up to date on Debian, Arch and Fedora, but AFAIK there is no explicit coordination: i.e. no one is explicitly notified when @tgingold pushes a tag (#514).

> NOTE: in this context, I consider the core of the project to be @tgingold, @Paebbels and myself, because we are the ones that have coauthored most of the build scripts in subdir `dist`.

We are aware of these ToDos, and there are several open issues:

- #497
- #507
- #744

Nonetheless, we do have pre-built binaries/tarballs for many more contexts than those in [releases](https://github.com/ghdl/ghdl/releases). Precisely, we have 7 mcode (https://travis-ci.com/ghdl/docker/builds/112342082), 5 LLVM (https://travis-ci.com/ghdl/docker/builds/112389031) and 4 GCC (https://travis-ci.com/ghdl/docker/builds/112389189) builds in the 'extended test matrix'. Furthermore, we can produce them for any of the platforms supported by docker and QEMU (amd64, i386, arm64v8, arm32v7, arm32v6, s390x and ppc64le). That's more than a hundred possible combinations! Distributing AppImages has been suggested too... So, although it is straighforward to add any of these to the main repo, we need cannot add all of them.

> @gasparka:
> I would suggest making Docker the default, and endorsed way of running GHDL. Any reasons against?

Docker is already the default for many of us. It is mostly used in CI environments, but it is used in desktop workstations too. For example, it is my preferred way to use GHDL on Win10. There are users which are building tools on top of some of the docker images that we provide. For example: https://section5.ch/index.php/2018/11/04/masocist-opensource-0-2-release/

The reason against not making this alternative more explicit in the README/Documentation is because we do not want to force users towards a technology that they don't know. Docker is gaining popularity and at some point using OCI containers will be common for any user. However, nowadays running Docker is not straighforward on some platforms where GHDL can be build and executed. Unfortunately, many universities and companies are still using Windows 7/Vista.

Ideally, we would be able to automate the process so that GHDL is available through the preferred package manager on each platform (apt, dnf, pacman, chocolatey, zypper). Note that up to 6 packages can be made available: mcode/LLVM/GCC, unstable/stable. That would offload the build to the resources/services of each platform.

For example, GHDL v0.35 is available on Debian Buster: https://packages.debian.org/buster/ghdl. Ref #377.

> Docker seems overkill, though it works. Is there a particular reason why the current Ubuntu 14.04 packages couldn't be upgraded into Ubuntu 18.04 packages?

I think it is because the default context in Travis are Ubuntu 14.04 VMs. Therefore, any user which does not add any custom option to the `.travis.yml` config file can directly retrieve a package, extract it and execute some tests.

Since it is straighforward to execute any of the provided docker containers on Travis, I think we may drop this requirement when #569 is closed (i.e. when we provide/tag stable docker images). @Paebbels, let us know what you think.

@PetteriAimonen, apart from that, packages for Debian Buster might work on Ubuntu 18.04. You might try: https://packages.debian.org/buster/amd64/ghdl/download. It is not an ideally solution, but I believe that it cannot be much more harmful than installing packages from Ubuntu 14.04.

Alternatively, you can wait until Ubuntu 19.04 is released. It seems that ghdl is available in it:

- https://qa.debian.org/developer.php?login=aeb%40debian.org
- https://launchpad.net/ubuntu/+source/ghdl

``` bash
# winpty docker run --rm -it ubuntu:19.04 bash
root@ca57efb87069:/# apt search ghdl
Sorting... Done
Full Text Search... Done
ghdl/disco 0.35+git20181129+dfsg-3ubuntu1 amd64
  VHDL compiler/simulator

ghdl-gcc/disco 0.35+git20181129+dfsg-3ubuntu1 amd64
  VHDL compiler/simulator (GCC backend)

ghdl-llvm/disco 0.35+git20181129+dfsg-3ubuntu1 amd64
  VHDL compiler/simulator (LLVM backend)

ghdl-mcode/disco 0.35+git20181129+dfsg-3ubuntu1 amd64
  VHDL compiler/simulator (mcode backend)
```

gasparka, 2019-05-20:
> Docker is already the default for many of us. It is mostly used in CI environments, but it is used in desktop workstations too. For example, it is my preferred way to use GHDL on Win10. There are [...]

Unfortunately the possibility of using GHDL over Docker is not even mentioned in the README, nor in the Documentation. I propose to add a Docker Hub badge to the README and some examples to the documentation. For example, i did not know it works in Windows 10 - useful example for sure.

> Docker seems overkill, though it works. 

IMHO the killer feature is that once you get it working, it stays working - which, debatably, is worth the small set-up overhead.

1138-4EB, 2019-05-20:
> Unfortunately the possibility of using GHDL over Docker is not even mentioned in the README, nor in the Documentation. I propose to add a Docker Hub badge to the README and some examples to the documentation.

No, it is not. That's a quite opinionated but deliberate decission. I was the one who contributed adding docker images to the project, and I keep them up to date. Unfortunately, I have not had time to write a *decent* 'Docker premier' for GHDL users, so the content is neither announced nor publicized yet. The 

Nevertheless, the GHDL GitHub organization contains two repos only. One of them is this, and the other one is named `docker`. I thought that it would be clear for any user that all the content related to docker is in the second repo.

Please, feel free to open any specific issue/PR, such as the suggesting the addition of a badge to the README and/or the docs.

Moreover, we might add [USE_CASES](https://github.com/ghdl/docker/blob/master/USE_CASES.md) to ghdl.readthedocs.io. I have not done so because I feel that the content is only useful to users who already have an intermediate to high knowledge of how containers work, and it might be frustrating for users with a 'more traditional' approach. Related to this, you might have a look at #506. In the [demo](https://1138-4eb.github.io/ghdl/) (see the docker badge in the footer) there is some content related to docker: https://1138-4eb.github.io/ghdl/guides/docker/. Once again, feel free to propose/contribute any enhancement to the README and/or Docs, based on the information you find there.

> For example, i did not know it works in Windows 10 - useful example for sure.

As a matter of fact, I do most of my development on Win10, but all the tools I use are for GNU/Linux. I normally start a `ghdl/ext:vunit-gtkwave` docker container with `x11docker`, and then install `terminator`. You get exactly the same environment as in any GNU/Linux host, which is pretty cool. Moreover, as explained in [USE_CASES](https://github.com/ghdl/docker/blob/master/USE_CASES.md), GtkWave can be used in a web browser.

Another interesting approach is using Visual Studio Code, which allows to open/attach the terminal to docker containers. In the near future, it will also be possible to open a directory inside the container: https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.vscode-remote-extensionpack

gasparka, 2019-05-20:
@1138-4EB I very much like the "USE_CASES" page,  definitely a good addition to ghd.readthedocks.io. Maybe it could be a separate chapter under 'GHDL usage' ? I think it is appropriate if you PR it.

Thanks for the x11docker hint, i will check it out - but it already looks like Nvidia + OpenGL will be trouble no matter what.

1138-4EB, 2019-05-20:
> @1138-4EB I very much like the "USE_CASES" page, definitely a good addition to ghd.readthedocks.io.

Thanks.

> Maybe it could be a separate chapter under 'GHDL usage' ? I think it is appropriate if you PR it.

I'll try to add it ASAP. But it won't be before july.

> Thanks for the x11docker hint, i will check it out - but it already looks like Nvidia + OpenGL will be trouble no matter what.

Actually, I did run some tests last week. You can use hardware acceleration with NVIDIA GPUs inside docker containers on Win10 as long as you use Cygwin/X (Xwin). See https://github.com/mviereck/x11docker/issues/148#issuecomment-491607833

raczben, 2020-03-02:
The files replaced.

this has worked for me:

```console
wget http://archive.ubuntu.com/ubuntu/pool/universe/g/gnat-4.8/libgnat-4.8_4.8.2-8ubuntu3_amd64.deb
wget http://archive.ubuntu.com/ubuntu/pool/universe/g/gnat-4.8/gnat-4.8-base_4.8.2-8ubuntu3_amd64.deb

sudo dpkg --install gnat-4.8-base_4.8.2-8ubuntu3_amd64.deb
sudo dpkg --install libgnat-4.8_4.8.2-8ubuntu3_amd64.deb
```