ghdl/ghdl #168: Missing "compilation error" line in GHDL mcode (Windows)

Paebbels, 2016-10-18:
Hello,

as reported by #167, GHDL emits a `ghdl: compilation error` line after an exception message.
This lines does not appear in the mcode version (tested on Windows).

GHDL, Linux 64-bit, GCC/GNAT 4.9.2, LLVM 3.5:
```
get_object_type_staticness: cannot handle IIR_KIND_COMPONENT_DECLARATION (generic_attribute.vhdl:2:13)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.9.2
In directory: /home/zabel/Dokumente/projekte/VHDL/bug_report/
Command line:
/opt/ghdl/0.34dev/bin/ghdl1-llvm --std=08 -P/opt/ghdl/0.34dev/lib/ghdl/v08/std/ -P/opt/ghdl/0.34dev/lib/ghdl/v08/ieee/ -c -o generic_attribute.o generic_attribute.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:158
Call stack traceback locations:
0x48e2d3 0x5d8033 0x5d8417 0x5d991c 0x5d9d0a 0x5dec75 0x5f1fcb 0x5ea643 0x5f2015 0x5e780b 0x5e7fed 0x5e8447 0x5c79d8 0x5c9a73 0x5c9fd7 0x5f3bad 0x5f4c30 0x5f836c 0x5fa4fe 0x60052a 0x6016af 0x5657f8 0x5c1ca5 0x46ff5f 0x430f0f 0x7f81c9d28b43 0x430360 0xfffffffffffffffe
******************************************************************
ghdl: compilation error
```

GHDL, Windows 64-bit, AdaCore GNAT 2016, mcode (32-bit)

```
get_object_type_staticness: cannot handle IIR_KIND_COMPONENT_DECLARATION (D:\git\poc\src\comm\comm.pkg.vhdl:38:18)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (commit: 2016-10-17;  git branch: paebbels/master';  hash: 2379a19) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
In directory: D:\git\poc\temp\ghdl\
Command line:
C:\Tools\GHDL\0.34dev\bin\ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --std=08 --work=poc D:\git\poc\src\comm\comm.pkg.vh

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:158
******************************************************************
```

The additional `compilation error` line makes it easier to scan raw GHDL outputs for warnings, errors or fatal errors (like this). Is it possible to add it to the ouput of mcode backended GHDLs?

tgingold, 2016-10-18:
No, because this message is displayed by the parent process when gcc/llvm backend is used.

Use the exit status to check for errors.

closed 2016-10-18 by tgingold

Paebbels, 2016-10-18:
So `ghdl` generates this line for the invoked `ghdl1-llvm` or `ghdl1-gcc` executables?

tgingold, 2016-10-18:
> So |ghdl| generates this line for the invoked |ghdl1-llvm| or
> |ghdl1-gcc| executables?

Yes.