ghdl/ghdl #172: winbuild

Paebbels, 2016-10-19:
Hello,

with #171, the build flow on Windows got more settings (add GHDL to PATH). Should I add a PowerShell equivalent like `./configure` to the flow?

Current settings:
* InstallDir -> like `--prefix`

Possible other settings:
* Register in Path: `Ask at Install`, `No`, `Machine`, `User`, `Session` -> GHDL is installed in a searched directory on Linux
* Add highlighting filters to global PowerShell script directory

Kind regards
    Patrick

tgingold, 2016-10-19:
I don't think you need a configure; you can pass options directly to the script.

Paebbels, 2016-11-10:
For now this idea is delayed :). I have a single one-liner file, which saves the installation directory. So a user can run the build/install script with `-Update`, which will read back the installation folder, saved while using `-Install [path]` the first time. It can be overwritten by using `-Install [other\path]` again or deleted while using `-Clean` to remove the build directory.

closed 2016-11-10 by Paebbels