ghdl/ghdl #173: GHDL's version name: 0.34dev

Paebbels, 2016-10-19:
Hello,

I'm perfectly fine with GHDL's version naming e.g. `0.34dev`, and I wouldn't request to change it, but other tools have complains about it :(.

I'm trying to package GHDL with Chocolatey as a NuGet package. NuGet's XML schema definition requires that version strings adhere to the [semantic versioning rules](http://semver.org/).

As a workaround, I'm currently using `0.34-latest` on my local machine for the latest build on Windows.

* Is this naming problem an issue for e.g. Debian packages too?
* Can we change the naming convention in the future to apply to their rules? 

**Proposed new naming scheme:**
* Release version: 0.34
* Development version: 0.34-dev or 0.34-latest
* Tagged versions: 0.34-tagged-20161019

tgingold, 2016-10-19:
I am ok with the dash, but I prefer 0.34-dev
Less disruptive.

Paebbels, 2016-10-20:
And the tagged version?

Dots `.` and plus `+` seem to be not supported.

commit 3c5dd6f1 (2016-10-20): Add a dash in version.

Fix #173
closed 2016-10-20 by tgingold