ghdl/ghdl #174: Automated builds for GHDL on Windows

Paebbels, 2016-10-20:
Hello,

shall I investigate how to setup a Travis like flow for Windows based on [AppVeyor](https://www.appveyor.com/)?

If it works, we could also deliver Windows builds per tagged commit.

Kind regards
    Patrick

Paebbels, 2016-10-20:
Can you get a documentation for the undocumented command line switches of GNAT GPL's installer? It has no builtin help accessible through: `-h`, `-help`, `--help`, `/h`, `/?`.

It seems that `/S` starts a silent installation to `C:\GNATPRO`.

tgingold, 2016-10-25:
On 20/10/16 12:04, Patrick Lehmann wrote:
> Hello,
>
> shall I investigate how to setup a Travis like flow for Windows based on
> AppVeyor <https://www.appveyor.com/>?

That would be nice.  The main issue is to install gnat to compile ghdl.

> If it works, we could also deliver Windows builds per tagged commit.

Yes!

Paebbels, 2016-10-25:
Is there a x86-64 GNAT compiler for Windows 64-bit, except for the MinGW64 gcc gnat?

I'll try to start a 32-bit console on Windows, maybe this console can invoke 32-bit programs without restrictions...

tgingold, 2016-10-25:
On 21/10/16 00:00, Patrick Lehmann wrote:
> Can you get a documentation for the undocumented command line switches
> of GNAT GPL's installer? It has no builtin help accessible through:
> |-h|, |-help|, |--help|, |/h|, |/?|.
>
> It seems that |/S| starts a silent installation to |C:\GNATPRO|.

IIRC, this is an NSIS installer.  Use /S for silent installation and 
/D=DIR to specify the destination.

tgingold, 2016-10-31:
On 25/10/16 20:43, Patrick Lehmann wrote:
> Is there a x86-64 GNAT compiler for Windows 64-bit, except for the
> MinGW64 gcc gnat?

No, AFAIK.  What is wrong with mingw64 ?

Paebbels, 2016-10-31:
I had some issues with the results from MinGW64 builds. Moreover, I think we need to ship MinGW with GHDL to run it. My MinGW folder has 4,00 GiB (I could remove MinGW32) but this leaves us with 2 circa 2 GB installation to run a <10 MB GHDL ...

MinGW64 builds are nice for people with MinGW64 experience, but it's no so lightweight like a native Windows GNAT GPL build.

Accoring to my filed issue at AppVeyor, it's possible to reboot the complete CI instance in a DEP disabled mode. Sounds strange to reboot the VM ... Let's see how the CI environment can handle such an event e.g. get the correct report messages and built results.

tgingold, 2016-11-01:
What are the issues with mingw64 ?
Are you compiling for mcode ?

Paebbels, 2016-11-02:
I did fresh MinGW64 compilations with mcode and llvm backend. Both builds compile in the MinGW64 Bash environment. There are no errors while analyzing the IEEE libraries. The installation directory contains all expected files (*.cf and *.o for the llvm backend).

I can run `ghdl.exe -v` from PowerShell with the llvm backend but not with the mcode backend. If I run `ghdl1-llvm.exe` I get no outputs or errors. I expect at least a complaint about a missing input file or command.

Ah now it's getting interesting ... I modified my PATH variable a bit and added MinGW's bin folder. I got:

```
  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (2016-09-14-176-g6abb2ed-dirty) [Dunoon edition]
  Compiled with GNAT Version: 6.2.0
  In directory: D:\git\PoC\temp\ghdl\
  Command line:
  C:\Tools\GHDL\0.34dev-mingw64-mcode\bin\ghdl.exe -r -v -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=te
st arith_prng_tb --ieee-asserts=disable-at-0
  Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
  Exception information:
  raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ortho_code-x86-emits.adb:2942
  Call stack traceback locations:
  0x6c48cb 0x5a7fb1 0x5a83c2 0x5a9109 0x5a92e1 0x556977 0x4d9c22 0x63668f 0x622a52 0x62ce28 0x622a91 0x637486 0x5edcc1 0x63d5e3 0x5f8d95 0x6418c2 0x649f6b 0x648b11 0x4da41e 0x402b15 0x4
013f6 0x401519 0x76b159cb 0x76d4a2df
  ******************************************************************
```

If I launch GHDL with llvm backend, then I get:

```
  C:\Tools\GHDL\0.34dev-mingw64-llvm\bin/ghdl1-llvm.exe -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --std=08 --work=poc -PC:\Tools\GHDL
\0.34dev-mingw64-llvm\lib/ghdl\v08\std\ -PC:\Tools\GHDL\0.34dev-mingw64-llvm\lib/ghdl\v08\synopsys\ -c -o my_project.o D:\git\PoC\tb\common\my_project.vhdl
  C:\Tools\GHDL\0.34dev-mingw64-llvm\bin\ghdl.exe: exec error
```

So `ghdl.exe` runs and writes bytes to STD_OUT, but the forked `ghdl1-llvm.exe` causes an error (or is it just silent?). According to some websites, I just need to add `C:\msys2\mingw64\bin` to `PATH`. This is what caused the exception from above.

**MinGW32**
* GHDL build with MInGW32 and backend set to llvm is silent like the 64 bit llvm version.
* **GHDL build with MinGW32 and backend set to mcode, works!**

So this is one out of four, if started from PowerShell :(.

Kind regards
    Patrick

------------------
Windows = Windows 7 64-bit
MSYS = MSYS64 / MSYS2
MinGW = MinGW32 or MinGW64
GCC / GNAT = 6.2.0
LLVM = 3.5.2

tgingold, 2016-11-02:
On 02/11/16 03:06, Patrick Lehmann wrote:
> I did fresh MinGW64 compilations with mcode and llvm backend. Both
> builds compile in the MinGW64 Bash environment. There are no errors
> while analyzing the IEEE libraries. The installation directory [...]

An x86-64 mcode ghdl for Windows is likely to crash, as this 
configuration is not supported.

Broken llvm backend (32 or 64 bits) are more surprising.  Missing dll ?

Tristan.

Paebbels, 2016-11-02:
Any idea how to degug it? There is no output and no failure indication except for the missing output and exe file. if *.o files exist, GHDL has emitted llvm bases code, right? So  there is no linker running to build the exe?

The missing exe file warning is from PoC as it does also warn for e.g. iSim or xSim if a exe result is missing.

tgingold, 2016-11-02:
> Any idea how to degug it? There is no output and no failure indication
> except for the missing output and exe file. if *.o files exist, GHDL has
> emitted llvm bases code, right? So there is no linker running to build
> the exe?

Try to add two -v options, to get more traces.
It is possible that a linker (or gcc) is missing.

Tristan.

Paebbels, 2016-11-02:
OK I found a bug on my side. I needed to configure PoC for `llvm`. An explicit elaboration step is needed if the backend is llvm and the OS is Windows. (PoC did only support mcode, which oversteps elab on Windows.)

Now I get:
> gcc.exe: error: C:\Tools\GHDL\0.34dev-mingw64-llvm\lib/ghdl\v08\std\std_standard.o: No such file or directory
> C:\Tools\GHDL\0.34dev-mingw64-llvm\bin\ghdl.exe: compilation error

Hmmm yes, std_standard.o is missing:
```PowerShell
Directory: C:\Tools\GHDL\0.34dev-mingw64-llvm\lib\ghdl\v08\std

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
------       02.11.2016     01:18            985 env.o
------       02.11.2016     01:18           2228 env_body.o
------       02.11.2016     01:18            589 std-obj08.cf
------       02.11.2016     01:18           3111 textio.o
------       02.11.2016     01:18          60275 textio_body.o
```

Actually, no VHDL version sub-directory in `lib/ghdl` contains this file. The std-obj08.cf file contains no entry for std_standard:
```
v 4
file . "../../src/std/env_body.vhdl" "deb806d24e820c331dfcedbec320bad47e701517" "20161102001810.309":
  package body env at 20( 893) + 0 on 14;
file . "../../src/std/textio_body.v08" "a24b0e4a66e69cbc17958d4c0d7e696a269e0ea2" "20161102001810.071":
  package body textio at 18( 760) + 0 on 12;
file . "../../src/std/textio.v08" "859f1cedaa82708600ff69335cad95b4c95964b7" "20161102001810.005":
  package textio at 19( 820) + 0 on 11 body;
file . "../../src/std/env.vhdl" "92c0566a64672bc27f707d4418586507a7e1eaf1" "20161102001810.248":
  package env at 20( 893) + 0 on 13 body;
```

**Questions:**
* Is it correct that there is no reference to std_standard ?
* Shall I dump the complete make output (for the library compilations)?

tgingold, 2016-11-02:
On 02/11/16 13:00, Patrick Lehmann wrote:
> OK I found a bug on my side. I needed to configure PoC for |llvm|. An
> explicit elaboration step is needed if the backend is llvm and the OS is
> Windows. (PoC did only support mcode, which oversteps elab on Windows.)

[...]

>   * Is it correct that there is no reference to std_standard ?

Yes, that's correct.

>   * Shall I dump the complete make output (for the library compilations)?

I suppose so.  Does your script build it ?
(See Makefile.in for how this file is created).

Paebbels, 2016-11-02:
I got one step further.
Because issue #181 is not merged yet, the makefile doesn't finish. I was aware of this fact, but I didn't notice that std_standard.o is build **after** all other libraries. So that explains the missing `std_standard.o` files.

So what's the current status?
The exe file exists (circa 3 MiB), but it crashes after launching the file. I also tried to launch the exe directly in PowerShell. => Same error.
[image]

This is the result from `gdb`:
```gdb
D:\git\PoC [paebbels/master = +7 ~9 -0 !]> gdb .\temp\ghdl\arith_prng_tb.exe
GNU gdb (GDB) 7.12
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from .\temp\ghdl\arith_prng_tb.exe...done.
(gdb) r
Starting program: D:\git\PoC\temp\ghdl\arith_prng_tb.exe
[New Thread 7308.0x4220]

Program received signal SIGSEGV, Segmentation fault.
__ghdl_run_through_longjump (func=0x4a0960 <grt.main.ghdl_elaborate_wrapper>) at ../../src/grt/config/win32.c:171
171       asm ("mov %%fs:(0),%0" : "=r" (prev));
(gdb)
```

What can I do next?

tgingold, 2016-11-02:
On 02/11/16 18:48, Patrick Lehmann wrote:
> I got one step further.
> Because issue #181 <https://github.com/tgingold/ghdl/pull/181> is not
> merged yet, the makefile doesn't finish. I was aware of this fact, but I
> didn't notice that std_standard.o is build *after* all other libraries.
> So that explains the missing |std_standard.o| files.

I am lost: pull #181 adds std_logic_misc, so what is the error when not 
merged ?
(Pull #181 is not yet ready as you know).

> So what's the current status?
> The exe file exists (circa 3 MiB), but it crashes after launching the
> file. I also tried to launch the exe directly in PowerShell. => Same error.
> image
> <https://cloud.githubusercontent.com/assets/956109/19940339/995c3e02-a12c-11e6-8d42-f7ea7db92a55.png>
>
> This is the result from |gdb|:
>
> D:\git\PoC [paebbels/master = +7 ~9 -0 !]> gdb .\temp\ghdl\arith_prng_tb.exe
> GNU gdb (GDB) 7.12
> Copyright (C) 2016 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "x86_64-w64-mingw32".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from .\temp\ghdl\arith_prng_tb.exe...done.
> (gdb) r
> Starting program: D:\git\PoC\temp\ghdl\arith_prng_tb.exe
> [New Thread 7308.0x4220]
>
> Program received signal SIGSEGV, Segmentation fault.
> __ghdl_run_through_longjump (func=0x4a0960 <grt.main.ghdl_elaborate_wrapper>) at ../../src/grt/config/win32.c:171
> 171       asm ("mov %%fs:(0),%0" : "=r" (prev));
> (gdb)
>
> What can I do next?

This SEH code is x86 specific, so it should be at least disabled on x86-64.

Tristan.

Paebbels, 2016-11-02:
Regarding #181: The change was active in my local branch, because I supplied the pull request :(.

OK, I tested `ghdl-0.34dev-mingw32-llvm` and it's now behaving like `ghdl-0.34dev-mingw32-mcode`, when all previously found changes are applied. This brings us to 2 out of 3(4).

SEH sounds like exception handling, right? (I found this: http://stackoverflow.com/questions/15670169/what-is-difference-between-sjlj-vs-dwarf-vs-seh)

How do I disable SEH or chose another exception handling mechanism for MinGW64 / GCC?
Another MinGW64 distribution provides different GCC setups SEH or SJLJ. Do I need to reconfigure GCC somehow in my MinGW64 setup (msys2 / msys64).

------------------
I tried to compile the the vendor libraries with this mingw32-llm build.

**Finding 1:**
I need to implement a max-file size switch for the PowerShell scripts too, if GHDL is using LLVM. (The Bash script has already this option.)

**Finding 2:**
I got these errors:
[image]
Shall I open a new issue for that?

Paebbels, 2016-11-02:
**As an update to the AppVeyor based approach:**
The support provided me a working yaml file, which disables the NX bit and reboots the entire VM.

Conversation at AppVeyor: appveyor/ci#1144
VM Log reports: https://ci.appveyor.com/project/Paebbels/ghdl/build/0.34-dev-25

So I can try to implement a GNAT GPL flow on top of that example.

Paebbels, 2016-11-03:
I tried to debug the displayed error message from "Finding 2". Here is my gdb session:
[image]

It shows the same error message. When I try to print the backtrace, `gdb`crashes.

tgingold, 2016-11-03:
On 02/11/16 22:30, Patrick Lehmann wrote:
> Regarding #181 <https://github.com/tgingold/ghdl/pull/181>: The change
> was active in my local branch, because I supplied the pull request :(.
>
> OK, I tested |ghdl-0.34dev-mingw32-llvm| and it's now [...]

Great!!

> SEH sounds like exception handling, right? (I found this:
> http://stackoverflow.com/questions/15670169/what-is-difference-between-sjlj-vs-dwarf-vs-seh)
>
> How do I disable SEH or chose another [...]

You need to edit win32.c and #ifdef/#endif all the code in 
__ghdl_run_through_longjump that is x86 specific.  For x86-64, the 
function will be:

int
__ghdl_run_through_longjump (int (*func)(void))
{
   int res;

   run_env_en = 1;
   res = setjmp (run_env);
   if (res == 0)
     res = (*func)();
   run_env_en = 0;

   return res;
}

Overflows won't be handled correctly on x86-64 but that could be postponed.

> I tried to compile the the vendor libraries with this mingw32-llm build.
>
> *Finding 1:*
> I need to implement a max-file size switch for the PowerShell scripts
> too, if GHDL is using LLVM. (The Bash script has already this option.)
>
> *Finding 2:*
> I got these errors:
> mingw32-llvm1
> <https://cloud.githubusercontent.com/assets/956109/19947124/02c2e0c0-a147-11e6-8d29-d69a8faf4541.PNG>
> Shall I open a new issue for that?

This looks like a memory overflow.  Isn't it a huge package ?

Tristan.

tgingold, 2016-11-03:
On 02/11/16 22:52, Patrick Lehmann wrote:
> *As an update to the AppVeyor based approach:*
> The support provided me a working yaml file, which disables the NX bit
> and reboots the entire VM.
>
> Conversation at AppVeyor: appveyor/ci#1144 [...]

That's great!

Paebbels, 2016-11-04:
Can I see the current memory consumption in GDB at the crash?

tgingold, 2016-11-04:
On 04/11/16 02:56, Patrick Lehmann wrote:
> Can I see the current memory consumption in GDB at the crash?

I don't think so.  Check with process explorer.

Paebbels, 2016-11-04:
**Regarding the memory problem.**

The MinGW64 llvm build is analyzing since minutes and uses around 3,050 MiB. So This could explain the memory problem in the MinGW32 llvm build. In contrast to Linux (3 GiB), Windows applications get only 2 GiB of user memory space.
[image]

I don't think we need to catch such an out of memory exception, because almost every user will use a 64-bit operating system and 64-bit applications. 32-bit application user should know that large designs might crash.

Regarding GDB, I found these 2 posts:
* http://stackoverflow.com/questions/2564752/examining-c-c-heap-memory-statistics-in-gdb
* http://stackoverflow.com/questions/5691193/gdb-listing-all-mapped-memory-regions-for-a-crashed-process

Paebbels, 2016-12-16:
Little progress on this issue:
- [x] MinGW64 is working (I needed to set 2 additional paths in PATH)
- [x] I can execute `configure --with-llvm-config`
- [ ] Start compilation
- [ ] Start test suite
- [ ] Deployment

Paebbels, 2016-12-16:
Hmmm, the preprocessor somehow includes files from Windows ...

I already excluded all path containing "Program Filer (x86)" from PATH, but it still searches these files:

```
In file included from C:\msys64\mingw64/include\llvm-c/Target.h:22:
In file included from C:\msys64\mingw64/include\llvm-c/Core.h:18:
In file included from C:\msys64\mingw64/include\llvm-c/Support.h:17:
In file included from C:\msys64\mingw64/include\llvm/Support/DataTypes.h:36:
In file included from C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\cmath:5:
In file included from C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\yvals.h:8:
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\crtdefs.h:10:10: fatal error
: 'corecrt.h' file not found
#include <corecrt.h>
         ^
```
AppVeyor log: https://ci.appveyor.com/project/Paebbels/ghdl/build/0.34-dev-25#L143

Or is this file / directory searched, because MinGW was build with Visual C++ from Visual Studio 14.0.

tgingold, 2016-12-16:
On 16/12/16 02:32, Patrick Lehmann wrote:
> Hmmm, the preprocessor somehow includes files from Windows ...
>
> I already excluded all path containing "Program Filer (x86)" from PATH,
> but it still searches these files:
>
> |In file included from [...]

That's strange.  Do you mean because clang was built with VC++ ?

I have never used clang on windows, so it is hard to comment.
Was it correctly installed ?

Tristan.

Paebbels, 2016-12-16:
The system contains "everything" ...

The `gcc -print-search-dir` command prints:
```
install: C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/
programs: =C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/;C:/msys64/mingw64/bin/../lib/gcc/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/x86_64-w64-mingw32/5.3.0/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/
libraries: =C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/;C:/msys64/mingw64/bin/../lib/gcc/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/lib/x86_64-w64-mingw32/5.3.0/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/lib/../lib/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../x86_64-w64-mingw32/5.3.0/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../lib/;C:/building/msys64/mingw64/x86_64-w64-mingw32/lib/x86_64-w64-mingw32/5.3.0/;C:/building/msys64/mingw64/x86_64-w64-mingw32/lib/../lib/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/lib/;C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../;C:/building/msys64/mingw64/x86_64-w64-mingw32/lib/
```

There is no Visual Studio 14 path in it.

tgingold, 2016-12-16:
On 16/12/16 20:03, Patrick Lehmann wrote:
> The system contains "everything" ...
>
> The |gcc -print-search-dir| command prints:
>
> |install: C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/
> programs: [...]

Yes, but the c++ file is compiled with clang.  So you need to execute 
clang -print-search-dir (if it exists).

Paebbels, 2016-12-17:
The `-print-search-dir` result for clang is:

```
programs: =
libraries: =C:\msys64\mingw64\bin\..\lib\clang\3.5.2:C:\msys64\mingw64\bin/../lib:C:\msys64\mingw64\bin/../x86_64-w64-mingw32/lib:C:\msys64\mingw64\bin/../lib/gcc/x86_64-w64-mingw32/5.3.0
```

A nice slash and backslash mixture, but no Visual Studio in the paths ...

Paebbels, 2016-12-17:
**GHDL was built, but ...**

Ok first the good news:
After I installed clang (with is not included in the llvm package) I got further. The next trick is to redirect all outputs from STDERR to STDOUT and to convert all pipeline objects to strings. (PowerShell is a object oriented and pipelined system in which processes (or command lets) exchange data through pipes with typed objects.) I don't know why gcc or make are writing to STDERR, especially if they don't write errors ...

However, as far as I understand the make outputs, we have a `ghdl_llvm.exe` and a `ghdl1_llvm.exe`.

```bash
ranlib libgrt.a
sed -e "\!^.[/\\]!d" -e "/-shared/d" -e "/-static/d" -e "/-lgnat/d" \
  -e "\X-L/Xd" < grt-files > grt-files.in
echo "@/libgrt.a" > grt.lst
for i in -ldbghelp; do echo $i >> grt.lst; done
cat grt-files.in >> grt.lst
/usr/bin/make GHDL=/ghdl_llvm.exe ANALYZE_OPTS="--GHDL1=ghdl1-llvm.exe" vhdl.libs.all libs.vhdl.standard
make[1]: Entering directory '/c/projects/ghdl/build/mingw64-llvm'
if [ -d ../../.git ]; then \
  desc=`cd ../..; git describe --dirty`; \
  sed -e "s/[(].*[)]/($desc)/" < ../../src/version.in > version.tmp; \
else \
  cp ../../src/version.in version.tmp; \
fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -aI../../src -aI../../src/vhdl -aI../../src/psl -aI../../src/vhdl/translate -aI../../src/ghdldrv -aI../../src/grt -aI../../src/ortho -aI../../src/ortho/llvm -aI../../src/ghdldrv -gnaty3befhkmr -gnatwae -aO. -gnatf -gnat05 -g -gnata\
 ghdl_llvm -bargs -E -largs 
gnatmake: "ghdl_llvm.exe" up to date.
mkdir -p lib/ghdl/v87
mkdir -p lib/ghdl/src/std
sed -e '/--!V87/s/^/  --/' -e '/--START-!V87/,/--END-!V87/s/^/--/' -e '/--START-V08/,/--END-V08/s/^/--/' < ../../libraries/std/textio.vhdl > lib/ghdl/src/std/textio.v87
sed -e '/--!V87/s/^/  --/' -e '/--START-!V87/,/--END-!V87/s/^/--/' -e '/--START-V08/,/--END-V08/s/^/--/' < ../../libraries/std/textio_body.vhdl > lib/ghdl/src/std/textio_body.v87
rm -f -rf lib/ghdl/v87/std
mkdir lib/ghdl/v87/std
cd lib/ghdl/v87/std; \
for i in std/textio.v87 std/textio_body.v87; do \
  cmd="/ghdl_llvm.exe -a --GHDL1=ghdl1-llvm.exe  --std=87 --bootstrap --work=std ../../src/$i"; \
  echo $cmd; eval $cmd || exit 1; \
done
/ghdl_llvm.exe -a --GHDL1=ghdl1-llvm.exe --std=87 --bootstrap --work=std ../../src/std/textio.v87
/bin/sh: line 3: /ghdl_llvm.exe: No such file or directory
make[1]: *** [std.v87] Error 1
../../libraries/Makefile.inc:137: recipe for target 'std.v87' failed
make[1]: Leaving directory '/c/projects/ghdl/build/mingw64-llvm'
make: *** [libs.vhdl.llvm] Error 2
Makefile:239: recipe for target 'libs.vhdl.llvm' failed
```

So now the bad news:
The problem seams to be the path to ghdl in first for-loop: ` cmd="/ghdl_llvm.exe -a`. There is either something missing (empty variable) or MinGW64 is not recognized by the makefile, because the makefile runs on my local Windows system with a MinGW64 installation. (We fixed that some time ago.)

Paebbels, 2016-12-17:
So I have found the problem, but I have no idea how to solve it ...

`$(PWD)` in the main Makefile is empty. But it is set in `libraries/Makefile.inc`. As far as I know, `$(PWD)` is a predefined make variable, right? How can it be empty? And who fill it, because there is no assignment to `PWD`?

Please see this build log for more details: https://ci.appveyor.com/project/Paebbels/ghdl/build/0.34-dev-39. You can search for `PWD=`:

* log line 165: main Makefile => empty
* log line 704: libraries/Makefile.inc => filled with:
  `https://ci.appveyor.com/project/Paebbels/ghdl/build/0.34-dev-39#L165`

tgingold, 2016-12-17:
On 17/12/16 09:23, Patrick Lehmann wrote:
> *GHDL was built, but ...*
>
> Ok first the good news:
> After I installed clang (with is not included in the llvm package) I got
> further. The next trick is to redirect all outputs from STDERR to STDOUT [...]

Great!

tgingold, 2016-12-17:
On 17/12/16 10:22, Patrick Lehmann wrote:
> So I have found the problem, but I have no idea how to solve it ...
>
> |$(PWD)| in the main Makefile is empty. But it is set in
> |libraries/Makefile.inc|. As far as I know, |$(PWD)| is a predefined
> make variable, right? How can it be empty? And who fill it, because
> there is no assignment to |PWD|?

No, I think that PWD is maintained by the shell.
So, you should try to do sth like: make PWD=xxxx

> Please see this build log for more details:
> https://ci.appveyor.com/project/Paebbels/ghdl/build/0.34-dev-39. You can
> search for |PWD=|:
>
>   * log line 165: main Makefile => empty
>   * log line 704: [...]

I will have a look.

Tristan.

Paebbels, 2016-12-17:
OK, we have pre-compiled IEEE libraries. I have to overwrite some return codes with 0. So we need to look into that issue later. I'm now trying to start `testsuite.sh`... The current build ID is 48, if you want to follow the latest logs.

tgingold, 2016-12-17:
On 17/12/16 11:49, Patrick Lehmann wrote:
> OK, we have pre-compiled IEEE libraries. I have to overwrite some return
> codes with 0. So we need to look into that issue later. I'm now trying
> to start |testsuite.sh|... The current build ID is 48, if you want to
> follow the latest logs.

It looks like there is \ issues:

cat grt-files.in >> grt.lst
install -m 644 libgrt.a C:ToolsGHDL-0.34-dev-mingw64-llvm/lib/ghdl/libgrt.a
install -m 644 grt.lst C:ToolsGHDL-0.34-dev-mingw64-llvm/lib/ghdl/grt.lst
install -m 644 ../../src/grt/grt.ver 
C:ToolsGHDL-0.34-dev-mingw64-llvm/lib/ghdl/grt.ver
install -m 644 -p libghdlvpi.dll C:ToolsGHDL-0.34-dev-mingw64-llvm/lib/
install -m 644 -p ../../src/grt/vpi_user.h 
C:ToolsGHDL-0.34-dev-mingw64-llvm/include/
.\dist\appveyor.test.ps1
Building GHDL and libraries...

Paebbels, 2016-12-17:
I'm using `--prefix=C:\Tools\GHDL-0.34-dev-mingw64-llvm`. This path must be translated to a MinGW compatible path `/c/Tools/...`.

tgingold, 2016-12-17:
On 17/12/16 12:01, Patrick Lehmann wrote:
> I'm using |--prefix=C:\Tools\GHDL-0.34-dev-mingw64-llvm|. This path must
> be translated to a MinGW compatible path |/c/Tools/...|.

Or try: --prefix=c:/Tools/...

I am not sure that /c/Tools/.. works.

Paebbels, 2016-12-17:
`/c/Tools/...` is the path used on my local machine :) That works. MinGW can use `/c/...`, whereas Cygwin requires `/cygdrive/c/...`

I disabled the following failing tests to get further:

- bug032
- bug063
- ticket24
- issue147
- issue67
- ...

Further problems:

- ticket30 throws an exception. Is this intended?
  ```
  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (2016-09-14-340-g12d3f12) [Dunoon edition]
  Compiled with GNAT Version: 5.3.0
  In directory: C:\projects\ghdl\testsuite\gna\ticket30\
  Command line:
  C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin\ghdl.exe --dir ieee
  Exception ADA.IO_EXCEPTIONS.DEVICE_ERROR raised
  Exception information:
  Exception name: ADA.IO_EXCEPTIONS.DEVICE_ERROR
  Message: Invalid argument
  Call stack traceback locations:
  0x57a540 0x57ba43 0x56e603 0x56e694 0x4d3685 0x4d38fa 0x4d4a7c 0x4e9064 0x409226 0x4022fc 0x4013eb 0x401529 0x7ff9979213d0 0x7ff99a1954e2
  ******************************************************************
  ```
- issue176 throws the same exception 2 times:
  ```
  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (2016-09-14-341-gcf68e16) [Dunoon edition]
  Compiled with GNAT Version: 5.3.0
  In directory: C:\projects\ghdl\testsuite\gna\issue176\
  Command line:
  C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin\ghdl.exe --file-to-xml t2.vhdl
  Exception ADA.IO_EXCEPTIONS.DEVICE_ERROR raised
  Exception information:
  Exception name: ADA.IO_EXCEPTIONS.DEVICE_ERROR
  Message: Invalid argument
  Call stack traceback locations:
  0x57a540 0x57ba43 0x56e603 0x56e694 0x468700 0x469f9c 0x4699ad 0x46baa2 0x46b6b4 0x46b7dc 0x4695bc 0x469657 0x46b839 0x46b6b4 0x4695bc 0x469657 0x46b839 0x4695bc 0x469657 0x46b839 0x4695bc 0x46d2f1 0x4e9064 0x409226 0x4022fc 0x4013eb 0x401529 0x7ffafbd913d0 0x7ffafc4654e2
  ******************************************************************
  ```
- ...

(This list may grow.)

tgingold, 2016-12-17:
On 17/12/16 15:10, Patrick Lehmann wrote:
> |/c/Tools/...| is the path used on my local machine :) That works. MinGW
> can use |/c/...|, whereas Cygwin requires |/cygdrive/c/...|

Ok, I didn't know that /c/xxx works.

> I disabled the following failing tests to get further:
>
>   * bug032
>   * bug063
>   * ticket24
>   * issue147
>   * issue67
>   * ...

To be investigated.  They all use more complex shell constructs.

> Further problems:
>
>   * ticket30 throws an exception. Is this intended?
>
>     |******************** GHDL Bug occured ****************************
>     Please report this bug on [...]

Device_error means there was an error when a file was opened.

Congratulation for this achievement.  Maybe you can do a pull request 
for the appveyor configuration file.

Tristan.

Paebbels, 2016-12-17:
Shall I move all AppVeyor related files into directory in `.\dist`?
This is the current layout in `.\dist`: https://github.com/Paebbels/ghdl/tree/paebbels/appveyor/dist

tgingold, 2016-12-17:
On 17/12/16 15:23, Patrick Lehmann wrote:
> Shall I move all AppVeyor related files into directory in |.\dist|?
> This is the current layout in |.\dist|:
> https://github.com/Paebbels/ghdl/tree/paebbels/appveyor/dist

As there are many files, I think you should create dist/appveyor
subdirectory.

Paebbels, 2016-12-17:
OK, I'll create the directory and remove the prefix.

Can you create a new `appveyor` branch (I don't think I have the rights for it) in your repo, so I can create a PR to this branch (paebbels/appveyor => appveyor). I don't think it's good to create a PR against master, because many commits might be needed until it works.

I'll create 2 commits in the PR. One for the appveyor settings and one for the disabled tests, so you can investigate/enable them again.

tgingold, 2016-12-17:
On 17/12/16 15:33, Patrick Lehmann wrote:
> OK, I'll create the directory and remove the prefix.
>
> Can you create a new |appveyor| branch (I don't think I have the rights
> for it) in your repo, so I can create a PR to this branch [...]

Done.

> I'll create 2 commits in the PR. One for the appveyor settings and one
> for the disabled tests, so you can investigate/enable them again.

Ok!

Paebbels, 2017-03-13:
The discussion about packaging the AppVeyor build results can be continued in #318.

Are there tests that are disabled or are all tests also running on AppVeyor?
If so, can we close this discussion/issue?

tgingold, 2017-03-14:
On 13/03/17 22:54, Patrick Lehmann wrote:
> The discussion about packaging the AppVeyor build results can be
> continued in #318 <https://github.com/tgingold/ghdl/issues/318>.

Ok.

> Are there tests that are disabled or are all tests also running on AppVeyor?

I think that all tests are also running on Appveyor.

> If so, can we close this discussion/issue?
ok.

closed 2017-03-14 by tgingold