ghdl/ghdl #177: Make stops on mv with an interactive question.

Paebbels, 2016-10-27:
Hello,

I did a new compilation without `make clean`. I get an interactive question from `mv` if a file should be overwritten. My system is a Debian 9 - testing.

[image]

Translation of the last line:
`mv: overwrite 'ghdl_main.ali', ignore mode 0644 (...)?`

I assume this is a new *feature* in Debian testing and I can configure `mv` to it's old behavior, but other users might get into the same trouble with their Linux installations...

Has `mv` a "force" option?

tgingold, 2016-10-30:
On 27/10/16 22:35, Patrick Lehmann wrote:
> Hello,
>
> I did a new compilation without |make clean|. I get an interactive
> question from |mv| if a file should be overwritten. My system is a
> Debian 9 - testing.
>
> image
> <https://cloud.githubusercontent.com/assets/956109/19784170/09ccc3ba-9c95-11e6-8aa6-d41cbc5e0c73.png>
>
> Translation of the last line:
> |mv: overwrite 'ghdl_main.ali', ignore mode 0644 (...)?|
>
> I assume this is a new /feature/ in Debian testing and I can configure
> |mv| to it's old behavior, but other users might get into the same
> trouble with their Linux installations...
>
> Has |mv| a "force" option?

I plan to allow: make MV='mv -f'

Tristan.

tgingold, 2016-11-01:
MV is now supported.

closed 2016-11-01 by tgingold

Paebbels, 2016-11-01:
So I have to provide a redefinition (alias) for mv to make, right?
Where does this mv behavior come from? Did Debian change something or is it related to changes in your makefiles?

tgingold, 2016-11-01:
On 01/11/16 14:34, Patrick Lehmann wrote:
> So I have to provide a redefinition (alias) for mv to make, right?

No, you can try: make MV="mv -f"

> Where does this mv behavior come from? Did Debian change something or is
> it related to changes in your makefiles?

I don't know.  The access rights look weird.

Tristan.